sumotoy/RA8875 #17: rectHelper has one pixel error

The-Experimentalist, 2015-03-11:
Your rectHelper function causes drawRect and fillRect to create rectangles larger in both height and width by 1 pixel

Should be: 

rectHelper(x, y, x+w-1, y+h-1, color, true);

sumotoy, 2015-03-12:
Oh, tanks! I'll check it out now!

sumotoy, 2015-03-12:
Ok, it's true. Unfortunatly the solution will not so easy as you suggested. I'm actually fixed this issue but found some other functions affected so I'm currently fixing. The incoming 0.69 version have all corrections applied (probably today) but I have to touch some example that result heavy affected (the FFT for mention one).
The main problem with RA8875 is that it's absolutely not out-of-range tolerant so if a value it's not in the range umpredictable results happens.
Thanks again for catch this, I really appreciate help.

closed 2015-03-13 by sumotoy