sumotoy/RA8875 #18: #define _SPI_HYPERDRIVE

The-Experimentalist, 2015-03-13:
Hi

I have been working extensively with the RA8875 and have used many libraries in my project trials. While you have been offline I have used 0.64 of your library and made some changes and additions. When trying 0.68 everything fell apart.

I have now had time to experiment and it seems all my troubles are due to you accidentally "cleaning" the define:

#define _SPI_HYPERDRIVE

I have updated 0.64 to work with Arduino 1.6.1 and Teensyduino 1.21 Beta 10 and found that the reason my code stops working with your versions is the missing define above. I presume your latest workis with some micro other than the Teensy Arm Cortex M range?

Anyway my code comes back to life when I add this #define back in. 

I have some other stuff such as other font ROMs I have tested and new methods that I have added which I will share in due course.

sumotoy, 2015-03-13:
I've the 0.69 soon to be released, lots of changes, cleaned and fixed all errors you point of (as I told you the solution was tricky, you will see changes soon).
I've dropped completely hyperdrive in 0.96, the max SPI speed it's 6.6Mhz and hyperdrive it was developed for drive at almost 30 Mhz!
If you have some cool addittion will welcome, I always mention in code who contributed and this help me a lot.
In the maintime I just discovered 2 minutes ago that my Eastrising chinese display has internal pullup on MOSI,MISO,CS and SCLK! 
This causes problems of any sort with most of the possible devices that share SPI and I'm desperately trying to get an SD work, not for the display (it's useless since it doesn't have fast way to transfer images or large pixel data) but for the Audio Board.
This happen once the display it's initialized so I'm suspecting it's the RA8875 that has another hardware bug.
I wonder if someone can confirm this...

The-Experimentalist, 2015-03-13:
I am using the 7" display ER-TFTM070-5 from East Rising which was ordered with the 3304-1 font chip but has the GT30L32S4W font chip installed. I am also using the on board SD card using Bill Greiman's SDFAT library (https://github.com/greiman/SdFat) and it is working fine at half speed. I have had the SDFAT library working at full speed but my current cables and bread board setup will not allow this. I am using the display in 256 colour dual layer buffer scrolling mode. So when using your library I set maxlayers to 2 for 800x480:

	} else if (_size == RA8875_800x480 || _size == Adafruit_800x480) {
		_width = 800;
		_height = 480;
		initIndex = 3;
		_maxLayers = 2;

and I have an extra method for buffer scrolling:

RA8875.h
//--------------- SCROLL ----------------------------------------
	void		EnableBufferMode(); // Ex - This one is mine
	void 		setScrollWindow(int16_t XL,int16_t XR ,int16_t YT ,int16_t YB);
	void 		scroll(uint16_t x,uint16_t y);
//-------------- DMA -------------------------------

RA8875.cpp

 /**************************************************************************/
/*!		
		Enable concatenated scrolling, creates one large area from both 
		layers

*/
/**************************************************************************/

void RA8875::EnableBufferMode()
{
	uint8_t temp = readReg(RA8875_LTPR0);
	writeReg(RA8875_LTPR0, temp | 0xC0);
}

I also define some more colours:

#define RA8875_LIGHT_GREY       0xB5B2 // 31727
#define RA8875_DARK_GREY        0x4A49 // 19017
#define RA8875_LIGHT_ORANGE	    0xFC80 // 64640
#define RA8875_DARK_ORANGE	    0xFB60 // 64352
#define RA8875_PINK		        0xF806 // 63494
#define RA8875_PURPLE		    0x281E // 10270

For the font chip:

		case GT23L32S4W:
		case GT30H32S4W:
		case GT30L32S4W:
		case ER3304_1:
			temp &= 0x1F; temp |= 0x80;
		break;

One further observation is that from 0.64 onwards the drawPixel method stopped working and when it did work on 0.64 the colour was wrong.

sumotoy, 2015-03-13:
Ok, 
so if you work at 256 colors you haved to modify something else in the init sequence since I'm actually init all displays for 65K colors. I should add an option for that.
About buffer mode, great idea, but I'm not sure about the name of the function since it's scrolling related, maybe change name in something more related?
Font chip adds are welcomed, I own only one.
About drawPixel, thanks for note, I will investigate this tonight, looks something simple since the colours for other functions looks ok.
Witch processor you actually using? I made all my testings with Teensy 3.1 actually.

sumotoy, 2015-03-13:
Also, since you have the same display I have, can you check if MOSI,SCLK and CS are pullup in your display? Eastrising refuse to give me the schematics so I'm investigating if it's an issue of the early batch screens (like mine).
To do that, if you are using a breadboard, open any example then disconnect SCLK wire from display and test if there's a positive voltage (it should not be presente since it's an input but in mine it's present and cause problems with any SPI device).
If you can test MISO and CS as well would be great!

The-Experimentalist, 2015-03-13:
I did originally add some extra defines for 8bpp colour but found that as soon as I enable layers at 800x480 resolution the chip drops the colour depth to 8bpp. After that I just removed them and did not bother any more.

I named the method after the datasheet ( I have version 1.6) on page 31, REG[52h] Layer Transparency Register0 (LTPR0), the description is "Buffer scroll (using Layer 2 as scroll buffer)". I am not precious about the name so I am happy with you calling it whatever suits, I guess setBufferScroll(bool) or enableBufferScroll would be better but I am happy with whatever you decide.

I am currently also using a Teensy 3.1

With regards to the pullups I will check for you later and report back.

The-Experimentalist, 2015-03-13:
Are we using the same screen? 

I thought you were using the 5 inch screen and the native touch functionality of the RA8875. I have the 7 inch screen with the capacitive touch screen and the FT5206 touch screen controller.

The-Experimentalist, 2015-03-13:
I have added more to your thread on the PJRC forum, probably better to continue this there:

https://forum.pjrc.com/threads/28034-the-infamous-RA8875-(again!)-pullups-on-every-SPI-pins

sumotoy, 2015-03-14:
The 8 bit support has many registers affected, color addressing has to be changed but before all to set the display at 8 bit this register has to be changed:
REG[10h] System Configuration Register (SYSR) bit 3,2
In my library I have this common for ll display: writeReg(RA8875_SYSR,0x0C);
This set display at 65K. To set at 8 bit:
writeReg(RA8875_SYSR,0x00);
I've already build up a new 0.69b2 version that has full 8 bit support and allow you to use even 65K where you need and change to 8 bit when you need layers and automatically expand layers to 2 as soon you are in 8 bit, then switch back to 1 layer when you are at 65K, I think that this it's a pratical solution! I'm just testing it to see if there's other stuff I need to change on the switchover.
Btw I cannot understand why you still need HYPERSPEED defined since I totally deleted all code from the library!
About the color issue, accordly to datasheet the RA8875 need a different way to set color when in 8 bit mode, I added support for that in the 0.95b2, the datasheet claim that RA it's able to write 16 bit color from a 8 bit one by using it's hardware capabilities. I really hope to post a working beta for you lately (actually I have to deal with a couple of issues).
I've also prolly fixed all graphic primitive dimension error you find plus many others.
Thanks for help, really appreciated!

closed 2015-04-19 by sumotoy