sumotoy/RA8875 #19: 0.69 not working for me

The-Experimentalist, 2015-03-13:
When adapting 0.69 with my additions I have back the exact same issues that I had before adding the 
#define _SPI_HYPERDRIVE back in to 0.68

I have not had time to investigate further at this point

sumotoy, 2015-03-13:
Well, this is weird since HYPERDRIVE it's gone from the 0.69, the code has been removed (now it's simpler). I'm introducing a new feature of 1.6.0 ide (and Teensyduino 1.21) that allow 16 bit SPI transfer and should activate only when you have the correct IDE, it should speedup a bit.
Try to rename the old library and use the library folder, test it, then add your adds.

The-Experimentalist, 2015-03-13:
Sorry just to make it clear I have not investigated in anyway. I have added the absolute minimum in order to get my project to compile and then downloaded to the Teensy 3.1 and the display is just the same as 0.68 before I add the HYPERDRIVE define back in. I have not tried to add it back in to 0.69 as I was aware you have made many changes and it is no longer used. I have made no further investigation yet. I will report back if I find anything. I may try and send you an example which demonstrates the fault if I get time.

The-Experimentalist, 2015-03-13:
Think my issue may be caused by the SD problems everyone is talking about but why would it work with 0.68 and not 0.69?

Could you post a link to a supplier for the 100nf capacitor in the following page so I get the right part?

https://github.com/sumotoy/RA8875/wiki/Fix-compatibility-with-other-SPI-devices

sumotoy, 2015-03-14:
It's a 100n polyester, any type, not critical, used just for decoupling supply, it can also ignored but better install for avoid oscillations.
A bit explain on HYPERDRIVE, it's optimized FIFO SPI routine that can reach very high SPI speeds and reach the limit of Teensy hardware capabilities. It's really cool, complicated stuff but actually I decided to rip off since the RA8875 works at max speed of 6.6Mhz in write and 3.3Mhz in read!
The FIFO helps to get SPI data faster but at these speed the improvements are really tiny and code became a big mess, also needs changes for the new version of Teensyduino and NOT work with Teensy LC since this MCU has different SPI FIFO structure and need completely different code (it also not capable of the same speeds). Actually the code it just use standard SPI calls that in a Teensy 3.1 are quite fast. Once code has been tested and no more errors I can add again HYPERDRIVE so you are able to reach the max speed possible (and now without the useless condenser should be much better), I saved all procedure and wait to add Teensy LC structure for compatibility.

The-Experimentalist, 2015-03-14:
I just tried version 0.69b2 with Arduino 1.6.1 and Teensyduino 1.21 Beta 12 and I have the same problem back and my project does not work. 

I just tried my modified version of 0.68 with Arduino 1.6.1 and Teensyduino 1.21 Beta 12 and all works fine. I have ordered 74HC125 and some 100nf capacitor but these will not arrive for some days. I don't intend to try any more with the new version until then really.

Everything works just fine with 0.68 including using the SD card. It seems to work faultlessly so I am a bit confused, no problems except for the incorrect pixel colour when drawing single pixels

Were you going to add the extra Font chip defines and the buffer scroll method at some point?

I am using enableBufferScroll() as the method name at the moment

sumotoy, 2015-03-14:
Wait, did you using the SD card and library whe you tested the 0.69b2?

sumotoy, 2015-03-14:
Yesterday arrived the 800*600 display here, just tested (without SD) and works!

The-Experimentalist, 2015-03-14:
My project does not really do anything without the SD card functionality. It initialises the RA8875, configures it, initialises the SD and then loads data from SD. So yes whatever library I am using I also use SdFat

The-Experimentalist, 2015-03-14:
The display and the SD card both initialise fine, and the initial display actions work fine.

Can you post a link to the 800x600, sounds interesting

sumotoy, 2015-03-14:
So send me the old 0.68 library, I don't have anymore here. I can see why it's working even if MISO goes UP after initialize chip, I'm pretty sure it's something related to SPI transaction.
sumotoy at gmail.com

sumotoy, 2015-03-14:
OOps my fault 800*480.

The-Experimentalist, 2015-03-14:
0.68 is effectively commit 82a6c4953bd811fb65a5ccc2d27bc5080cf52211

sumotoy, 2015-03-14:
Accordly commit 82a6c4953bd811fb65a5ccc2d27bc5080cf52211, only Adafruit 640x480 was added! cannot be, lot's of changes till now. Nevermind, I will look around.
I'm pretty sure that you got working for some weird reason, maybe the SD card you are using has strong MISO signal, maybe whatever! In some minute I got the 74HC125 adaptor ready for the 800*480 and will test with SdFat.
I'm only thinking that you have a version with SPI transaction not fully implemented.

The-Experimentalist, 2015-03-14:
If you just clone your own repo to a new folder:

git clone https://github.com/sumotoy/RA8875.git

and then checkout that commit:

git co 82a6c4953bd811fb65a5ccc2d27bc5080cf52211

That is 0.68 !

The-Experimentalist, 2015-03-14:
0.64
====
commit 23f5fa6e9122c437942ada451df83a377d37faeb

0.68
====
commit 82a6c4953bd811fb65a5ccc2d27bc5080cf52211

0.69
====
commit 99bdfa838f5ff4008e5feb00df6c439303bf9d0b

0.69b2
======
commit 8303d0bc026d8801b8addca72ebad2d579dac2c3

The-Experimentalist, 2015-03-14:
Just checked and my SdFat library was out of date. However, I have cloned the latest from:

https://github.com/greiman/SdFat.git

. . . and retested, everything still working with 0.64 and 0.68 but not with 0.69 onwards

sumotoy, 2015-03-14:
0.64 -> clearly an error, the commit is exact!
0.68 -> Adafruit 640x480 added
0.69 -> Only text changed?
0.69b2 -> Only text changed?
Following commit seems that I have changed almost nothing! Check yourself
It cannot be. I've found a 0.67 and 0.68 untouched and lot of changes but apart SPI.transfer16 and HYPERDRIVE nothing related to SPI. Since HYPERDRIVE it's fast it can happen that CS close transaction so fast that SdFat can interact without colision but this depends of so many stuff (the type of SD card, lenght of cables, etc. etc.). 
To proof that, I just used the 0.68 with SdFat, standard SPI line, a level 10 SD card, a level 4 SD card and...nothing work.
Switched on 0.67 (with Hyperdrive ON), same thing. After 6 reboot I got the Level 4 initialize but RA8875 stop working after 10 sec.
As I said, too much variables.<br>
Installed SPI circuit with 74HC125 - > RA8875 MISO.<br>
All tre variants works with SD and SdFat, both SD cards.

The-Experimentalist, 2015-03-14:
I have a different folder for each time I have cloned your repo. The last commit you made on master represents the version at the time as I see it. I am no Git expert but seems sensible :0)

I am off to bed now but I will sendawebster

The-Experimentalist, 2015-03-14:
Thanks for all your time, good night, speak soon

sumotoy, 2015-03-15:
No problem, I will be happy only when everithing worked as it should be.
I've tested only 4 and 8 giga SD card btw, it's almost impossible find smaller one today.
I have to thank you for help!

sumotoy, 2015-04-19:
I have received the same display of yours, tested with the last release here and... surprise, everithing works as it should, no freeze, layers switch automatically at 8 bit as supposed and every example (included layers one) worked, apart an issue with the maxy fft display that's has error in example I have to fix.
I've not tested against other SPI libraries so no SD or SDfat till now but it's enough to say that library works as it should.
At this point I'm pretty sure you have some problem with your application or better (I suspect) you are using SdFat from Greyman without changing the SPI Transaction option inside greyman library.
I close this here but if you have any other info you can open another...

closed 2015-04-19 by sumotoy