sumotoy/RA8875 #22: it doesn't work in waveshare LCD

mamaikus, 2015-04-01:
i use lcd 7 inch from waveshare http://www.wvshare.com/product/7inch-Capacitive-Touch-LCD-C.htm
but i my lcd only appear grid line.could you help me?

thanks

sumotoy, 2015-04-01:
well, I'll try. Actually my display it's fried and also never used this one. 
Are you sure it's setup to work for SPI?
Can you provide me a link for product datasheet?

mamaikus, 2015-04-01:
even i didn't try the SPI yet, i meant that i just give it power which is 3,3V and i got my lcd panel show gridline instead of white blank.for the first time i think there was problem with my lcd panel but after i change the lcd panel for the third time the problem is the same.
unfortunately, i only got the pin table(in that website) instead of datasheet.

sumotoy, 2015-04-02:
ah, you NEED to ask at list the schematics to your vendor because with just the link you provide me it's not possible, the RA chip can work in SPI 3 wire, SPI 4 wire, I2C, Parallel 16 bit, Parallel 8 bit and more, you know how it's actually configured? They MUST provide you something to configure the display, with eastrising ones I had to configure some solder jump on the back.

sumotoy, 2015-04-02:
Wait a moment... After a closed look I've seen a pin called PS (26), you should connect to 3V3 to get SPI 4 wires working.
Then you should use the SPI LCD port (MOSI,MISO,SCK and CS). You will need to comment out USE_EXTERNALTOUCH in RA8875.h file to avoid the library management of the RA8875 resistive touch since you are using a display that have capacitive one drived by I2C (probably, cannot see more from picture).

mamaikus, 2015-04-03:
still the same. Here the gridline which i meant
https://fbcdn-sphotos-h-a.akamaihd.net/hphotos-ak-xpf1/v/t34.0-12/10917630_10203546769689426_1741575791_n.jpg?oh=451571eb80970795c951f37d7e7249fe&oe=55200C19&__gda__=1428162184_9d62beb038bdb707acd007d67ca764b2

i try to buy the new lcd but it's hard to find here

sumotoy, 2015-04-03:
Looks like the classic screen of a display that has not been inited! Maybe it's not fried as you think...
Can you tell me the speed of the Xtal mounted near the RA8875?

sumotoy, 2015-04-03:
Try also this, connect all SPI lines but LEAVE unconnected reset pin!

mamaikus, 2015-04-03:
it use 20Mhz for Xtal.
oh,so the problem is not the lcd panel but the driver right?
i am trying to trace the circuit to get the schematic. i will inform you as soon as possible.
thank you for your great support and help

sumotoy, 2015-04-03:
Driver it's fine,has been proof that worked with several displays.

sumotoy, 2015-04-04:
Ok,
please provide me an high resolution image of the SPI connector (the 14 pin on the bottom/right), I cannot see in any image the text.
Also...
1) Wich processor you are using?
2) Wiring.

mamaikus, 2015-04-04:
here
[image]
[image]

sumotoy, 2015-04-04:
Thanks.
So you connected as this one?
https://github.com/sumotoy/RA8875/wiki/Wavesharing-Connections-%28beta%29
Also:
Wich processor/board you are using?

sumotoy, 2015-04-19:
no responses, in the meantime I got the same display and...works. I close here.

closed 2015-04-19 by sumotoy

thehose, 2015-04-30:
Sumotoy, I'm starting a Arduino project and part of that is this exact LCD/TFT board (maybe not such a good choice) so I was happy to find your work. Did you say (to mamaikus) that you now have this board and confirmed that it works with your library ?
If so, I expect to receive this in a couple of weeks and plan to use your wiring connections (WiKi) to test the unit.

DrewLX, 2015-04-30:
I'm trying to get the Waveshare module to work with an Arduino Mega.

The SPI pins all pass through a logic level shifter module from SparkFun. However I still get the same gridline garbage on the screen.

sumotoy, 2015-04-30:
It should work, any board I've seen have RA8875 SPI direct exposed (apart Adafruit board has a level converter). I got positive responses from UNO users both on adafruit and eastiring, 472x480 and 800x480, works with no problem. However I don't know the waveshare board, I've ordered a 800x480 but still not here, there's no reason should not work btw. Did you have a datasheet or a schematics to see?
On the Mega should work as in UNO, I'm using standard SPI calls, just be sure to use the SPI pins for the mega board (50 (MISO), 51 (MOSI), 52 (SCK), 53 (SS)). If you see lines it's probably not inited there's a problem on SPI or the waveshare has it's internal reset (try to cold start without RST cable).The best is take a look to schematics.

sumotoy, 2015-04-30:
I've noticed that waveshare has a pin to select the port, I got a 472x480 time ago in a shop and we tested as the schematics I've writed here https://github.com/sumotoy/RA8875/wiki/Wavesharing-Connections-%28beta%29  for a couple of min and it's turned on and showed 'hello world' but was on Teensy without a level converter.

thehose, 2015-04-30:
I will be using an Arduino Leonardo for my project. Can you answer a question for me given your connection (Beta 29). The SPI interface is for the LCD and the I2C is for the TFT interface, is this correct ??

sumotoy, 2015-04-30:
yes

thehose, 2015-04-30:
It does, they are using a Goodix GT811.... I'll monitor this thread and as soon as I get the board, I'll post my results...

DrewLX, 2015-04-30:
Its possible in my setting up and testing I gave it a 5v SPI before realising I was being an idiot. Might this have killed the RA8875?

I'm going to switch back to the Adafruit interface and make sure I can get your library to work with that, then continue to experiment with the Waveshare model.

thehose, 2015-04-30:
Ouch !!! that sucks

sumotoy, 2015-05-01:
I don't think so... It happened to me few times and still alive, it's not 5v tolerant but for a small time can survive. Did you use 5v for long time?

DrewLX, 2015-05-01:
Probably no more than a few minutes. I wouldn't expect that to kill it. I've got some time next week to experiment and try a bunch of different things. I'll test the Waveshare with an Uno or a Teensy. and I'll test the Adafruit board with my Mega (tested with adafruit library)

mamaikus, 2015-05-05:
i give up for the waveshare board, i bought 3 times and still same..

the code work so well when i use eatrising board.

bytheway,is there any lcd controller which is good for diplay image file?because the ra8875 is too slow to diplay image

sumotoy, 2015-05-05:
I suspect that waveshare board it's not configured for SPI, there's some missing. They put a pin to do that but looking at RA8875 datasheet its' not enought touch one pin! Are you sure that you can't get a schematics from waveshare?

thehose, 2015-05-28:
Well I'm back and have all the components for my project. I just connected the SPI (only) interface on the WaveShare 7" LCD/TFT to my arduino Leonardo according to your diagram (except for the 3.3v - PS jumper and no CS). When I power the board, I get a vertical rainbow of single lines.
I'm going to download your latest library (0.70) and use the Drawing_Tests.ino from your examples to test the video. 
I'll let you know what happens.....

sumotoy, 2015-05-28:
Did you using a 5V -> 3V3 level converter?

thehose, 2015-05-28:
No, I just took the 3.3V from the arduino

sumotoy, 2015-05-28:
Stop! You need a level converter! You can damage the chip!

thehose, 2015-05-28:
Is this what you are talking about... https://www.sparkfun.com/products/12009

sumotoy, 2015-05-28:
Probably it's still working, I've noticed that RA it's well made and can resist even for some sec to polar inversion but on datasheet it's clearly specified that this chip it's not 5v tolerant.
Use a CD 4050 (powered at 3v3 volt) for MOSI,CS and SCLK, MISO can go directly to arduino

thehose, 2015-05-28:
The Leonardo provides both 5v and 3.3v. What is the difference ??

sumotoy, 2015-05-28:
I prefere to use 74HC4050 for that job, are much  faster, the level converter based on mosfet are terrible with fast signals.
Leonardo has 3v3 but his I/O works at 5V!

thehose, 2015-05-28:
I'll have to get one of those and work it into the board. However do you think I can test with the 3.3V from arduino ?

sumotoy, 2015-05-28:
It's near the limit of the Leonardo internal regulator, it should supply enough current. But you cannot connect SPI wires directly from leaonrdo to RA8875 without a level converter, you will damage your TFT for sure!

sumotoy, 2015-05-28:
But you have datasheet for your tft? Waveshare should supply to you some documentation...

thehose, 2015-05-28:
OK, I'll get a 74HC4050 chip and wire that into my bread board. In the mean time, since I'm not a hardware guy, I'll google for a 74HC4050 / SPI schematic. I'll let you know when that's in place.

sumotoy, 2015-05-28:
It's really simple, 74HC4050 it's 5V tolerant and it's just 6 non-inverting ports, supply it at 3V3, connect arduino to it's input and 4050 output to RA. The MISO from RA you can connect directly to Arduino.

sumotoy, 2015-05-28:
I don't know the touch chip of waveshare, prolly a I2C if capacitive. I think it's a ITXsomething, in that case my library doesn't support it yet, remember that for I2C you need a special level converter as well and this time not the 74HC4050 since I2C it's bi-directional!

thehose, 2015-05-28:
This board uses the Goodix GT811. Does your lib support it ?

thehose, 2015-05-28:
Teensy does not require a 74HC4050 for 3.3v power ?

sumotoy, 2015-05-28:
Teensy does not require a 74HC4050 for 3.3v power ? -> NO It has I/O 3v3 but all digital I/O are 5V tolerant as well so you don't need a downside voltage level converter.
I do not support GT811, I don't have it, maybe in the future if waveshare send me a display I will.

thehose, 2015-05-29:
So I looked at the Teensy 3.1 and decided to get one; It should be here tomorrow. At this point is just a matter of connecting SPI to SPI directly on both the display and Teensy.
While your lib will (may) not support the I2C touch, I should be able to load and run one of your examples such as "Mandelbrot" to test the setup.

sumotoy, 2015-05-29:
Teensy and DUE works as is, they are both 3V3. Teensy it's easier since it have the same pinout of UNO and much faster, smaller, smarter, cheaper and supported than DUE!
Since Teensy 3.1 has 3V3 regulator inside chip you will need an external supply for display, something like this
www.ebay.com/itm/AMS1117-3-3V-3V3-800mA-AMS1117-3-3-Linear-Voltage-Regulator-Module-UK-SELLER-/261858129034
or similar.
About Waveshare TFT
I don't know witch exact model you have:
http://www.wvshare.com/product/7inch-Capacitive-Touch-LCD.htm
http://www.wvshare.com/product/7inch-Capacitive-Touch-LCD-B.htm
Both LCD Interface:24-bit parallel (not work in SPI)
http://www.wvshare.com/product/7inch-Capacitive-Touch-LCD-C.htm
LCD Interface:8080 parallel (not work in SPI)
You have to **force** waveshare to give you a datasheet (of the display, not the RA8875) or documentation about your display, the info online doesn't mention how to force display to run in SPI mode!

thehose, 2015-05-29:
I have the http://www.wvshare.com/product/7inch-Capacitive-Touch-LCD-C.htm, so this will not work with your linbrary or with teensy (pins). Is that correct ?

sumotoy, 2015-05-29:
I'm really confused about this display,:
Interface:8080 series interface
What does mean? 8080 it's parallel but they put word 'series'.
You have to ask waveshare shematics of this particular display and/or datasheet, they HAVE to provide you infos or it's impossible for any library access this display.
RA8875 can be configured for many protocol by jumpers, resistors, pullups, etc.

thehose, 2015-05-29:
They have all their docs, schematics and datasheets @ http://www.waveshare.com/wiki/7inch_Capacitive_Touch_LCD_(C)

sumotoy, 2015-05-29:
Good catch! I'm really happy for this, finally I can figure out configuration and help.

thehose, 2015-05-29:
Well it would be really nice if I can use my WaveShare with your library. I expect to have it up and running (soldering the pins and running some samples. I already have a 3.3v regulator on my breadboard, so I should be good to go by Sunday. 
Good luck with the new release, I'll come back to this thread when my Teensy 3.1 is up and running...

Thanks for all the help and advise so far....

thehose, 2015-05-30:
Well my new Teensy 3.1 is DOA... Should have a replacement by Monday or Tuesday....

sumotoy, 2015-05-31:
DOA?

thehose, 2015-05-31:
Yeah, like a door nail small led blinks once when you plug it in and THATS IT ! Mac does not see the USB connection. It happens here and there.
Have you looked at the datasheets for my LCD. I did a bit more digging and that display sells in Amazon and it's advertised as 4 wire SPI capable...

sumotoy, 2015-05-31:
Well to know, cross fingers,, I've sent an email to waveshare, if they don't respond I will write here in wiki and it's clearly not a great publicity for them, it's the third time I send an email and no response. Btw their code on the page you give me yesterday it's all for parallel 24 and 16.
To get working on SPI without any info from them (even in your link there's no word about!) the only way it's compare the adafruit board schematics and discover witch RA8875 line has to be set on the waveshare. The adafruit board works only in SPI and has schematics as well.
Too bad for the Teensy, I've buy more than 50 and no one has troubles, never, when you solder pins please pay attention because several tiny components are very near pin, it's a tiny device but really powerful, much more than DUE at a fraction of the cost.
I have totally replaced all arduino's from 3 years thanks Teensy.

thehose, 2015-05-31:
I saw this reference in Amazon here:

 http://www.amazon.com/Waveshare-Capacitive-Multicolor-Display-Backlight/dp/B00ID965PO/ref=sr_1_cc_2?s=aps&ie=UTF8&qid=1433078850&sr=1-2-catcorr&keywords=waveshare+7+inch

[image]

However I could not find that same reference in the Waveshare site... 

I hope you have better luck that I've had getting Waveshare to respond to emails.

As for Teensy, I was very careful when soldering the headers, in fact it was one of the best looking soldering jobs I've done, looked like it was dipped at the factory. Who knows, it was not a board from PJRC so it could have been a Chinese clone. The new one I have coming is from PJRC and has the pins already soldered, so it SHOULD be good to go.....

thehose, 2015-06-01:
Well, I heard from China after all. I'm not sure if they are saying no SPI or they don't have any arduino info... As usual Chinese support is useless.

Hi Jose Martinez

We have development information for this LCD here http://www.wvshare.com/downloads/accBoard/7inch-Touch-LCD-C.7z 
But we don't have information for connect this display SPI/I2C to arduino....  sorry about this.

Best regards
Johanna

在2015年05月29 08时39分，"Jose Martinez Jr."<thehose@aol.com>写道：

I purchased this display through AliExpress; Do you have any support documents on how to connect this display SPI/I2C to arduino ?

sumotoy, 2015-06-03:
The code it's for 24 bit/16bit parallel and only STM series processors, I've spend some time to send email to them, offer support code but they NEVER respond! The last chance you have it's comparing Adafruit board schematics with the Waveshare one and try to figure out what you need to jump/pullup/pulldown to force RA8875 to run in SPI mode, than theorically it's possible use my library, I say 'theorically' because eastrising and adafruit uses 2 different variants to turn on display TFT.

thehose, 2015-06-04:
Well this waveshare will not work on SPI for sure, so I will return it. I looked in your GitHub page and I beleive this one will work with your RA8875 library (and others)…

http://www.buydisplay.com/default/7-inch-lcd-module-capacitive-touch-screen-panel-i2c-spi-serial <http://www.buydisplay.com/default/7-inch-lcd-module-capacitive-touch-screen-panel-i2c-spi-serial>

Can you confirm this for me (before I order) and advise on the configuration, specially the Font Chip….

Thanks

thehose@aol.com <mailto:thehose@aol.com>

sumotoy, 2015-06-04:
They both work (look wideo on wiki) but when you order select **4-wire-SPI-interface**

thehose, 2015-06-04:
Great, I will not get the Font Chip. I will let you know when I get it up and running with Teensy 3.1...

BTW: I noticed you are in Milano. I was there for two weeks on business a few years ago and stayed in a very nice hotel next to the Duomo. There's a gelato store in a promenade near by named "1954" or something like that that was irresistible to me and I gained 6 pounds in that time. Of course great food too. Wish I could do that again !!! :-))

Thanks again for all the help.....

[3 more replies]