sumotoy/RA8875 #24: Request for update to previously included code

The-Experimentalist, 2015-04-05:
Hi

After further work with layers and your code version 0.68 I have recently updated my previously submitted enableBufferScroll method and think you will want to include the improvements as the code is a better fit with the datasheet.

My changes:

RA8875.h

Removed: 
void		enableBufferScroll();
Added:
enum RA8875scrollMode{ SIMULTANEOUS, LAYER1ONLY, LAYER2ONLY, BUFFERED }; //for LTPR0
void		setScrollMode(enum RA8875scrollMode mode);

RA8875.cpp

Removed:
/**************************************************************************/
/*!	
Enable concatenated scrolling, creates one large area from both
layers
Author: The Experimentalist

*/
/**************************************************************************/
void RA8875::enableBufferScroll() {
	uint8_t temp = readReg(RA8875_LTPR0);
	writeReg(RA8875_LTPR0, temp | 0xC0);
}

Added:
 /**************************************************************************/
/*!		
		Sets the scroll mode. This is controlled by bits 6 and 7 of  
		REG[52h] Layer Transparency Register0 (LTPR0)
*/
/**************************************************************************/
void RA8875::setScrollMode(enum RA8875scrollMode mode)
{
	uint8_t reg = readReg(RA8875_LTPR0);
	reg &= 0x3F;			// Clear bits 6 and 7 to zero
	
	switch(mode){			// bit 7,6 of LTPR0
		case SIMULTANEOUS: 	// 00b : Layer 1/2 scroll simultaneously.
			// Do nothing
		break;
		case LAYER1ONLY: 		// 01b : Only Layer 1 scroll.
			reg |= 0x40;
		break;
		case LAYER2ONLY: 		// 10b : Only Layer 2 scroll.
			reg |= 0x80;
		break;
		case BUFFERED: 		// 11b: Buffer scroll (using Layer 2 as scroll buffer)
			reg |= 0xC0;
		break;
		default:
			return; 		//do nothing
		break;
	}
	
	//TODO: Should this be conditional on multi layer?
	//if (_useMultiLayers) writeReg(RA8875_LTPR0,reg);
	writeReg(RA8875_LTPR0,reg);
}

sumotoy, 2015-04-05:
Ok, accepted. About conditional or not, LTPR0 affect both multilayer capable or not display, so probably the switch must have some conditional statement in case of multilayer enabled or not, have to check but I think we can leave as it is...

The-Experimentalist, 2015-04-06:
Okay, great. I have been using the above amendment for some time with 0.68 version of the library and it has been working fine form me.

closed 2015-04-19 by sumotoy