sumotoy/RA8875 #29: setRotation...

sumotoy, 2015-04-19:
Thanks Morgan for setRotation, I was planning to include but you was faster, really appreciate!.
The last beta has your setRotation included, btw I'm currently modify a bit since if the function has to mimic the adafruitGFX some other variable should be affected.
For example, setRotation(1) will result in a W:800, H:480 (in case of 800x480 display) but some function will fail because it should be W:480, H:800!
Since the current code uses the screen width and height for some calculations I have to check code if switching rotation dinamically needs some mod so several functions.

sumotoy, 2015-04-22:
The setRotation bring more issues than I was suspecting, in portrait mode the setCursor doesn't work as it should and after digging I have to modify both setCursor and setRotation.
Another problem it's the setFontRotate that if used when text is in portrait mode (rotation 1 o 3) it messup everithing, fixing this added quite a lot of code so I'm wondering to send setFontRotate into private area and leave as is (I'm not sure, I don't know if someone use this function a lot, to me can be useful only for charts labels).
I know that must be something else to fix so I still not posted fix but will be finally fixed everithing in the next beta.

sumotoy, 2015-04-24:
I've released 0.69b21 that fix all the issues of setRotation. At the end of the story was not easy at all fix this without compromising code in therm of speed! I've also find another small bug of the RA8875, when the direction register has been touched the RA8875 forgets the textColor assigned, this has also been fixed in library.
Now you can run any example in any rotation set without issues, I've also erased the Morgan basicRotation example with a brand new one that demostrate the correct positioning of the drawed object on screen when you perform a rotation (without library support on this it's a nonsense).
Since this was tricky to fix maybe something else can be still affected, if someone finds out more problems please let me know.

MorganSandercock, 2015-04-26:
I've reviewed your code in the latest beta. That looks good. I'm not going to use it myself as I think the performance hit of swapping x and y all the time is going to be bad for my application on a 16MHz Arduino, but doing it that way will help the people who just want to download the library and have it work right the first time.

sumotoy, 2015-04-26:
There's no other way to get rotation since RA8875 has it's internal buffer but leaving rotation at 0 doesn't affect performance, the degradation it's almost not visible with things like Teensy 3.
That's why it's possible to use such large display with 8 bit MCU like arduino uno.

sumotoy, 2015-04-27:
Just a trough, I'm using an inline swap function, really fast, just checked performances on a UNO and really tiny degradation!
A great result if you consider that 100% of the library still working as it should at any rotation :)

sumotoy, 2015-04-27:
Apart resistive touch screen that I still have to fix for rotations....

MorganSandercock, 2015-04-27:
Well, that should be simple to just reverse the swap when returning numbers from the touchscreen.
if(_rotation==2) y=-y;

sumotoy, 2015-04-28:
Yes, I know. But have to reconnect my 480x272 and actually I'm stuck with this beautiful 800x600, it will take some minutes to fix but get also excited with font rendering, I will do tomorrow.

closed 2015-04-28 by sumotoy