sumotoy/RA8875 #32: userCustomChar issues with font scale not equal to 0

sumotoy, 2015-04-24:
Looks like another RA887 bug. the user custom char works but if font scale it's different from 0 it will not work and give back random stuff. I'm investigating...

MorganSandercock, 2015-04-26:
I looked at this too. There's no reason why it shouldn't scale custom chars exactly the same as built-in but the simple answer is "It doesn't." The random stuff on the screen (in the right colour and in the right place) seems to indicate that it's doing something. Maybe it's reading from another area of memory - you might have custom chars defined for different sizes in different places.

sumotoy, 2015-04-26:
Looks like an RA8875 bug, I've some email exchanges with RAio technician a couple of days ago.
During the set rotation digging I got another couple of bugs that has been confirmed by Raio, I will report here and try to figure out a solution.

sumotoy, 2015-05-30:
It's definetively an hardware bug, cannot get around it, with big screen like 800x480 it's even worst, I leave this note for postery since cannot be solved, RAiO confirmed.

sumotoy, 2015-06-30:
Fixed in the new upcoming 0.70b5 library.
Since it's impossible get around this, I have eliminated the influence of font scaling so it will not affect showUserChar function.

closed 2015-06-30 by sumotoy

MorganSandercock, 2015-06-30:
Thanks for the feedback. It's disappointing seeing how many bugs are in this chip. If there weren't projects like this trying to make use of all the advertised functions, RAIO may never have found some of them.

WerkAG, 2015-09-25:
Hi sumutoy

In first place, congratulations for your project of making a full featured RAIO 8875 library.

The userCustomChar issues (RAIO bug) may not be only related to the font scale .
I'm doing some tests with an EastRising 800x480 display and your library:
The userCustomChar example included in the library works great, but when I try to print a custom user char with  some code I'm writing, just after the command to display the custum char, the display seems to zoom only in the horizontal position, colors also change (I'm not using layers).
Until now I'm unable to determine what in my code, breaks the userCustomChar function, so I'm only reporting it.
Below is an image of the screen, after printing one custum user char.

[image]