sumotoy/RA8875 #35: oldpapa

Oldpapa, 2015-04-29:
Using adafruit with teensy 3.1
Black screen using tft.begin(Adafruit_800x480);
If tft.begin(RA8875_800x480); I get a white screen.  

Either case, nothing displays.

sumotoy, 2015-04-29:
Did you have any other stuff connected to SPI?

sumotoy, 2015-04-29:
I think I find the issue, fixing now...wait

Oldpapa, 2015-04-29:
Nothing, only display

sumotoy, 2015-04-29:
looks like the release online missed the gpio command where adafruit connected the display backlight. In a min I give a piece of code to add and you should be ready to go

Oldpapa, 2015-04-29:
ok, be here

sumotoy, 2015-04-29:
In the RA8875::initialize() function , after displayOn(true); add...

	if (_size == Adafruit_480x272 || _size == Adafruit_800x480) GPIOX(true);//only for adafruit stuff 
	PWMsetup(1,true, RA8875_PWM_CLK_DIV1024);//setup PWM ch 1 for backlight
	PWMout(1,255);//turn on PWM1

sumotoy, 2015-04-29:
tft.begin(Adafruit_800x480);

Oldpapa, 2015-04-29:
It Works..!!

sumotoy, 2015-04-29:
so there's an issue on backlight, I have to fix this...

Oldpapa, 2015-04-29:
Ok, well, I was not crazy after all
I can now use your library to do more..

Great, TY Sir!

sumotoy, 2015-04-29:
This library it's much more fast than adafruit and also doesn't have a lot of errors never fixed on adafruit one, the list is too long, this is wy I waste my time to develop it,

Oldpapa, 2015-04-29:
You are not wasting time..  Not at all..

sumotoy, 2015-04-29:
Can you please go tho Teensy forum and says this is solved? Thanks for your time!

Oldpapa, 2015-04-29:
Yes I will..

closed 2015-04-29 by sumotoy