sumotoy/RA8875 #40: Having issue to build examples on DUE

gcharles81, 2015-05-04:
When building any of the examples sketches on DUE having below errors

Arduino: 1.6.3 (Windows 7), Board: "Arduino Due (Programming Port)"

\Documents\Arduino\libraries\RA8875\RA8875.cpp: In member function 'void RA8875::drawPixels(uint16_t*, uint32_t, int16_t, int16_t)':

\Documents\Arduino\libraries\RA8875\RA8875.cpp:2259:7: error: 'class SPIClass' has no member named 'transfer16'

   SPI.transfer16(*p);//should be fixed already

       ^

\Documents\Arduino\libraries\RA8875\RA8875.cpp: In member function 'uint16_t RA8875::getPixel(int16_t, int16_t)':

\Documents\Arduino\libraries\RA8875\RA8875.cpp:2291:16: error: 'class SPIClass' has no member named 'transfer16'

   color  = SPI.transfer16(0x0);

                ^

\Documents\Arduino\libraries\RA8875\RA8875.cpp: In member function 'void RA8875::getPixels(uint16_t*, uint32_t, int16_t, int16_t)':

\Documents\Arduino\libraries\RA8875\RA8875.cpp:2331:7: error: 'class SPIClass' has no member named 'transfer16'

   SPI.transfer16(0x0);//dummy

       ^

\Documents\Arduino\libraries\RA8875\RA8875.cpp:2338:17: error: 'class SPIClass' has no member named 'transfer16'

    color  = SPI.transfer16(0x0);

                 ^

\Documents\Arduino\libraries\RA8875\RA8875.cpp: In member function 'void RA8875::writeData16(uint16_t)':

\Documents\Arduino\libraries\RA8875\RA8875.cpp:3446:7: error: 'class SPIClass' has no member named 'transfer16'

   SPI.transfer16(data);//should be fixed already

       ^

\Documents\Arduino\libraries\RA8875\RA8875.cpp: In member function 'void RA8875::gPrint(uint16_t, uint16_t, const char*, uint16_t, uint8_t, const FONT_DEF*)':

\Documents\Arduino\libraries\RA8875\RA8875.cpp:3807:11: error: 'class SPIClass' has no member named 'transfer16'

       SPI.transfer16(buffer[i]);//should be fixed already

           ^

Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

sumotoy, 2015-05-04:
Ouch, the Arduino team added SPI.transfer16 for all processor but not Arduino DUE!
Try now, I think it's fixed

sumotoy, 2015-05-04:
Since I don't have DUE, can you tell me if it's working?
If yes, you can get more speedtrough (almost double), I will tell you how.

gcharles81, 2015-05-04:
HI Sumotoy

Its working fine In my case my setup is as follows 

Together with 5" TFT from buy.display.com  I have also Ethernet card with SD 

So I connected as following since pin 4 & 10 are already used by CS of SD and ethernet 
#define RA8875_INT 23 
#define RA8875_CS 22 
#define RA8875_RESET 24

Do you suggest that I should use the The 74HC125 driver/buffer to isolate the other devices on SPI bus.

So there is still possibility to go faster ? 

Thanks 
Charles

sumotoy, 2015-05-04:
You need the 74HC125, the RA8875 pulls up MOSI,MISO and SCLK and this will interfere with SD and other SPI devices.
I don't know if DUE has SPI transactions library, I think (and hope) yes. If so, go to RA8875UserSettings.h file in _utility folder and modify

	#elif defined(__SAM3X8E__)							 // due
		#define MAXSPISPEED 			8000000//was 6600000
	#else		

into

	#elif defined(__SAM3X8E__)							 // due
		#define MAXSPISPEED 			14000000//was 6600000
	#else

sumotoy, 2015-05-04:
And thanks for reply, I need feedback for getting the best from the library

gcharles81, 2015-05-04:
Tested but without the 74HC125 I have an issue on those speeds .

I will build the buffer circuit tomorrow and let you know the outcome.
I am willing to test the DUE part for you , do you have plan to support the other I2C capacitive touch controller GSL1680 ? 

thanks

sumotoy, 2015-05-04:
GSL1680  it's mounted on waveshare if I'm not wrong but I don't have it, I need a display with this chip to build upa library.
I can optimize a little the SPI code for DUE in the future, but it's important that SPI library for DUE support SPI transaction or it's a problem, RA8875 use SPI mode 3 (mosi,clock,miso HIGH) and SD use mode 0 (the opposite), the two chip uses different SPI speed also and eastrising has added pullups resistors on all lines, keeping live all together in harmony it's a challenge, some extra hardware and software tricks are needed.

closed 2015-05-21 by sumotoy

gcharles81, 2015-05-23:
Hi 

Today I made some speed benchmarks with DUE and 2 speed settings 8000000 / 12000000 , I also tested any differences between using CS pin 4 and CS Digital pin 22 , below are the results

CHIP Select  pin 4 - Speed 800000 In usersettings.h

screen:800x480 - rotation:0
Benchmark                Time (microseconds)
Screen fill              16944
Test Pixel               147
Test Pixels              97793
Text                     8825
Lines                    125832
Horiz/Vert Lines         99745
Rectangles (outline)     280385
Rectangles (filled)      280472
Circles (filled)         273986
Circles (outline)        270413
Triangles (outline)      33890
Triangles (filled)       75514
Rounded rects (outline)  257249
Rounded rects (filled)   1030068
--------------------------------

screen:480x800 - rotation:1
Benchmark                Time (microseconds)
Screen fill              16746
Test Pixel               147
Test Pixels              58689
Text                     8824
Lines                    125806
Horiz/Vert Lines         99766
Rectangles (outline)     280714
Rectangles (filled)      280900
Circles (filled)         274018
Circles (outline)        270466
Triangles (outline)      30879
Triangles (filled)       72750
Rounded rects (outline)  257209
Rounded rects (filled)   1030319
--------------------------------
CHIP Select  pin 4 - Speed 1200000 In usersettings.h

screen:800x480 - rotation:0
Benchmark                Time (microseconds)
Screen fill              16761
Test Pixel               137
Test Pixels              91387
Text                     8239
Lines                    122540
Horiz/Vert Lines         98725
Rectangles (outline)     279633
Rectangles (filled)      279518
Circles (filled)         264093
Circles (outline)        260736
Triangles (outline)      32843
Triangles (filled)       74747
Rounded rects (outline)  244755
Rounded rects (filled)   1018779
--------------------------------

screen:480x800 - rotation:1
Benchmark                Time (microseconds)
Screen fill              16710
Test Pixel               137
Test Pixels              54845
Text                     8224
Lines                    122516
Horiz/Vert Lines         98845
Rectangles (outline)     279452
Rectangles (filled)      279147
Circles (filled)         264150
Circles (outline)        260784
Triangles (outline)      29870
Triangles (filled)       71522
Rounded rects (outline)  244921
Rounded rects (filled)   1019665
--------------------------------
CHIP Select  pin 22 - Speed 800000 In usersettings.h

screen:800x480 - rotation:0
Benchmark                Time (microseconds)
Screen fill              16938
Test Pixel               146
Test Pixels              97031
Text                     8768
Lines                    125422
Horiz/Vert Lines         99005
Rectangles (outline)     280619
Rectangles (filled)      280349
Circles (filled)         271949
Circles (outline)        268424
Triangles (outline)      33808
Triangles (filled)       75382
Rounded rects (outline)  255819
Rounded rects (filled)   1028811
--------------------------------

screen:480x800 - rotation:1
Benchmark                Time (microseconds)
Screen fill              16747
Test Pixel               146
Test Pixels              58230
Text                     8768
Lines                    125430
Horiz/Vert Lines         99029
Rectangles (outline)     280710
Rectangles (filled)      280433
Circles (filled)         272023
Circles (outline)        268470
Triangles (outline)      30821
Triangles (filled)       72389
Rounded rects (outline)  255817
Rounded rects (filled)   1028698
--------------------------------
CHIP Select  pin 22 - Speed 1200000 In usersettings.h
screen:800x480 - rotation:0
Benchmark                Time (microseconds)
Screen fill              16741
Test Pixel               137
Test Pixels              90629
Text                     8195
Lines                    122179
Horiz/Vert Lines         98457
Rectangles (outline)     279003
Rectangles (filled)      279339
Circles (filled)         262080
Circles (outline)        258749
Triangles (outline)      32714
Triangles (filled)       74377
Rounded rects (outline)  243563
Rounded rects (filled)   1018885
--------------------------------

screen:480x800 - rotation:1
Benchmark                Time (microseconds)
Screen fill              16920
Test Pixel               137
Test Pixels              54388
Text                     8165
Lines                    122210
Horiz/Vert Lines         98475
Rectangles (outline)     279003
Rectangles (filled)      279414
Circles (filled)         262137
Circles (outline)        258794
Triangles (outline)      29710
Triangles (filled)       71146
Rounded rects (outline)  243722
Rounded rects (filled)   1017491
--------------------------------