sumotoy/RA8875 #43: BTE_Move doesn't work on layer 2 in portrait

MorganSandercock, 2015-05-08:
The updates to BTE_Move() have broken portrait mode for layer 2. The high bit 0x80000000 must be set on the Y coordinate to indicate Layer 2 but the updated BTE_Move() then applies the portrait X-Y swap in BTE_fromTo() after setting that bit. So it can never work on layer 2.
There was a reason why I didn't use the BTE helper functions when I wrote it. I also advise against using the BTE_sourceLayer() and BTE_destLayer() functions in the BTE_Move() because those functions will take more time than the move itself does.

sumotoy, 2015-05-08:
Ok, I will find a solution, the reason I used helper is that BTE have many methods so write hudge functions will result in massive code.
There's still a problem with this function with the 800x480 displays, simple move works but apply an efx result in garbage on screen, I've sent an email to RAiO and they are investigating, they assert it should work. Initialize the 800x480 screen at 8 bit get things worse, this is quite strange since I was thinking that error was in this area (there's a function that downgrade color dept as soon layer are used) but looks like there's something else.
I have a suspect that BTE using the first color byte only on 16 bit screen, I haven't a 272x480 actually (mine it's dead last week definetively, I have another arriving from china), can you try the monochrome bte demo at 8 bit on that screen?

MorganSandercock, 2015-05-08:
I'm not sure if there's something wrong in the way you're using 8-bit mode. drawPixel() still sends 16 bits of color information and the display ignores half of it. Is there a way to only send 8 bits?

For the simple 8-wide symbols, it works in 8-bpp mode by simply writing the 8 bits in the upper half of the word. For example:
tft.drawPixel(i*2, 0, battery1[i]<<8);
(from my example, using your 8-bit wide battery symbol.) This draws fine on the 400x272 screen, using the same ROP as 16-bit mode.

But the pixels in memory are still 16 bits wide and I can't work out how to write to the other half of the word using the provided functions. So no monochrome symbol can be more than 8 units wide at the moment.

MorganSandercock, 2015-05-08:
It is looking like a bug in the RA8875: when testing my 32-bit wide character, two different bytes in memory affect the same place on the screen. I'm still investigating.

sumotoy, 2015-05-31:
Looks fixed now...

closed 2015-05-31 by sumotoy