sumotoy/RA8875 #45: Eastrising 5" 800x480 TFT RA8875 not responding

ReneMeeuw, 2015-05-21:
Continued from  https://forum.pjrc.com/threads/28655-Eastrising-5-quot-TFT-RA8875-with-sumotoy-lib-not-responding

I'm trying to get an Eastrising RA8875 5" 800x480 TFT working with the sumotoy lib, but still at most I get the backlight on and nothing more.
I've tried everything suggested already:
- 2 different Eastrising 5" 800x480 TFT's, with and without R1/2/3 and C1/2
- using 5V power from USB-3 for both Teensy and TFT (J8 open)
- Used 2 different Teensy 3.1's (tested with 48 and 72MHz)
- Used Sumotoy RA8875 lib <B>0.69b36</B> and it's examples modified to use pin14 for SCLK (also earlier)
- Using direct connection Teensy/TFT and also via 74HC125
See pictures (cables are about 3" long) and sourcecode below.

<b>Note to sumotoy:</b> My setup is 800x480 Eastrising direct connected to teensy 3.1 and powered at 5v. Measured exact 5.0V on TFT and J8 is open for 5V. See large foto's below for exact connection and jumper settings.

<b>WHAT am I doing wrong???</b>

[image]
[image]

#include <SPI.h>
#include <RA8875.h>

#define RA8875_CS 10              //see below...
#define RA8875_RESET 255     //any pin or nothing!
#define RA8875_MOSI 11
#define RA8875_SCLK 14
#define RA8875_MISO 12

RA8875 tft = RA8875(RA8875_CS, RA8875_RESET, RA8875_MOSI, RA8875_SCLK, RA8875_MISO);

void setup() {
  Serial.begin(38400);
  long unsigned debug_start = millis ();
  while (!Serial && ((millis () - debug_start) <= 5000)) ;
  Serial.println("start");
  tft.begin(RA8875_800x480);            //initialize library
  if (tft.errorCode() != 0) {
    Serial.println("Initializing error!\n");
    uint8_t error = tft.errorCode();
    if (bitRead(error, 0)) {
      Serial.println("display not recognized!");
    } else if (bitRead(error, 1)) {
      Serial.println("MOSI or MISO or SCLK out of permitted range!");
    } else if (bitRead(error, 2)) {
      Serial.println("CS out of permitted range!");
    }
  }
  Serial.println("RA8875 initialized!");
  tft.print("Hello World");
}

void loop() { }

sumotoy, 2015-05-21:
Looks like this one needs a different initialization. I have a 800x480 but it's  a 7", you have a 5".
All wiring looks correct to me. let me try to check the init code, I will give you new values in a min.

sumotoy, 2015-05-21:
I have modified the library to get a brand new initialization index **RA8875_800x480_5**
It probably don't work yet, I'm trying to figure out the correct parameters for your specific TFT.
A question: did you try the Adafruit_RA8875 to see if TFT initialize?

ReneMeeuw, 2015-05-21:
I just tried to initialize with "Adafruit_800x480". Via the serial monitor I also get the "RA8875 initialized!", but nothing else happens. I also tried "RA8875_480x272" with the exact same result.

ReneMeeuw, 2015-05-21:
If you want to I can send one of my TFT's to you (I have 2) so you can test it yourself. Just let me know where to send it to and I'll mail it right away.

sumotoy, 2015-05-21:
where are you living?

sumotoy, 2015-05-21:
I have upped a new test, let me know if it's working.
The really interesting thing is that at this link http://www.buydisplay.com/default/5-inch-tft-lcd-module-800x480-display-w-controller-i2c-serial-spi I found an software example and just discovered it's my library, the 0.55 early beta!

ReneMeeuw, 2015-05-22:
Hi Max,

 

I live in the Netherlands (Gerben van Manenstraat 110 in Drachten). I can send you my second 5” 800x480 TFT and you can keep it as a thanks for your efforts on the library, you don’t have to send me another one. I’ll just send it in a thick envelope, so it should be in Italy in a few days.

 

I just tested your latest lib with the RA8875_800x480_5, but still nothing happens.

 

Best regards, René Meeuwisse.

 

Dit e-mailbericht is gecontroleerd op virussen met Avast antivirussoftware.
http://www.avast.com

sumotoy, 2015-05-22:
thanks but I prefere send you one in any way. Netherland will take few days, normally from 3 to 7 to here.
Send an email here: sumotoy at gmail.com

sumotoy, 2015-05-22:
Witch type of touch controller it's mounted?

ReneMeeuw, 2015-05-22:
It's a resistive touch panel. I didn't connect the interrupt pin for that yet (see first photo), because I first want the TFT to display something. I'll send my address to your gmail and you can reply with yours.

sumotoy, 2015-05-22:
Hi Renè,
be sure it's well packet, the post here are like in zambia!
My address:
Egidio Massimo Costa
Via Giordano Bruno 3, 20154 Milano, Italy.
Tel. +39-02-36503770

Give me your address too!

Before send, just a couple of questions:
1) Are you sure that you never set display in the past at 3V3 and supply 
5V (for error I mean)?
2) Actually the display react in some way (the backlight turns on and 
you get black screen).
Max

sumotoy, 2015-05-22:
Font Chip it's mainly for chinese,arabic and asian languages, I already managed how to deal with it in library but I use rarely.
I'm actually enabling the keypad that can be really useful and adding support for big custom fonts.
Also I'm adding text support for automatic centered text inside objects (really useful for design interfaces) and gauges, the goal it's create complex interfaces really easily

ReneMeeuw, 2015-05-22:
Sounds good, I really appreciate your efforts!
I've just mailed my 5" 800x600 TFT to you. Should arrive in 2..4 working days.

ReneMeeuw, 2015-05-22:
I checked the TFT Datasheets (ER-TFTM050-3) and I think that /RESET is on pin 11 of JP3 (although it mentions (NC), so maybe not connected). On the eastrising connections.png it is shown on pin 12 (the datasheet says this is used to select 8080/6800 interface and internally connected to ground). Maybe if reset is needed and we connected it on the wrong pin this may explain why the TFT is not working yet. What do you think?

sumotoy, 2015-05-22:
OOpps, my fault! I will correct png asap.
Btw I have tested with and without RST and no changes, the library use a software reset when RST not connected.
Atrough if you set the instance to use RST pin and it's not connected the RA chip it's not resetted correctly...

gcharles81, 2015-05-22:
Hi 

I am using the same exact display with arduino DUE and this library and works fine what I have done is the below 

Connecting Pin 10,11 from tft CON2 to arduino DUE and digital PIN and adding the below in define define  
#define RA8875_INT 23 
#define RA8875_RESET 24

Maybe this can help you

ReneMeeuw, 2015-05-22:
Thanks, I'll try right now...

sumotoy, 2015-05-22:
crossing fingers....

sumotoy, 2015-05-22:
Hey gcharles, what about to to use DUE extended SPI ? I have just ordered DUE

gcharles81, 2015-05-22:
I can try it but in my case I have to move tft cs to pin 52 because 4 and
10 are already occupied by sd and Ethernet.

Let me know until your due will arrive if you want me to test anything for
you .

ReneMeeuw, 2015-05-22:
I tested with /RESET on TFT pin 11. Tested with new Teensy, new ER-TFTM050-3 with all jumpers carefully set and removed R1/2/3 C1/2, all doublechecked. Tried begin(RA8875_800x480_5/RA8875_800x480/RA8875_480x272/Adafruit_800x480) just to be sure, tested on 72 and 48MHz. But nothing worked.
I noticed 2 things: 
- With reset pin set to 255 backlight usually did not switch on, with 8 it did, so Reset helped somewhat.
- tft.errorCode() always returned 0,ALSO when using a wrong tft.begin parameter.

sumotoy, 2015-06-01:
Display arrived now. I have checked carefully and discover one thing:
1) This display doesn't have the 5V to 3v3 regulator (looking on the back the LD1117 is missed) so leave jumper 8 open it will simply break the supply.<br>
This mean that **TFTM050-3 works ONLY at 3V3!!!**<br>
2) This display needs a separate 3v3 regulator, the Teensy 3.1 internal one it's not able to supply enough current and goes in protection when backlight led goes on.<br>

Fixed the 2 issues, I connected as normally, instanced with RA8875_800X600 and...<br>
It turn's on and work perfectly, if you need a photo I can give you.

ReneMeeuw, 2015-06-01:
Thanks sumotoy! I see what you mean, U2 is empty opposed to holding the LD1117 in the pictures of the other displays in you documentation. I'll send a note to Eastrising about this, hoping the'll fix this in the future.

Did you just solder an LD1117 to the back or did you supply 3v3 externally?

sumotoy, 2015-06-01:
You cannot solder an LD since there's no space for the output condenser.
Besta way it's supply it from external and NOT from teensy, at 3v3 it drain too much current.

closed 2015-06-01 by sumotoy