sumotoy/RA8875 #46: When using SD tft.fillRoundRect & tft.fillRoundRect & tft.fillCircle do not works properly on DUE

gcharles81, 2015-05-25:
HI 

Today I noticed that if I have an SD card inserted in my ethernet shield and  after I initialize the card the  tft.fillRoundRect & tft.fillRoundRect & tft.fillCircle somehow are drawn in wrong coordinates if I use Version:0.69b38 while it works perfect with Version:0.55(early beta) note i downloaded this from buydisplay.com website.

my display is 5" 800x640 buydisplay.com and I am using the 74HC125 buffer circuit v3 as per drawing in documentation

sumotoy, 2015-05-25:
It's because 0.69b38 use faster SPI speed than 0.55. I've managed to fix this in the upcoming 0.69b50.
circleHelper cannot work at more than 10Mhz, in the new code I've introduced a pach that slow down at 10Mhz some operation and brings up automatically at 14Mhz. If you want a temp fix open RA8875UserSettings and around #if defined(__SAM3X8E__)// DUE set
#define SPI_SPEED_WRITE 	SPI_CLOCK_DIV4 to #define SPI_SPEED_WRITE 	SPI_CLOCK_DIV6

sumotoy, 2015-05-27:
Try new beta...

gcharles81, 2015-05-30:
I am using the new version now since Thursday and it's fine so much faster.

Have one question about text center function - can if it be used to place
text in center of 4 coordinates instead of screen center.?

Thanks
Charles

sumotoy, 2015-05-30:
I have a very useful function for center text around x,y:<br>
tft.setCursor(x,y,autocenter);<br>
For example tft.setCursor(80,80,true);<br>
If you now use tft.print("ABCDEF"); -> the text will be centered to 80,80 like this ABC|DEF<br>
Whitouth autocenter flg text will start from 80,80. -> |ABCDEF<br>
Did you confirm it works with DUE? I have another version even faster with more features almost ready.

gcharles81, 2015-05-30:
Thanks will try the center function  today .

Yes of course it's working fine with DUE I am using it with 5" tft from
buydisplay.com

Regards
Charles

sumotoy, 2015-05-30:
0.69b60 online, can you try with DUE? (make a backup of the 0.69b50 first).
I don't have DUE so I'm doing blind programming for this CPU!

sumotoy, 2015-05-31:
0.69b62, this is better.

sumotoy, 2015-05-31:
If it's working I will close this, I'm not able to test with DUE.

gcharles81, 2015-05-31:
Hi Nothing is working anymore with the last 2 versions on DUE also the examples tft seems is not initialized

sumotoy, 2015-05-31:
Ok, I will fix tonight, I know where's the problem, really thanks for beta test!
I just tested all the rest of the boards and everithing works, te only one it's DUE that I don't have but I can get trough.

gcharles81, 2015-05-31:
Wanted to give you some more info on version b50 using my own program it's working fine .

but wanted to let you know that examples also do not work example I tested benchmark  test the draw functions makes screen full of garbage and what I have noticed is only fillscreen works good.

sumotoy, 2015-05-31:
oh, thanks, this info it's very important

sumotoy, 2015-05-31:
I got an Arduino DUE with programming port with solder problems (brand new from Arduino site, Grrr), it's a pain to program because it works once every 20 but at the end I solved...
Now DUE works in normal SPI, Extended SPI and Normal SPI with CS port manipulation (faster), you can choose one of the method from RA8875UserSettings as you like.
I've runned every demo all run smoothly without any artifacts. I'm shooting in a min a video proof and post in Wiki in some minutes along with benchmark results.
If you got any garbage please check if the other SPI devices on the same line are using SPI Transactions, you have to enable for all of them otherwise the SPI speed modulation I use for fix the shit going on inside RA8875 brain will conflict and everithing broken up, if you need help about add (or in most case enable) SPI transaction please open another issue here.
I will up the library in 1 hour, just want to check if there are other stuff I missed, the modifications are a lot but now library can work in any possible mode of every CPU so very flexible.

gcharles81, 2015-06-01:
Thanks Sumotuy 

I will check the point you mentioned about the other devices because I cannot use SD anymore with latest version but i have to check how I can see if my SD library is using the SPI transactions as you mentioned 
In my project I am using both original SD and ETHERNET libraries

sumotoy, 2015-06-01:
Do not worry, there's a solution for everithing...
1) The Ethernet board works at very slow SPI speed, it's really simple to add SPI transaction but I need to know which chip you are using.<br>
2) SD card was nightmare because SD cards have weak SPI lines, each brand has different characteristics and working at fast SPI speed as know mean choose very carefully SD holder, isolate the RA8875 as I show in wiki. and have short wires. Of course you can work at slow SPI speed with the SD card and get rid of the long wires and SD holder but if you want high speed access you better pay attention to this. Some SD library have SPI Transaction, the one from Bill Greyman have (you must enable in settings).<br>
The W5100 ethernet chip have also an error bug that need an external chip, I can show you the circuit.<br>
Oh well, why before was working everithing? Because at every SPI call the device update his access speed and use it, if the W5100 that works at 500-1Mhz(sigh) make a transaction the SD and display will retain that speed until there's another speed change, this is not good!<br>
With the SPI transaction every device works as it's best so TFT works at 22Mhz, W5100 at 1Mz and SD can works really fast if it's HC10 (even 24Mhz), you will notice overall speedup and no garbage on screen or whatever, SPI transaction are new thing but are here to stay, in the future everithing will use them.

gcharles81, 2015-06-04:
Thanks sumotoy

For Ethernet I am using w5100 on the Ethernet shield .

Can you let me know where I can find a circuit diagram for the w5100 fix , or else I was thinking of buying a new Ethernet module based on w5200 

As for the sd I will try the library you suggested and let you know 

Thanks Charles

sumotoy, 2015-06-04:
Did you isolating the RA8875 with the 74HC125 as show in wiki?
Also, tell me witch library you are using for W5100, where I can download?

gcharles81, 2015-06-04:
Yes i  am using 74HC125 for RA8875 v3 of your schematic.

As for the Ethernet & SD I am using the ethernet shield and  the library  Tinywebserver but modified to work for DUE and it uses the official SD and ETHERNET libraries.

The original Tinywebserver is not compatabile with DUE . 

Do you want me to send you a copy of the library ? ,

BTW the last two days I was trying to make the  capacitive touch screen of my 5" RA8875 tft from buydisplay.com using GSL1680 ic working on DUE and finally I managed to make it run.

sumotoy, 2015-06-05:
I have ordered a display with that touch controller, capacitive touch are great, the FD don't need calibration and simplify a lot the code.
About the Ethernet lib, I can take a look to add SPI transaction if you provide a link.
In the meantime try add in the setup this 2 lines:
  pinMode(CS_W5100, INPUT_PULLUP);
  pinMode(CS_SD, INPUT_PULLUP);
Or maybe better a 10K resistor trough W5100 CS pin and VCC and SD_cs and VCC.
The W5100 CS pin it's floating when not inited, the chip can catch interferences during initialization and not startup correctly when inited.

closed 2015-06-20 by sumotoy

gcharles81, 2015-06-29:
Thanks Max

It took me some time to reply to you because I decided to scrap the
official ethernet , and designed my own SD , and ethernet shield using
Wiz820io module and only sd socket direct  with incorporated 74HC125 for
the tft , now I can already run with Full speed on the sd together with
 TFT using your library uning a breadbord version and now I am just waiting
for the designed PCB from fab.

Regards