sumotoy/RA8875 #51: Touch Screen Calibration Fails to complete.

sbfreddie, 2015-06-20:
While try to use the supplied touch screen calibration example, the first step appears to work OK (Top Left Corner), but while waiting for the Bottom Right Corner to be touched, nothing happens after the red dot is touched no matter how hard you press.
It just hangs there waiting for me to press the red dot.

Any possible solutions would be much appreciated.

Thanks,
Ed

sumotoy, 2015-06-21:
Can you tell me witch display you are using and controller?

sbfreddie, 2015-06-21:
Max:
I am using the ER-TFTM043-3 display which has the RA8875 controller, and the Teensy 3.1 Microprocessor (using the alternate SPI pins).
So far all the other examples work just great!
Thanks,
Ed

sumotoy, 2015-06-21:
Looks like there's a wrong or out of range matrix resistor on board, it happened to another user 2 month ago. In that case I need to modify a bit the calibration sketch.
I planned a 0.70b3 update between today and tomorrow with 3 more features and a workaround for this.

sbfreddie, 2015-06-21:
Max:
Thanks for all you do and have done!

Ed

sumotoy, 2015-06-22:
Haved bad time with another hardware bug on RA8875 that affect fillTriangle so I've loosed hours to find out a solution, tomorrow at list I will release 0.70b3

sbfreddie, 2015-06-22:
Max:
Have you fixed the touch screen calibration yet?

Thanks,
Ed

sumotoy, 2015-06-23:
the good news, your display seems ok. There's something broken in the library when I added screen rotation so I'm currently fixing, it will not take long.
Sorry for delay but I have to get a resistive touch display from my old project.
Today or tonight I release the fixed library.

sumotoy, 2015-06-23:
Almost solved, after 5 hours of storage oscilloscope and tons of tweaks I'm able to rearm RA interrupt even at 22Mhz.
But timings are critical so I'm forced to completely recode all the Touch Screen section of the library to use hardware interrupt or on fast processors the RA can get stuck again.
Have to take a break, the most it's done. Maybe I'm not able to release today but for sure tomorrow it's ready.
Thanks again to point me to this, sorry but this chip it's a register nightmare and needs a lot of work

sumotoy, 2015-06-24:
sbfreddie,
I'm still working on beta3, lot of changes.
Btw if you want a temporary fix, try following changes and let me know:
Open RA8875.cpp
find this function:

void RA8875::touchEnable(boolean enabled) 
change to:
```
void RA8875::touchEnable(boolean enabled) 
{
	if (_touchPin < 255){
          if (_currentMode) changeMode(false);
		if (!_touchEnabled && enabled) {
			_INTC1Reg |= (1 << 2); //bit set 2
			writeReg(RA8875_INTC1, _INTC1Reg);
			_TPCR0Reg |= (1 << 7); //bit set 7
			writeReg(RA8875_TPCR0, _TPCR0Reg);
			_touchEnabled = true;
		} else if (_touchEnabled && !enabled) {
			_INTC1Reg &= ~(1 << 2); //clear bit 2
			writeReg(RA8875_INTC1, _INTC1Reg);
			_TPCR0Reg &= ~(1 << 7); //clear bit 7
			writeReg(RA8875_TPCR0, _TPCR0Reg);
			_touchEnabled = false;
		}
	}
}
```

it's a temporary workaround but should work, I'm working at 0.70b3 that have tons of changes.

sbfreddie, 2015-06-24:
Max:
Thanks, that fixed the calibration problem on my display!

Ed

sumotoy, 2015-06-25:
looks an easy fix but took 2 days of desperation since it's an obscure bug of RA8875, the touch registers of touch screen cannot be touched if text mode it's selected, there's no word of that in the datasheet.
I'm pretty sure that this chip it's korean but designed by a moron.
Btw this is a temporary fix, I'm releasing the 0.70b3 that has definitive fix, much more features but unfortunatly a lot of changes and several functions changed, expecially touch screen one, I plan to release once I checked capacitive touch and fix another bug ob SD.

sumotoy, 2015-06-25:
Another fast fix
find this function

void RA8875::begin(const enum RA8875sizes s,uint8_t colors) 

inside, find the code

```	

#if !defined(USE_EXTERNALTOUCH)
		_touchPin = 255;
		_clearTInt = false;
		_touchEnabled = false;
		if (!touchCalibrated()) {//added by MorganSandercock
			_tsAdcMinX = 0; _tsAdcMinY = 0; _tsAdcMaxX = 1023; _tsAdcMaxY = 1023;
		} else {
			//We have a valid calibration in _utilities\RA8875Calibration.h
			//Note that low may be a smaller value than high
			//These values may be updated by any rotation/orientation setter.
			_tsAdcMinX = TOUCSRCAL_XLOW;
			_tsAdcMinY = TOUCSRCAL_YLOW;
			_tsAdcMaxX = TOUCSRCAL_XHIGH;
			_tsAdcMaxY = TOUCSRCAL_YHIGH;
		}
	#else
/*		Touch Panel Control Register 0     [0x70]
		7: 0(disable, 1:(enable)
		6,5,4:TP Sample Time Adjusting (000...111)
		3:Touch Panel Wakeup Enable 0(disable),1(enable)
		2,1,0:ADC Clock Setting (000...111) set fixed to 010: (System CLK) / 4, 10Mhz Max! */
		_TPCR0Reg = RA8875_TPCR0_WAIT_4096CLK | RA8875_TPCR0_WAKEDISABLE | RA8875_TPCR0_ADCCLK_DIV4;
		//_TPCR0Reg = RA8875_TPCR0_WAIT_32768CLK | RA8875_TPCR0_WAKEDISABLE | RA8875_TPCR0_ADCCLK_DIV128;
	#endif
```	

into

```	
_activeWindowXL = 0;
	_activeWindowXR = WIDTH;
	_activeWindowYT = 0;
	_activeWindowYB = HEIGHT;
	#if !defined(USE_EXTERNALTOUCH)
		_touchPin = 255;
		_clearTInt = false;
		_touchEnabled = false;
		if (!touchCalibrated()) {//added by MorganSandercock
			_tsAdcMinX = 0; _tsAdcMinY = 0; _tsAdcMaxX = 1023; _tsAdcMaxY = 1023;
		} else {
			//We have a valid calibration in _utilities\RA8875Calibration.h
			//Note that low may be a smaller value than high
			//These values may be updated by any rotation/orientation setter.
			_tsAdcMinX = TOUCSRCAL_XLOW;
			_tsAdcMinY = TOUCSRCAL_YLOW;
			_tsAdcMaxX = TOUCSRCAL_XHIGH;
			_tsAdcMaxY = TOUCSRCAL_YHIGH;
		}
/*		Touch Panel Control Register 0     [0x70]
		7: 0(disable, 1:(enable)
		6,5,4:TP Sample Time Adjusting (000...111)
		3:Touch Panel Wakeup Enable 0(disable),1(enable)
		2,1,0:ADC Clock Setting (000...111) set fixed to 010: (System CLK) / 4, 10Mhz Max! */
		//_TPCR0Reg = RA8875_TPCR0_WAIT_4096CLK | RA8875_TPCR0_WAKEDISABLE | RA8875_TPCR0_ADCCLK_DIV8;
		_TPCR0Reg = RA8875_TPCR0_WAIT_32768CLK | RA8875_TPCR0_WAKEDISABLE | RA8875_TPCR0_ADCCLK_DIV128;
	#endif
```

sbfreddie, 2015-06-25:
Max:
I installed this change and ran the calibration again, and it works fine still.  I got different numbers this time:
XLOW 75
YLOW 161
XHIGH 927
YHIGH 912

Thanks,
Ed
PS I’ll wait until Sunday and then install your new version.

sumotoy, 2015-06-26:
these numbers looks fine to me!

sumotoy, 2015-06-29:
I'm late with the release, have to fix some other bug and finishing the render font engine.
Btw it's work with the proposed mods?

sbfreddie, 2015-06-29:
Max:
So far I have not seen any problems with your changes, however I am not currently using the touch functions in my code, but will be in the not to distant future.
One strange problem is with the getCursor(x,y) function:
Code:
if(CmdlineBufferEditPos == CmdlineBufferLength)
                                {
                                    // destructive backspace
                                    // echo backspace-space-backspace
                    
                                    Serial.print(F("\b"));
                                    Serial.print(F(" "));
                                    Serial.print(F("\b"));
                                    tft.getCursor(cursorPositionX, cursorPositionY);
                                    cursorPositionX--;  // Decrement X position (Backspace).
                                    tft.setCursor(cursorPositionX, cursorPositionY);  //set cursor position to write text.
                                    tft.setTextColor(RA8875_WHITE, RA8875_BLACK);  // First variable sets the text color, second variable sets the background color.
                                    tft.print(F(" "));  // echo space character to the TFT Screen.
                                    tft.getCursor(cursorPositionX, cursorPositionY);
                                    cursorPositionX--;  // Decrement X position (Backspace).
                                    tft.setCursor(cursorPositionX, cursorPositionY);  //set cursor position to write text.
                                    // decrement our buffer length and edit position
                                    CmdlineBufferLength--;
                                    CmdlineBufferEditPos--;
                                }

In this top section of code the compiler has no problem finding the getCursor(cursorPositionX, cursorPositionY) function as indicated by the color of the code.  (I am using Xcode)
However in the bottom section of code the exact same code cannot be found by the compiler, if issues an error “No matching function for call to ‘RA8875::getCursor(int&,int&}’ “.
I can’t seem to get past this problem, even though I just copied and pasted the code from the top section to the bottom section.

    //digitalClockDisplay();  // This puts the time on the Computer monitor.
    tft.getCursor(cursorPositionX, cursorPositionY);  // Save cursor position.
    tft.setTextColor(RA8875_WHITE, RA8875_BLACK);  // First variable sets the text color, second variable sets the background color.
    tft.setCursor(XCOL47,YLINE1);   //Move cursor to column 17, row 35 out of 60 for time & date display.
    tftTime();  // This puts up the time on the TFT display.
    tft.setCursor(cursorPositionX, cursorPositionY);  //set cursor position to original position.

Do you have any suggestions?
Thanks,
Ed

sumotoy, 2015-06-29:
because you have to use this:
 tft.getCursor(&cursorPositionX, &cursorPositionY);
getCursor uses the & operator to get the value, so using the function above will transfer the cursorx,y values to cursorPositionX and cursorPositionY.

sbfreddie, 2015-06-29:
Max:
That does not work, it gives the same error, then I have two errors not one.
I figured out the problem, it seems that getCursor must have the type “int16_t", not type int.  I guess the compiler is very fussy about the correct type being passed to functions. 

Thanks,
Ed

sumotoy, 2015-06-29:
Yes, getCursor  must have <b>int16_t</b>, it's the basic value of all coordinates. 
Since the compiler does some optimizations you will not see any differences in therms of memory between int and int16_t.

closed 2015-06-30 by sumotoy

github4mat, 2018-03-05:
I have the same problem with the touch screen calibration as "sbfreddie" had years ago.  The problem I found is, that first time the tft.touched() routine after I touched my screen delivers true-value for return  but then it stays false! The reason may be that _RA8875_INTS is not being cleared and so the statement "if (bitRead(_RA8875_INTS,0))" wont deliver true after first time passed!
Does this statement not clear _RA8875_INTS &= ~(1<<0); _checkInterrupt(2); the variables correctly?
I dont know how to check this?

My hardware:
Teensy 3.2
5' Adafruit 800X480 TFT Screen

#define USE_RA8875_TOUCH //resistive touch screen is being set!
Interrupt pin ist pin 0;
Version of RA8875: 0.70b11p10
Can someone help me?