sumotoy/RA8875 #54: SD Tests at 800x480 not working correctly after 0.69 beta 50

The-Experimentalist, 2015-06-24:
I do not have any versions in between 069 beta 50 and 0.69 beta 64

I am using a Teensy 3.1 and have been testing with different library, Arduino and Teensyduino versions

I am still looking for a solution where my project still only works on version 0.68 and no later versions. It does work perfectly on 0.68 but I obviously do not get the benefit of later improvements.

My testing however relates purely to the included examples. I have compared the following examples with many combinations of library, Arduino and Teensyduino:
RA8875\Examples\_Teensy3\_800x480_only\SDfatTest_800x480
RA8875\Examples\_Teensy3\_800x480_only\SDTest_800x480
RA8875\Examples\_Teensy3\_800x480_only\manyGauges_800x480
RA8875\Examples\_Teensy3\_800x480_only\vbarsGauge800x480

I have included 3 images, a failure matrix and the good and bad results.

I am using a test rig but not on breadboard I am using a matrix PCB, similar to that shown in your early documentation. I am using the Buy Display 7" 800x480 display ER-TFTM070-5 using your V3 isolation circuit and a 74HC125. Therefore the RA8875 is totally isolated.

The display has R1, R2, R3, C1 and C2 removed.

The only changes to the examples are to convert to hardware serial as I find this much better to work with.

I have the following configuration changes to the SdFat library (SdFatConfig.h)

-#define USE_LONG_FILE_NAMES 1
+#define USE_LONG_FILE_NAMES 0

-#define ARDUINO_FILE_USES_STREAM 1
+#define ARDUINO_FILE_USES_STREAM 0

-#define ENABLE_SPI_TRANSACTION 0
+#define ENABLE_SPI_TRANSACTION 1
[image]
[image]
[image]

I would appreciate any pointers you may have to anything stupid I am doing or any other feedback you can offer.

Thanks for a all your efforts and a great library!

The-Experimentalist, 2015-06-24:
With regard to the SD test columns the Y = Good (top image), N = BAD (bottom image)

sumotoy, 2015-06-24:
Ah, this should quite simple. As you noticed the image is offset in the y axis and get worst when the buffer has reloaded, this probably because for SD I need to code a function that define first the area that should be filled, set the X,Y, fill a one line buffer and write the line buffer, then repeat.
Actually it uses a generic buffer, not good because the y will goes wrong.
The old version was working but worst because they use pixel write, extremely slow.
I can fix this in following days for sure.
I was really busy because of another stupid bug of the RA8875 that affect resistive touch screen, solved definetively some minute ago after 2 days of desperation, it will part of the incoming 0.70b3 with a lot of new stuff like a new function to create round gauges, much easier, faster and bug free.

sumotoy, 2015-07-03:
Ok, I fixed this issue, I'm able to load the image correctly now and I will release a 0.70b5 preview between today and tomorrow.
Unfortunatly I'm very un-luky with SD cards, I have 10 and all create problems, this together the long cables and breadboard force me to use QUARTER SPEED on SDFat.

sumotoy, 2015-07-05:
0.70b5 preview online. Have tested with Paul's optimized SD and SD Fat and both working at high speed (only when using short cables, a decent SD card and the SD card holder provided by Paul, the same mounted on Audio Board.
Tested with 800x480 and 480x272 and Teensy3.1 and both working, no excuses.
If something wrong please check connections, cable lenght, sd card, code, etc.
The modified SD examples are in Teensy examples folder, just added one line and the SDfat800x480 uses a modified version that doesn't need buffer.

The-Experimentalist, 2015-07-08:
Working fine now:

OK!

Loading image 'keya.bmp'
File size: 657654
Image Offset: 54
Header size: 40
Bit Depth: 24
Image size: 800x274
Loaded in 2520 ms
end...

sumotoy, 2015-07-08:
should be much faster now, correct?

The-Experimentalist, 2015-07-14:
Sorry been away working and going away again so trying to answer your questions

0.69b50:

Loading image 'keya.bmp'
File size: 657654
Image Offset: 54
Header size: 40
Bit Depth: 24
Image size: 800x274
Loaded in 2853 ms
end...

0.70b5:

Loading image 'keya.bmp'
File size: 657654
Image Offset: 54
Header size: 40
Bit Depth: 24
Image size: 800x274
Loaded in 2500 ms
end...

353ms faster, shaving 12% of the original time, hope that helps

sumotoy, 2015-07-15:
cool, not a big improve but I will refine this, the SD library from Paul Stoffregen with t3 optimizings enabled it's actually faster but works only when card it's really close to Teensy

closed 2015-07-15 by sumotoy