sumotoy/RA8875 #55: Brightness Settings

The-Experimentalist, 2015-06-26:
When I use the Adafruit library and set the brightness it seems to work, very minor stroboscopic effects. When I use your library I get a very pronounced stroboscopic effect.

I have a sketch that sets the background up as follows on each library:

	tft.clearScreen(0xB5B2);
	brightness = 226;
	tft.brightness(brightness);

Can you try it and see what you think does it, same setup as previous issue.

sumotoy, 2015-06-27:
Commands and register affected are the same, the only difference it's the system speed that probably affect the RA8875 pwm generator. I will look the datasheet to see if there's a relation with that.

sumotoy, 2015-06-27:
I just test with my last beta:
tft.brightness(5); //dimmed, no flicker
tft.brightness(255); //full on, no flicker
There's no evidence of of stroboscopic efx or flickering!

sumotoy, 2015-06-28:
Tried

tft.clearScreen(0xB5B2);
tft.brightness(226);

All test on Eastrising 5 and 7, no issue.
The only thing I can tell it's when switch libraries you need a complete power cycle, my library and the adafruit are too different and the RA8875 it not always reset all it's parameters as it should.

The-Experimentalist, 2015-06-28:
Well I stand confused. There is obviously nothing wrong with my screen as when using the Adafruit library it works fine. 

I have just tried this with the full colour palette as defined in RA8875ColorPresets.h

All the pure colours seem fine regardless which library I use.

The ones causing the strobe effect are:
RA8875_LIGHT_GREY
RA8875_LIGHT_ORANGE
RA8875_DARK_ORANGE
RA8875_PURPLE

However they all work fine with the Adafruit library.

Oh well, thanks for looking, I am all out of ideas.

The-Experimentalist, 2015-06-28:
If I change the PWM setup code to use RA8875_PWM_CLK_DIV128 as follows the strobe effect goes but the display is audibly louder, you can hear it working (high frequency whistle, quite annoying). Tested it then put it back since not sure what effect it may have on the hardware.

void RA8875::backlight(boolean on) //0.69b31 (fixed an issue with adafruit backlight)
{
	if (_size == Adafruit_480x272 || _size == Adafruit_800x480/* || _size == Adafruit_640x480*/) GPIOX(on);
	if (on == true){
		//setup PWM ch 1 for backlight
		PWMsetup(1,true, RA8875_PWM_CLK_DIV128);
		PWMout(1,_brightness);//turn on PWM1
	} else {
		PWMsetup(1,false, RA8875_PWM_CLK_DIV1024);//setup PWM ch 1 for backlight
	}
}

sumotoy, 2015-06-28:
1) Did you tried to power cycle CPU and display when switch to this library?
2) Adafruit BOARD use PWM2 for display, all other display use PWM1. Wich display are you using (model, brand)?

Apart SPI differences there's a lot of other parameters set differently between the 2 library, first I'm using a much higher system clock and a lot of parameters are related to this.

This chip it's really nasty, one wrong parameter and you will have a blank screen and it have tons of parameters, some obscure like fade blanking, PCLK inversion, PCLK Period, all the Horizontal/Vertical non-display period registers, 
The register related to brightness could be P1CR, the brightness change the dutyCycle by touching register P1DCR but P1CR set the frequency. Try look at the <b>void RA8875::backlight(boolean on)</b> function, I'm actually using a  1024 division but maybe a different frequency it's better for your display.
The 2 display I have here works fine, I have carefully tweaked all parameters, maybe your display have a different TFT that need a tweak of one parameter or 2 but I need more info since I cannot reproduce the problem with the boards I have here.

sumotoy, 2015-06-28:
You are modding the frequency of the PWM, it was /1024 and drop to /128 result too high, try using different parameters but not going so high or you can damage the mosfet that drive the backlight

sumotoy, 2015-06-28:
What I suspect is that PWM have nothing to do with your problem since you should have flickering with all colors.
Can be something related to  fade blanking, PCLK inversion, PCLK Period, ,Horizontal/Vertical non-display period registers

The-Experimentalist, 2015-06-28:
I am using a test rig but not on breadboard I am using a matrix PCB, similar to that shown in your early documentation. I am using the Buy Display 7" 800x480 display ER-TFTM070-5 using your V3 isolation circuit and a 74HC125. Therefore the RA8875 is totally isolated.

The Adafruit library I am using for the comparison tests is using PWM1:
https://github.com/PaulStoffregen/Adafruit_RA8875.git

It makes no difference if I power cycle the display or not. I can switch the divider from 1024 to 256 and lose the strobe effect but the sound made me suspect there would be an issue so I only ran it this way for a few seconds to prove the point.

Another option I have to work around it is to switch to white and drop the brightness right down and this will do for the minute. Just wish I understood it as it makes little sense to me.

sumotoy, 2015-06-28:
Ok, I have the exact display you are using in my spares, I'm actually using a ER-TFTM070-2, an 800x480 but 5".
Let me see if I can reproduce the problem.
did you confirm that
tft.clearScreen(0xB5B2);
tft.brightness(226);
cause flickering?

sumotoy, 2015-06-28:
It's really important you power cycle between libraries, I confirm that some parameter will not be changed between resets (hardware and software), this has been confirmed by RAiO some day ago. Two weeks ago I loosed an entire day after tweaking some parameters then discovered that a PLL register never changed and after a cold start my display remained blank, I spend several hours to discover that some system parameter not change even after trigger the reset pin.

The-Experimentalist, 2015-06-28:
Okay I will make sure to power  cycle the display on library change. 

I  can confirm that the 

tft.clearScreen(0xB5B2);
tft.brightness(226);

does cause flickering

In fact I can confirm that I get an almost scroll like strobe effect from top towards bottom of the screen for all brightness settings from around 60 upwards but not when at 255 for all the colours mentioned earlier:

RA8875_LIGHT_GREY
 RA8875_LIGHT_ORANGE
 RA8875_DARK_ORANGE
 RA8875_PURPLE

ER-TFTM070-2 is surely a 7" I thought ? 

(070 = 7", 050 = 5" I thought. I also have a ER-TFTM050-3 5" in my spares)

Going to bed now so I will respond to any other questions tomorrow, thanks for all your hep and time as ever :0)

sumotoy, 2015-06-29:
(070 = 7", 050 = 5"
No, the ER-TFTM070-2 it's a 5" 800x480.

sumotoy, 2015-06-29:
I have connected ER-TFTM070-5, tried with:
tft.clearScreen(0xB5B2);
tft.brightness(226);
No flickering, steady grey screen at any angle.
Tried
RA8875_LIGHT_GREY
RA8875_LIGHT_ORANGE
RA8875_DARK_ORANGE
RA8875_PURPLE
All steady, no flickering. So it's something else....
Also tried this in loop:
  for (b = 0; b <= 255; b++) {
    tft.brightness(b);
    delay(10);
  }
Result in a smooth fade out, no flickering.
I only noticed that RA8875_PWM_CLK_DIV2048 works smoothly on fast SPI speed o I'm think to set it for Teensy and SAM as default and leave 1024 for slower processors.

I wonder if you can try:

1) Connect ER-TFTM070-5 directly to Teensy 3 and bypass everithing.
If works, maybe the 74HC125 goes in oscillation due high speed SPI, in that case check if there's a good decoupling capacitor very near 74HC125 supply and if the 74HC125 ground its short, also do not leaved open un-used ports, tie to ground.
If not works, there's many options:
A) The PWM interfere in some way with the power supply and create digital noises, if you have an oscilloscope check if supply it's clean when using brightness (it happen to me when using 555 in the past), also check if your supply doesn't change voltage when using PWM.
B) The display mounted on your ER-TFTM070-5 it's different than mine, in that case I need some HiRes photo to confirm this.
C) ???

The-Experimentalist, 2015-06-29:
Thanks, you got it. Sorry to waste your time!

Having run through the checks suggested I found a couple of issues.  I did have one unused port on the 74HC125 that was not grounded and the ground line may have been a little long. I also checked the supply when the PWM was active as you suggested I found it was dipping. I have just finished an evening of redesign and testing. 

I had been using USB power from a powered USB hub rated at 1A per USB port. I have now added an external supply and mounted the connector so as to make the ground connection to the 74HC125 as short as possible. I also grounded the unused port on the 74HC125. 

I think the power supply dipping was the main issue but I am not sure which was the problem as I fixed all three at once. Whichever it was the display is now solid without flickering. I tried with division up to 8192 without issue but had a pronounced strobe effect at 16384. Therefore it works fine at the current 1024 and proposed 2048. What I would say is the strobe effect was worse at 2048 with the original wiring so to provide low power capability it may be worth leaving the library as is with the 1024 divisor. I really want to try and make this a USB powered project, I know it is borderline but I really want to avoid an external supply if I can. My guess is if I use a good quality high power USB cable it will work fine with the original wiring. I guess that the cores in the USB cable are not up to supplying the required current without causing the voltage drop.

It must have been right on the limit of going wrong with the other library also. I would not have wasted your time had the other library not worked without issue producing no strobe effect. I guess driving it at the higher clock speed must have pushed the power consumption up just enough to cause the issue with my original wiring.

Anyway, thanks very much for your time, and sorry to waste it on something that turns out to be an issue that is nothing to do with the library.

sumotoy, 2015-06-30:
No problem happy you solved!

closed 2015-06-30 by sumotoy