sumotoy/RA8875 #56: New beta 0.70b5

sumotoy, 2015-06-28:
I'm releasing soon a new beta that will be the basis of all upcoming ones. It's taking time because I changed really a lot of stuff but believe me you will love it.
First, bug fixes! All the touch Screen stuff it's brand new, much faster than old one, easier and can use internal ISR but you can switch to digitalRead method with one command so you can protect other code as needed.
The most important new feature it's the brand new external font rendering engine, now part of the text methods and can use the same commands like write,print,println with all variants print(data,HEX(etc.), setCursor and all it's variants (like CENTER and autocenter flag), etc.
The font engine can use normal or compressed fonts and it's line based rendered that has the same quality of the pixel rendered but much faster and since that the scaling it's really high quality and it's not limited to 3 so you can draw big chars..
Now you can use all internal fonts, external rom fonts and user fonts with the same commands all together and sharing colors and cursor position.
I have also fixed the pixel streaming to use with SD cards, now line based rendered.
I'm sorry but some function have changed name, basically all the touch stuff, I will provide a list of changes to easy change your code.
I have added some very useful graphic macro as well.

The-Experimentalist, 2015-06-29:
With regard for your aim of no issues I currently have some more issues to report relating to using the following command with the latest library versions:

 tft.setScrollMode(BUFFERED);

I get strange results with text (extra random characters appended when using tft.print()) and the 2nd layer does not render properly. This all worked fine in earlier library versions. I have not had time to report these properly yet but I will try and find some time later.

sumotoy, 2015-06-29:
All text issues will be fixed in upcoming version, I have redesigned completely the textWrite function to incorporate font rendering and I'm actually carefully checking text in all situations.
But thanks to point me to the scroll issue, I will check this.

The-Experimentalist, 2015-06-29:
Now I have the brightness issue sorted I will try and spend some time to get you a sketch to reproduce the layer issue as described above. I should be able to do this in the next couple of days.

I have also managed to fix the issue I was having with all versions of your library over 0.68. This would cause

The-Experimentalist, 2015-06-30:
I wrote a sketch to demonstrate the issue and found my demo worked fine!

 I have refactored my code and fixed the issue but once again I do not understand it. I originally had a for loop that reads text from the SD card and then writes it to the display adjusting virtual co-ordinates to absolute co-ordinates on a particular layer (L1 + L2 in buffered scroll mode so virtual screen size is W x (2 x H)). In each loop iteration it first read from the SD, separate file per iteration, and then wrote the text from the file to the display. If I do it this way it does not render correctly and the both layers get corrupted.

If I read all the text into an array in one loop and then write it out to the display in another loop it works fine. Therefore it seems to have trouble switching quickly between the SD and the display. I have 10K pull ups on both chip select pins and the RA8875 fully isolated.

Any ideas what I might be doing wrong or anything I can check or try? You have way more experience and electronic ability than me I am truly a hobbyist at both programming and electronics.

sumotoy, 2015-07-01:
can you send me the sketch? sumotoy at gmail.com

The-Experimentalist, 2015-07-01:
Sending you a sketch now that demonstrates my issue. At the top of the sketch:

bool showGood = true;

Run it like this, then run again with it set to false.

sumotoy, 2015-07-05:
I just published the 0.70b5 preview, massive changes but still have to fix up something. At list the SD issues solved (tested personally with 3 types of display and Paul's SD optimized for Teensy and SDfat).
I didn't have time to check your code but I will, maybe with last changes of the library some issue gone.

sumotoy, 2015-07-11:
Issue fixed?

closed 2015-07-18 by sumotoy