sumotoy/RA8875 #57: The infamous square qr-code like that appear sometime on top-left...

sumotoy, 2015-07-05:
In some situation (almost related to Layers) sometime appears on top-left a qr-code like square that doesn't go away.
This not happen always!
Since it's not easily replicable I have some trouble to understand what cause this but I'm pretty sure can be fixed really soon, I'm pretty sure it's caused by an RA8875 hardware bug related to some register timing during startup initialization.
I have noticed that appear more if you are not using the reset pin and this point me to an initialization tweak.

The-Experimentalist, 2015-07-08:
This was the issue I had from all versions of the library after 0.68 that I could just not consistently reproduce. It drove me mad. I have not had the issue since I reconfigured my test rig and added a more powerful power supply with your guidance.

I explain exactly how I fixed it here:

https://github.com/sumotoy/RA8875/issues/55#event-344221373

sumotoy, 2015-07-08:
Thanks, but I've noticed it return from time to time. I've finally detect where is the problem and get fixed in some days!

sumotoy, 2015-07-11:
Ok, find the cause, it's a power supply problem! The RA8875 expect exact 3V3 volts, going under that value create problems with his internal memory.
Since the Eastrising display (some) has an internal LDO regulator from 5V to 3V3 that has a drop, giving less than 4.8V don't permit LDO to compensate the drop.
The solution it's simple, don't supply less that 5 volt (or 3V3).

closed 2015-07-11 by sumotoy