sumotoy/RA8875 #58: Color565 wrong parameter names

gmartau, 2015-07-08:
First, I want to thank you for your priceless work.
The chip uses the RGB (MSB-LSB) order color. The inline member function Color565 assemble the color parameters in BGR order.
Maybe instead of the inline function it is better a macro, so we can define a list of basic colors (currently, there is no definition as constants for basic colors).

sumotoy, 2015-07-08:
basic color are defined in a separate file! Check settings

gmartau, 2015-07-14:
In the RA8875ColorPresets there are defined basic colors
const uint16_t	RA8875_RED              = 0xF800;
const uint16_t	RA8875_GREEN            = 0x07E0;
const uint16_t	RA8875_BLUE             = 0x001F;
So the order is RGB considering byte order MSB-LSB

The function Color565(r,g,b) assemble the parameters in the BGR order which is wrong.
Probably this is why some examples uses the parameter reversed like
tft.Color565(bc, gc, rc)

Now if you use the function with parametes (255, 0,0) instead of red, you will get blue.
If the color order depends of chip initialization then I think that this fact should be hidden in the library. However, I did not found a such information in the datasheet.

sumotoy, 2015-07-15:
Ok, you right. The chip actually initialized as RGB. There's a datasheet 2.1 that shows how to change order, I've used it initially for maintain compatibility with the code from my other library but dropped because the 16 to 8 color bit conversion that RA8875 perform with his hardware accelleration produces unpredictable results.
Thanks for point me this bug, it's embarassing one, Ill fix it and change all examples.
What about your macro proposal?

closed 2015-07-18 by sumotoy