sumotoy/RA8875 #59: Support for unicode fonts chip

gmartau, 2015-07-08:
There are any chances to support the font chip GT30L16U2W?

If not, can you please make public the modified version for "lcd image converter" utility so we can add a basic font like Arial (a big size). The embedded chip font is horrible scaled and the above external font chip is not supported in the current version. The already included fonts in library are too small and do not look good scaled, or are too cool for a serious display (those present in the 0.70 preview release that were removed now).
Thank you.

sumotoy, 2015-07-08:
For LCD Image Converter, it's planned, but I need to send to author and get an authorization since it's not my code, I just modified it. The reason it's simple, it's the only converter I find around that it's able to compress font and not using a much faster way to store font (instead of a big array for glyph and another array for offset).
Btw, I can convert a font if you point me to an usable arial that it's has copyright and it's free!
About GT30L16U2W, should be easy but without it I cannot really know if working, look inside library the function that set the external ROM fonts, basically it sets 3 (or 4 bits) that correspond to any font chip handled by library.
About unicode, It's planned very soon, I just taken some time because the 0.70 preview was a big mess and I'm still fixing some small error, I have planned Unicode and 2 or 3 bit font decoding.
The external ROM fonts has 2 methods, direct code and indexing, this last one I never tried and there's actually no examples of this method (at list googling it give no results) and since I'm 1 byte language based I have some difficulties for start but I really want to solve it.

Thanks for try this library and please let me know if there's any bug.

sumotoy, 2015-07-11:
Just a note, the GT30L16U2W should be the same of GT23L16U2W.
Accordly datasheet the bit 7..5 of the register SFRSET should be set to 001.
I have upped a new version that should support GT30L16U2W.

closed 2015-07-11 by sumotoy

gmartau, 2015-07-12:
I can confirm that the font chip GT30L16U2W works in the latest version. I tested the STANDARD, ARIAL and ROMAN font family. I found that the working encoding is ASCII and for chars something between '!' and '~' (numbers, usual punctuation, English letters). For the rest of combinations I get garbage.

sumotoy, 2015-07-12:
Because it needs proper decoding, ASCII use one byte but here 2byte are needed and each encoding needs a proper method.
Inside ROM there's a lookup table that redirect the 2byte address to the correct glyph.
The Arduino Print/Write command it's limited to const char so I need to write a brand new method for this, together a decoded for the ROM loockup table, It's planned but I have to fix all the library bugs first.