sumotoy/RA8875 #60: Font centering using internal font is no longer centered

The-Experimentalist, 2015-07-08:
When I compile with 0.70 fonts are correctly centered, with 0.70b5$RFV5tgb6yhn7ujm

sumotoy, 2015-07-08:
ah, ok. It's an easy fix. I'll check and fix later

sumotoy, 2015-07-11:
Are you sure about this?
```
  tft.setFontScale(3);
  //tft.setCursor(CENTER, CENTER);
  tft.setCursor(tft.width()/2, tft.height()/2,true);
  tft.print("AA AA");
  tft.drawLine(0, 0, tft.width(), tft.height(), RA8875_RED);
  tft.drawLine(0, tft.height(), tft.width(), 0, RA8875_RED);
  tft.drawFastHLine(0, tft.height() / 2, tft.width(), RA8875_RED);
  tft.drawFastVLine(tft.width() / 2, 0, tft.height(), RA8875_RED);
```

Looks perfectly centered, at any scale and using CENTER or autocenter flag!!!!

The-Experimentalist, 2015-07-14:
Sorry for slow reply, been away. Not had chance to test this yet. All I know is when I compile with the later code all my button legends went off centre. Will probably not get any time to work on this again for a couple of weeks

sumotoy, 2015-07-15:
no prob, I'll close it, eventually you can still open again issue if you find something else I've missed.

closed 2015-07-15 by sumotoy