sumotoy/RA8875 #63: PROGMEM problems with 8bit arduino's

sumotoy, 2015-07-19:
A couple of users reported compiler errors with the external fonts with 8bit arduinos, I'm sorry about that, I simply missed to insert a macro in the last release but I will include in the next version.
The next version will have a working compressed font decoder as well.

GJakobsche, 2015-07-29:
Dear Sumotoy:

Thanks for the great library.

The README file for V0.70b10 says you fixed the missing macro that was causing PROGMEM problems for 8-bit Arduinos. However, I get compile errors using V0.70b10. In arial_numbers.c, I get about two dozen compile errors. Mostly "Unknown type name 'tImage'" such as on the line:

static const tImage arial_numbers_0x20 = { image_data_arial_numbers_0x20, 21};

but also "Unknown type name 'tChar'" on the line:

static const tChar arial_numbers_array[] = {

and "Unknown type name 'tFont'" on the line:

const tFont arial_numbers = { 21, arial_numbers_array,0,32,0 };

I am compiling using the embedXcode plug-in (http://embedxcode.weebly.com/) for Xcode (the Apple IDE, https://developer.apple.com/xcode/). I am using an Arduino Mega 2560 clone made by Inland. 
Do you think there is a conflict between your RA8875 library and the embedXcode plug-in?

I don't get this error when using the Arduino IDE. Instead, I get other errors. However, I gave up using the Arduino IDE long ago, and I don't want to go back to it. embedXcode & Xcode make a much nicer IDE.

Best regards,
George

sumotoy, 2015-07-29:
hello,
oops! arial was not fixed (and complete), it's just a test and I forget in the last release, please do not use it.
I have tested with a mega2560 and works.
I have not tested with other ide, I will try embedXcode but I can tell you that I dubt spi transcaction are supported with xcode ide.

GJakobsche, 2015-07-29:
I have been using embedXcode and the earlier version of the RA8875 library
(0.70b2) without trouble. SPI worked fine.

Best regards,
George

sumotoy, 2015-07-29:
Hello,
SPI works aniway, I have several workaround in the library that allow SPI work with or without SPI Transactions, however with SPI transactions it's much faster since I can modulate SPI speed and always use the max speed possible, this is not so evident with 16Mhz processors but with Teensy, DUE and Energia it makes a big difference.
Apart the ifamous arial_numbers (that will be released complete with letters and symbols in the b11 release) did you get any other error with embedXcode?

GJakobsche, 2015-07-29:
Thanks for your message. Previously, I did not appreciate the difference
between getting SPI to work and using SPI Transactions. I just ordered a
Teensy 3.1, so I will have to investigate this aspect.

I look forward to the b11 release. Thank you for working on this great
library.

No, I have had no other trouble compiling my program using the RA8875
library and embedXcode. I am still writing code. My program is now up to
about 50 pages of source code, although many lines are comments.

I like embedXcode + Xcode much better than the Arduino IDE. For example,
the user interface in embedXcode + Xcode handles libraries, and multiple
source file programs in general, better than Arduino IDE. The Arduino IDE
could not even compile some combinations of libraries I wanted to use. By
the way, embedXcode also supports Teensy and several other boards and
processors. Like the RA8875, embedXcode includes code that is selectively
compiled, based on the target processor/environment.

In case you are interested, my program drives a digital signal synthesizer
to generate radio frequency signals, which are fed to an antenna system
under test. A circuit generates a signal indicating how much power is
reflected from the antenna, and the program calculates a voltage standing
wave ratio (VSWR), which is a figure of merit of the antenna system. I use
a 7" EastRising capacitive touch screen as the UI to accept user inputs
selecting frequencies to generate and plot resulting VSWR values. The UI
includes a scroll area (not using the display scroll feature, because I
need to scroll more than twice the height of the screen) for user selection
buttons, start and clear buttons, buttons to save date to a microSD card
and load and display the saved data, change user settings, etc. The save
and load features are not yet implemented.

-George

sumotoy, 2015-07-30:
Hi George,
good work! I'm happy that library it's useful in some way.
Btw I cannot use embedXcode because I have only Win and Linux here.
You can verify if your embedXcode support SPI Transaction by put this in your sketch:

#if defined(SPI_HAS_TRANSACTION)
Serial.println("SPI Transactions compatible");
#else
Serial.println("NOT SPI Transactions compatible");
#endif

Did you already tried SD card with RA library? Have a look in the wiki!

GJakobsche, 2015-07-31:
Thank you for the code to test compatibility with SPI Transactions. I compiled and uploaded the code with embedXcode to both my Mega 2560 clone and to my Teensy 3.1. In both cases, the output indicated NO SPI Transactions compatibility. I used the Arduino IDE (V1.6.1 with Teensyduino 1.24) to compile and upload the code to both boards, and the output also indicated NO SPI Transactions compatibility.

This was surprising. What is SPI Transactions compatible?

-George

sumotoy, 2015-08-01:
Arduino IDE (V1.6.1 with Teensyduino 1.24), SPI library have SPI transactions for both Arduino and Teensy with this version so I'm really surprised.
The SPI transaction has been introduced in standard Arduino as well so you need to install Teensyduino only if you use teensy.
Infos:
http://dorkbotpdx.org/blog/paul/spi_transactions_in_arduino

GJakobsche, 2015-08-13:
'Sorry, I made a silly typographical error when I entered the test program. SPI Transactions >>are<< supported when I compile using Arduino V1.6.1 + Teensyduino 1.24 (for both Arduino Mega 2560 Inland clone and Teensy 3.1) and when I compile using embedXcode. As far as I know, embedXcode uses the same compiler and libraries as are used by the Arduino IDE. The Arduino IDE must be installed before installing embedXcode, so one would expect the same library support in both environments.

To answer an earlier question, no, I have not yet tried accessing the microSD card reader on the EastRising display board using the RA8875 library. Very early on, I ran a test program using some other library that supported only an microSD card reader, and I was able to access the microSD card. However, as noted in various places, I need to include a buffer if I want to access both the microSD card reader and the display using the same bus. I have the chips. Maybe this weekend I will get a chance to wire them in.

-George

sumotoy, 2015-08-13:
With Mega 2560 you can try without chip isolation, maybe it works since the low SPI speed and SPI transactions but pull up RA8875 and SD cs pin's to be sure that SPI don't interfere.
Btw I strongly suggest the RA8875 isolation (wiki) due SPI Mode 3, not compatible with SPI Mode 0 of the SD card.

sumotoy, 2015-10-12:
b11 just released has correct and no-messy workaround for MCU that need PROGMEM

closed 2015-10-12 by sumotoy