sumotoy/RA8875 #67: fast draw image

langwadt, 2015-08-07:
Not really and issue just a comment.  
The wiki seems to indicate that you can't do a fast data transfer to the screen, I know you can (atleast with the whole screen). I use a 800x480 buydisplay lcd as  frame buffer in a different project ~3fps in 16 bit color ~6fps in 8bit color with 20MHz spi 

set activewindow, set startaddress, cmd 0x02,  then stream the all the pixel data while keeping CE low

sumotoy, 2015-08-07:
It's exact what I'm doing here, even faster thanks to the 22Mhz SPI and SPI Transactions.
Btw I cannot call 'fast'  3fps and 6fps!

langwadt, 2015-08-08:
ok, I misunderstood it as the chip not supporting a command to stream data directly to memory, it's hard to fault the chip for not being able to transfer data faster than the interface allows. Are you saying there is a limitation when running in parallel mode? 

I don't use the library, it is a totally different project using an FPGA with a cpu running linux, I don't use any special functions, I just set up with a slow spi and then use something similar to DMA to stream the data at maximum rate

sumotoy, 2015-08-08:
dunno really but I suspect that the presence of the GPU force limitations, datasheet says from 9ns to 16ns so probably it can arrive at 10, 12fps max but this just by looking datasheet.

langwadt, 2015-08-09:
afaict tell from the datasheet the limit on parallel access is 15MHz, that comes out to almost 40fps

closed 2015-08-13 by langwadt