sumotoy/RA8875 #69: Version:0.70b10 not working

prathapchandra, 2015-09-04:
first of all, thank you for your work on this immensely capable library

i am pulling the plug too early, sorry about this
i have a set up where my DUT is communicating with Arduino UNO and sending some data. i am trying to demonstrate the DUT capabilities using a 5" LCD (800X480) along with Adafruit RA8875 breakout board (both purchased from Adafruit)
the set up works fine with Adafruit library , but am not able to get your "basic-setup" code to work .
all i am getting is a silver screen
the RA8875 is acknowledging that it is being activated
i am trying to look into your .h and .cpp files to find out if i need to tweak something to fix the issue
any help will be highly appreciated

thanks again
-Prathap

sumotoy, 2015-09-04:
i'm still in vacation so I don't have the stuff with me. Once back I promise to look at this.
I have tried before going vacation with a custom board with Mega2560 that can work at 3v3 and it was working, this CPU uses code compatible 100% with UNO so it should work, but I have tried not the Adafruit board.
The Adafruit board uses a RA GPIO port to turn on display and a level converter, when you connect the stuff you get a black screen then goes silver?

prathapchandra, 2015-09-04:
thanks for your reply and sorry to disturb you on your vacation
the screen goes directly to silver or rather white-out

sumotoy, 2015-09-05:
this mean no SPI communication, I will check next week but in the meantime please double check your connections and try again!

prathapchandra, 2015-09-23:
hello
hope you had a fun filled vacation
I am still not able to get your code to work, I need to upload bitmap images to the LCD, and am using a SD Data logger from Adafruit.
the code provided by Adafruit takes 80s to load a 24bit depth 800x480 screen
i was hoping your code could achieve rendering the data faster, as my understanding is that you are using highest possible speed on the SPI bus.
when i run the SD test sketch , the serial monitor says that RA8875 was detected and that the image was detected and its size and also the turn around time, however, as is the problem statement, I see nothing but a white screen.
one observation though, when I tried to load "Alert.bmp" the screen would blink, as if being turned on and off periodically.
Any thoughts would help
thank you

sumotoy, 2015-09-23:
Yes, I'm just back from vacation, thanks.

I just checked mine 800x480 eastrising with a MEGA2560 board at 3V3 volt, the adafruit should be same (apart a GPIO for turn one screen that I already manage on library), just run the benchmark and...works.
1) Before attach SD holder, have you tested the display with UNO directly?
2) How you are powering up the display? From UNO?

Before attach SD you have to be sure that everithing works, the 800x480  use a lot of current and library initialization I use power up with the max values (the adafruit library not), some user had the same problem, try to connect a voltmeter and check the voltage of the display during all power-up, if it goes under 4,50V it's not sure the onboard regulator can supply 3v3 and the RA goes in limbo state where it doesn't do nothing.

prathapchandra, 2015-09-24:
1) Before attach SD holder, have you tested the display with UNO directly?
Yes, the LCD works just fine with rectangles and triangles

2) How you are powering up the display? From UNO?
Yes, 5V output from UNO

I didn't notice any loading on the 5V source from UNO, to be certain i hooked up a DC power supply with 8V as input power supply, I still see the same screen

sumotoy, 2015-09-24:
I have prepared a CD5050 board to connect an arduino board to my 800x480 and tested and...even this works.
Can you please take a picture of the display and give me the details? Not the board, just the display.

sumotoy, 2015-09-25:
In the meantime I have almost ready a new version of the library that has been tested with many MCU's, it has a new font management that can handle many user fonts.

prathapchandra, 2015-09-28:
below is the LCD that i am using
https://www.adafruit.com/products/1596
thank you

sumotoy, 2015-09-28:
It looks a 800x480 5" and not 7" so it should not consume a lot...
I have asked to another user that has an Adafruit board with a 5" to check current library and he confirm that works, I've tested with a uno two different eastrising 5" and 7", both 800x480 and works. At this point and since your display works with adafruit I have to see how you are initializing the display, can you post only the setup code (any, just want to see how you are trying to initialize and wich pin you are using, etc.)

closed 2015-09-28 by sumotoy

prathapchandra, 2015-09-29:
Hello
am using Arduino UNO Pin 8 for CS of the RA8875 breakout board and the Pin 10 as CS for the SD card data logger.

rest of the drivers is directly from Adafruit
thanks

sumotoy, 2015-09-29:
....I was asking the what there's inside:

void setup(){
//please post the code you have here
}

prathapchandra, 2015-09-29:
void setup() {
 Wire.begin();                // join I2C bus (address optional for master)
 Serial.begin(9600);          // start serial communication at 9600bps

 if (!tft.begin(RA8875_800x480)) 
  {
    Serial.println("RA8875 Not Found!");
    while (1);
  }
  Serial.println("Found RA8875");
  
  xScale = 1024.0F/tft.width();
  yScale = 1024.0F/tft.height();

  tft.displayOn(true);
  tft.GPIOX(true);                              // Enable TFT - display enable tied to GPIOX
  tft.PWM1config(true, RA8875_PWM_CLK_DIV1024); // PWM output for backlight
  tft.PWM1out(255);
  
  HomeScreen();         //Initialise the LCD to display the Home Screen
  
  pinMode(RA8875_INT, INPUT);
  digitalWrite(RA8875_INT, HIGH);
  tft.graphicsMode();
  tft.touchEnable(true);
    
  Serial.print("Status: "); Serial.println(tft.readStatus(), HEX);
  Serial.println("Waiting for touch events ...");
}

please see

sumotoy, 2015-09-29:
prathapchandra, I strongly suggest you to read the wiki before claim that not works!
The adafruit library work out-of-the-box only for adafruit products but this library has been designed for all spi drived RA8875 so it has m ore options to adapt different brands.
In the initialization yoiu can use:

RA8875_480x272,
RA8875_800x480,
RA8875_800x480ALT,
Adafruit_480x272,
Adafruit_800x480

Guess what you beed to use?

if (!tft.begin(Adafruit_800x480)) 

You have an adafruit and I was sure you are using+

prathapchandra, 2015-09-29:
aargh.. it works now.. apologies for the confusion
i used Adafruit_800X480 and it works great
the visuals are just what i wanted
i wish i could have solved this issue three weeks ago
nevertheless, i appreciate your passion for helping others

sumotoy, 2015-09-29:
no problem. But read wiki, there's very important infos about use this chip with other SPI device!

prathapchandra, 2015-09-29:
I will
thank you