sumotoy/RA8875 #70: Custom Char issue...

WerkAG, 2015-09-25:
Hi sumutoy

In first place, congratulations for your project of making a full featured RAIO 8875 library.

The userCustomChar issues (RAIO bug) may not be only related to the font scale .
I'm doing some tests with an EastRising 800x480 display and your library:
The userCustomChar example included in the library works great, but when I try to print a custom user char with  some code I'm writing, just after the command to display the custum char, the display seems to zoom only in the horizontal position, colors also change (I'm not using layers).
Until now I'm unable to determine what in my code, breaks the userCustomChar function, so I'm only reporting it.
Below is an image of the screen, after printing one custum user char.

[image]

MorganSandercock, 2015-09-25:
Does it make a different what position you print the char to?
It seems like the custom chars are set up for spraying bits into memory and there's no checking if the destination is actually the display memory.

sumotoy, 2015-09-26:
Due RA8875 Hardware bug the custom char are badly affected by char zoom command, I was sure I already fixed this time ago (forcing custom char to avoid zoom) but your code seems bypass my fixups, I will like to see code, send sumotoy at gmail.com

sumotoy, 2015-09-26:
...Or wait the next b11, it has a new user char support with a pyton code to convert any font so you can avoid custom char and use custom font that support zoom as well.

WerkAG, 2015-09-27:
Hi, thank for the replay,
Yes, I did a look at the library and saw the code to avoid zooming the custom char.
To try to illustrate the issue, I will post below, a modified UserCustomChar code. The issue appears to affect all text printed before the custom char be printed.
The Custom Char is not an important feature to me, custom fonts will be more! I can wait for the next b11.  Thanks for your work , with this great library.

/*
An example of how easy it's upload custom 8x16 symbols.
 RA8875 has 255 locations reserved for user custom designed symbols.

 Works with Arduino 1.0.6 IDE, Arduino 1.6.x IDE
 */

#include <SPI.h>
#include <RA8875.h>

/*
Arduino's
You are using 4 wire SPI here, so:
 MOSI:  11//Arduino UNO
 MISO:  12//Arduino UNO
 SCK:   13//Arduino UNO
 the rest of pin below:
 */

#define RA8875_CS 10 //see below...
/*
Arduino's 8 bit: any
*/
#define RA8875_RESET 9//any pin or nothing!

RA8875 tft = RA8875(RA8875_CS,RA8875_RESET);//Teensy3/arduino's

  #define RA8875_DARKGRAY        0x2945
  #define MenuBar_Background     RA8875_DARKGRAY

//define some custom char (8x16)
const uint8_t battery1[16] = {
  0b00111000,
  0b00111000,
  0b11111111,
  0b10000001,
  0b10111101,
  0b10111101,
  0b10111101,
  0b10111101,
  0b10111101,
  0b10111101,
  0b10111101,
  0b10111101,
  0b10111101,
  0b10111101,
  0b10000001,
  0b11111111
};

const uint8_t battery2[16] = {
  0b00111000,
  0b00111000,
  0b11111111,
  0b10000001,
  0b10000001,
  0b10000001,
  0b10111101,
  0b10111101,
  0b10111101,
  0b10111101,
  0b10111101,
  0b10111101,
  0b10111101,
  0b10111101,
  0b10000001,
  0b11111111
};

const uint8_t battery3[16] = {
  0b00111000,
  0b00111000,
  0b11111111,
  0b10000001,
  0b10000001,
  0b10000001,
  0b10000001,
  0b10000001,
  0b10111101,
  0b10111101,
  0b10111101,
  0b10111101,
  0b10111101,
  0b10111101,
  0b10000001,
  0b11111111
};

const uint8_t battery4[16] = {
  0b00111000,
  0b00111000,
  0b11111111,
  0b10000001,
  0b10000001,
  0b10000001,
  0b10000001,
  0b10000001,
  0b10000001,
  0b10000001,
  0b10111101,
  0b10111101,
  0b10111101,
  0b10111101,
  0b10000001,
  0b11111111
};

const uint8_t battery5[16] = {
  0b00111000,
  0b00111000,
  0b11111111,
  0b10000001,
  0b10000001,
  0b10000001,
  0b10000001,
  0b10000001,
  0b10000001,
  0b10000001,
  0b10000001,
  0b10000001,
  0b10000001,
  0b10000001,
  0b10000001,
  0b11111111
};

//next 3 are part of the same char
const uint8_t battery6a[16] = {
  0b00000111,
  0b00000111,
  0b11111111,
  0b11000000,
  0b11000000,
  0b11000000,
  0b11110000,
  0b11111100,
  0b11111111,
  0b11111111,
  0b11111111,
  0b11111111,
  0b11111111,
  0b11111111,
  0b00000000,
  0b00000000
};

const uint8_t battery6b[16] = {
  0b11100000,
  0b11100000,
  0b11111111,
  0b00000011,
  0b00000011,
  0b00000011,
  0b00000011,
  0b00000011,
  0b00000011,
  0b11000011,
  0b11110011,
  0b11111111,
  0b11111111,
  0b11111111,
  0b00000000,
  0b00000000
};

const uint8_t battery6c[16] = {
  0b00011111,
  0b00000011,
  0b00001111,
  0b00000011,
  0b00011111,
  0b00000000,
  0b00011111,
  0b00000011,
  0b00001111,
  0b00000011,
  0b00011111,
  0b00000000,
  0b00000000,
  0b00000000,
  0b00000000,
  0b00000000
};

void setup() 
{

  uint8_t i;
  tft.begin(RA8875_800x480);
  //upload chars in address 0x00 to 0x04
  tft.uploadUserChar(battery1,0);//0x00
  tft.uploadUserChar(battery2,1);
  tft.uploadUserChar(battery3,2);
  tft.uploadUserChar(battery4,3);
  tft.uploadUserChar(battery5,4);//0x04
  tft.uploadUserChar(battery6a,5);//
  tft.uploadUserChar(battery6b,6);//
  tft.uploadUserChar(battery6c,7);//

 tft.setTextColor(RA8875_WHITE, RA8875_BLUE);
 tft.setFontSpacing(3);
 tft.setFontScale(1,0);
 tft.setCursor(10, 11);
 tft.print(F("Just a line with some text"));
 delay(5000);
 //now custom char are stored in CGRAM
 tft.setTextColor(RA8875_WHITE,RA8875_BLACK);
 tft.setFontScale(0);
  for (i=0;i<5;i++){
    tft.setCursor(tft.width()/2,tft.height()/2);
    tft.showUserChar(i);//retrieve from 0x00 to 0x04
    delay(200);
  }
 tft.setTextColor(RA8875_WHITE, RA8875_BLUE);
 tft.setFontSpacing(3);
 tft.setFontScale(1,0);
 tft.setCursor(10, 300);
 tft.println(F("Now, after printing the custum char"));
 tft.println(F("the first line of text is enlarged"));
 tft.print(F("and the color was changed"));
 tft.setFontScale(0);
 delay(10000);
}

uint8_t rot;
void loop() 
{
  uint8_t i;
  tft.clearScreen();//exact as tft.fillScreen();
  tft.setRotation(rot);
  tft.setTextColor(RA8875_WHITE,RA8875_BLUE);
  //custom char are managed as text but need to be called
  //by a special function
  //the only text parameter that is not accepted is setFontScale!
  for (i=0;i<5;i++){
    tft.setCursor(10,10);
    tft.showUserChar(i);//retrieve from 0x00 to 0x04
    delay(200);
  }
  
  //now another feature, you can design a char wider
  //by using many char slot combined.
  //below an example, 3 chars wider
  tft.setTextColor(RA8875_LIGHT_ORANGE, RA8875_BLACK);
  tft.setCursor(30,11);
  tft.showUserChar(5,2);
  delay(1000); 
  if (rot > 3) rot = 0;
  rot++;
}

sumotoy, 2015-09-28:
Just a couple of non-related questions:
1) Are you using UNO or similar?
2) How you are supply power to the display?
Thanks for code, it's  useful to see what is causing the break.

sumotoy, 2015-09-28:
Ok, find a temporary fix but should work.
in RA8875.cpp find this function:
void RA8875::showUserChar(uint8_t symbolAddrs,uint8_t wide) 

then change this:
 	if (_useMultiLayers){
		if (_currentLayer == 0){
			writeTo(L1);
		} else {
			writeTo(L2);
		}
	} else {
		writeTo(L1);
	}

into this:

 	if (_useMultiLayers){
		if (_currentLayer == 0){
			writeTo(L1);
		} else {
			writeTo(L2);
		}
	} else {
		//writeTo(L1);//fix the problem
	}

WerkAG, 2015-09-28:
>Just a couple of non-related questions:
>1) Are you using UNO or similar?
>2) How you are supply power to the display?

I'm using a custom board (still a prototype) which uses an ATMEGA 1284P
I tried several ways to supply power to the display, inclusive install a 3.3V regulator in the display board (ER-TFTM050-3), not recommended because it's not easy to install the filter capacitors. I ended using a separate switched 5V power supply, sharing the ground signal with my custom board, this solutions is working well. I'm now waiting for the 7" version, which can be powered directly by 5V.

I take the opportunity to say that the solution exposed in the wiki to isolate the RAIO SPI signals (MISO bug) to allow operation with other devices in the same SPI bus, works great. I'm using the display with a SPI flash memory, and both work flawlessly.

>I cannot remember why I'm force writing in a layer even if multilayers are not enabled, looks silly now but maybe at the time I wrote the piece of code there was a reason?

:-) I understand this perfectly.

Thanks for your help. I will try the proposed changes as soon as I can (I'm away from home for a week). Then, I will post here the results

closed 2015-09-29 by sumotoy

WerkAG, 2015-09-30:
Confirmed, your fix solves the problem!
Thank you.

sumotoy, 2015-09-30:
:)