sumotoy/RA8875 #72: ER3301-1 External font support

WerkAG, 2015-10-09:
Could you please add support for the external font ER3301-1, or give me some pointers on how to do it myself?
This external font is important for those who need Latin characters.
Thank you.

sumotoy, 2015-10-09:
Ok, I will order one and see, unfortunatly every chip it's different and datasheet doesn't help as it should.

WerkAG, 2015-10-10:
Thank you :-)
I have spent last two nights trying to make some changes in your library to use the font chip, but without any success. I'm a very poor programmer too!
No rush, until you add support for it, I will continue trying, this way I also learn something.

WerkAG, 2015-10-17:
Hi,

Seems that the ER3301-1 font ROM, is equivalent to Genicomp GT30L16U2W.
It works if I define GT30L16U2W as external font chip

sumotoy, 2015-10-19:
thanks, in upcoming b6 you can assign as ER3301_1

closed 2015-10-19 by sumotoy