sumotoy/RA8875 #75: setFontScale bug

gmartau, 2015-10-19:
The function sends to chip display wrong values (x for vertical and y for horizontal instead of vice versa).

sumotoy, 2015-10-19:
are u using b10 or b11?

gmartau, 2015-10-19:
I'm using 0.70b11p5.
I used a previous version that had parameters named vertical, horizontal (I noticed the change). It was OK and the getFontWidth and getFontHeight returned correct values. In the latest version these return reversed values. If text is scaled on Y only, the getFontWidth return scaled value and the getFontHeight return the non scaled value not as expected. My local fix was to send to display switched values and the getFontWidth and  getFontHeight remain in sync with the redered text.

sumotoy, 2015-10-19:
Ok, it's my stupid error definetively, fixed in p6 (incoming in a couple of minutes)

closed 2015-10-19 by sumotoy