sumotoy/RA8875 #79: Arduino DUE & FT5206

gcharles81, 2015-11-27:
I am testing the capacitive touch with eastrising 7" and up to now cannot make it working ,

I am using SCL1 & SDA1 on DUE

sumotoy, 2015-11-27:
display it's working or just touch? I had recently a feedback from an user that was working on DUE with capacitive as well but cannot remember witch I2C channel he was using.

gcharles81, 2015-11-27:
The display works fine only the touch is not working I changed the wire
commands from Wire.send to Wire1.send etc but cannot make it
working  tomorrow I will try to add a pull-up to the interrupt pin maybe
it's flooting.

Thanks charles

Thanks

sumotoy, 2015-11-27:
Mmm, I don't even expected to use wire1, have to look inside code. I coded all DUE stuff by looking datasheet since I don't have it.

sumotoy, 2015-11-28:
Try the last version, see inside RA8875UserSettings.h to use Wire1.

gcharles81, 2015-11-29:
Hi I could make it running with the below changes :user Version:0.70b11p6 

Usersettings file still left as Wire. because this is calling the Wire library.

Edit file RA8875.cpp rename all Wire. to Wire1.

Also added 10K pullup to arduino pin serving as touch interrupt in may case I am using pin 17.

Works like a charm thanks for your effort making this amazing library.

P.S one last question if i need only one touch and I define MAXTOUCHES to 1 , is there and call to get only X&Y coordinates .

thanks

sumotoy, 2015-11-29:
This has been fixed yesterday with the p7, the new user settings has an option to use alternative wire for DUE, I never tested but shoul work, maybe you can test it and approve so I will fix as gold master. There's an handly trick to change wire without rename everithing, take a look at the top of cpp file.

gcharles81, 2015-11-29:
Thanks verified working with 	Version:0.70b11p7

sumotoy, 2015-11-29:
Hurra!

closed 2015-11-29 by sumotoy

sumotoy, 2015-11-29:
About last question:
"P.S one last question if i need only one touch and I define MAXTOUCHES to 1 , is there and call to get only X&Y coordinates "
The FT chip it's quite primitive, it simply return all it's registers so why I'm using an array. Setting MAXTOUCHES to 1 will just break the filling routine (and save some ms) so at the end you have location 0 filled with the vars you need (see example).