sumotoy/RA8875 #80: ESP8266 still not working.

sumotoy, 2015-11-29:
The RA8875 uses incompatible SPI mode and the current ESP8266 development it's in early stage, it simply has too much issues but I'm still try to get something work, actually the p8 release has a lot of changes to force compatibility but the hardware mode of the RA blocks any SPI activity and even an isolator chip 74HC125 it's not enough, I'm not sure if the internal SPI Flash chip it's sharing SPI, if yes this will explain why.

sumotoy, 2016-02-09:
The current new ESP8266 beta fixed something in SPI and looks promising, I got a b12 of this library correctly initializing my RA8875 800x480 and perform some task!

deejayspinz, 2016-05-10:
Hi.  I am trying to get a RA8875 working with a NodeMcu ESP8266  (12e) .  Wondering if you have made any progress with the SPI issues you have been experiencing?  My display is a 5" tft from Buydisplay with a RA8875 driver and I'm using this  http://www.banggood.com/Geekcreit-Doit-NodeMcu-Lua-ESP8266-ESP-12E-WIFI-Development-Board-p-985891.html

sumotoy, 2016-05-10:
It doesn't depend on library, the SPI support for ESP8266 it's still not compatible with SPI mode 3 and since ESP8266 has internally an SPI Flash chip sharing the same pin and RA8875 uses SPI mode 3 (with relative pullup on MISO) it will stuck the ESP. If Igrr or someone else will fix this in future ESP8266 core this library will works immediately. I've tried to avoid this using some hardware but believe me it's a challenge.

deejayspinz, 2016-05-10:
Thanks for the quick response.  Can you make any other suggestions on what
I can use? I have an Adafruit CC3000 WiFi sheild that uses the Texas
Instrument wifi device. I could use that and an Arduino Uno maybe?  Ideally
I am looking to create an IoT device to display and show weather
information on a 5" TFT LCD?

Dave

sumotoy, 2016-05-10:
Adafruit CC3000 has some issue with SPI, it uses SPI Mode 3 but I read somewhere that has a pullup on MISO, this limitate a lot the SPI devices you connect with it, if you plan to use with RA8875 you should isolate SPI of the RA by reading my wiki.
Also you should use Paul Stoffregen version of the CC3000 library that is SPI Transaction compatible.
Since RA8875 has most of the stuff onboard, if you avoid external fonts (space consuming) you can probably use with UNO and CC3000 seccesfully, or use UNO and ESP8266 in serial mode?

deejayspinz, 2016-05-10:
It's unfortunate.  I thought with all the excitement around IoT projects
vendors like those who make the ESPs would ensure they had compatibility
with display devices and not make it so complicated.  Thanks again for your
quick response. I'm really hoping to find a way to make an IoT controller
to work with larger LCD.

I wish I knew more about coding at this level so I could help out.  It's
nice to have people like you who are passionate about doing this stuff and
making the world for makers like me easier.  Here's my blog if you are
interested in seeing some of my projects - www.plastibots.com

Dave

Dave

deejayspinz, 2016-05-21:
Hi Max,

 

I thought I would give you an update on the issue I have been having below in hopes it might help you with your RA8875 driver. Long story short, I have an Adafruit RA8875, Adafruit Huzzah ESP8266 and SD card reader.  I was able to successfully connect all of these together with a shared SPI connection between the RA and SD card reader and the Adafruit RA8875 driver.  I can connect to Wifi, display text and read from the SD card.  However, I was having problems with rendering external fonts (seems the Adafruit library may not support this), so I tried yours (RA8875_0.70) and noticed that it wont even start the display using the instructions here (https://github.com/sumotoy/RA8875/wiki/Connecting-the-Adafruit-RA8875-to-your-controller-(and-this-library)) .   What I didn’t do is tri-state MISO with a 74HC125.  I also did change to this (Adafruit_800x480) for tft.begin.

What I am wondering is if maybe there is a fix hiding in the latest adafruit RA8875 driver code that could help you (given that works for me?) – with my specific hardware?  I’d like to use your driver as it supports the external fonts, but at this point, it won’t even start the tft.begin.   

I’ve included the sketch below in case you are interested.  For this case, all the hardware noted above is wired up, but the SD card is not being used. Just trying to get the font rendering working with your /uno/fontrender.ino sample as a base.  It does not get past: tft.begin(Adafruit_800x480);

If interested, I could include a (long) sketch I have that works with the above hardware and adafruit ra8875 library

 

/*

This shows the capabilities of the library text engine that is

extremely flexible and simple! All commands are the same for all type of text

(INT,EXT or Render ones) and things like println works too.

You can use all the print() options (like print(var,HEX), etc. Take a look the print command 

in the arduino guide!) 

*/

 

#include <SPI.h>

#include <RA8875.h>

#include "fonts/akashi_36.c"//minipixel

 

//#define sd_cs 4    //*******  SD CS Pin - need to check the LCD to see which it is

//Adafruit RA8875 Library only supports hardware SPI at this time

// for Huzzah ESP8266 MISO, MOSI, SCLK pins are hard coded: MISO>>pin12, MOSI>>pin13, SCK>>pin14, CS>>pin15 (for RA8875) >> CS(this can be moved)

//dont need define statement as they are built in - I think!

#define RA8875_INT 0  //TX pin on the Huzzah

#define RA8875_CS 15  //pin 15

#define RA8875_RESET 2  //pin 2

//// end for RA8875 LCD

 

RA8875 tft = RA8875(RA8875_CS, RA8875_RESET); 

 

void setup()

{

  Serial.begin(38400);

  //long unsigned debug_start = millis ();

  yield();

  tft.begin(Adafruit_800x480);

  Serial.println("found RA8875");

  tft.setRotation(0);//works at any rotation as well

  tft.setFont(&akashi_36);

  tft.setFontScale(3);

  tft.setCursor(CENTER,0);

  tft.setTextColor(RA8875_GREEN);//notice that! After grandient text will be this color!

  //grandient it's one shot, text color will be reset as the one choosed by setTextColor

  tft.setTextGrandient(RA8875_RED,RA8875_CYAN);//works also with rendered text!

  tft.println("CD 0123");

  yield();

  tft.println("ABCD");//notice that correctly goes in another line

  tft.setFont(INT);

  tft.setTextColor(RA8875_WHITE);//Force white

  tft.println("this is the internal text.");//notice that correctly goes in another line too!

 

}

 

void loop()

{

 

}

sumotoy, 2016-05-21:
Adafruit library talks to RA8875 using SPI Mode 0 that is supported by ESP but the RA8875 clearly demand SPI Mode 3, why they act like this? Dunno, probably they don't read datasheet!
Question. RA8875 works with SPI Mode 0? Mostly of times yes but modulating SPI (as I do for getting the most from RA) is not supported and RA crash, this is why they use single (low) SPI speed for their library.
Another problem, I'm using SPI Transaction, this is necessary for modulating SPI speed and protect RA from other SPI comms, Adafruit clearly don't use this.

> SPI
> SPI library supports the entire Arduino SPI API including transactions, including setting phase (CPHA). Setting the Clock polarity (CPOL) is not supported, yet (SPI_MODE2 and SPI_MODE3 not working). 
> 

I'm surprised that your ESP worked in such a situation, as far I know ESP is NOT compatible with SPI_MODE_3 but recently I've see this:
It claims that CPOL has been updated and working so from the very last version it may be compatible with SPI_MODE_3, have to verify.
If this is true I will try again to include ESP8266 into my library, I already hardly tried but without CPOL it's impossible, so stay tuned.

deejayspinz, 2016-05-21:
I'm not sure why it works, but I have been able to get the combination of Adafruit Huzzah ESP8266 + Adafruit RA8875 + Adafruit 5" TFD + Blynk connected -  all working and can even connect to the web (in my case to download weather data).    Happy to share the entire sketch if it helps.  Also happy to test any changes you want to try. Could it be possible that Adafruit has a 74HC125 built into their RA8875?  I don't know what CPOL is and dont entirely understand this stuff.  Happy to help where I can though.  Thanks for looking into this!

sumotoy, 2016-05-21:
Probably the 75HC125 isolate the RA and protect the ESP SPI.
However everithing works at low speed and I'm not surprise if you get some kind of SPI problems in future.
However I'm currently relase the new TFT_ILI9163C that has some ESP8266 speedups, I will add to RA8875 as well and try to add compatibility as well so stay tuned.