sumotoy/RA8875 #82: Add Code to use PWM on the LED backlighting Pin 14 (BL_Control).

sbfreddie, 2016-01-02:
Sumotoy:
Please add code to impliment PWM of the backlighting control (pin 14) on Teensy 3.x.
I think this would be a great improvement to control the amount of power the 7" display is using.

Thanks for considering this improvement,
Freddie

sumotoy, 2016-01-02:
The brightness command already use the internal pwm pin to control the backlight! Did you try it?

sbfreddie, 2016-01-02:
Max:
I cannot find any mention of the brightness command on the WIKI or the examples, or which pin to connect it to on the Teensy 3.1.
Have you any help located anywhere else?
Thanks,
Freddie

sumotoy, 2016-01-03:
The RA8875 has 2 GPIO with PWM capabilities, Eastrising displays and Adafruit use one of these GPIO's for backlight.
The brightness command uses PWM of the GPIO and Don't Need any MCU extra pin!
On startup and initialization, display has been set at max (255), using brightness at less than 255 just after 'begin' will force display to consume much less current.

sbfreddie, 2016-01-03:
Max:
Thanks for that info.  Please add this info to the WIKI.
What is pin 14 on the ER-TFTM070-5 7" TFT Display Board Connector JP3 for?

Thanks,
Freddie

sumotoy, 2016-01-03:
I think it was provided for use with a possible external pwm source, If I remember well there's a jumper for that, normally it's connected to RA8875 PWM GPIO

closed 2016-01-03 by sumotoy