sumotoy/RA8875 #83: Font files will not build.

sbfreddie, 2016-01-04:
Max:
All the font files blow up my compiler with a ton of errors:
"static const uint8_t image_data_aerial_22_0x20[58] __PRGMTAG_  = {
Expected '=', ',',';', 'asm' or '__attribute__'before'__PRGMTAG_'"
It appears that Xcode thinks this is C++ code and this construct is not allowed in C++.
This one file produces 370 errors.
I am using Xcode 7.2 on Mac OS 10.10.5 with Embed Xcode.
[Compiler errros for aerial_22.c..pdf](https://github.com/sumotoy/RA8875/files/77858/Compiler.errros.for.aerial_22.c.pdf)

Thanks,
Freddie

sumotoy, 2016-01-05:
PRGMTAG it's a placeholder and it's handled by RA8875_CPU_commons.h , it will be replaced in  " " or "PROGMEM" (depends of the CPU) and everithing it's called sequentially by RA8875.h.
This error it's clearly an attempt of xcode to compile fonts BEFORE has been processed by RA8875_CPU_commons.h!
Why Xcode do not respect the RA8875.h directives it's the real problem.

closed 2016-01-17 by sumotoy

pabbott0, 2016-02-28:
I think the problem is more to do with the build system attempting to build the freestanding _font dir of .c files. Those things won't compile by themselves - it doesn't matter if it happens before or after they get included in a header.

I had the same issue with the ILI9163C module when attempting to build via platformio on OSX. I renamed the font folder to ._font and changed the include directive in the main module header , which apparently hid the .c files well enough for platformio to not attempt compiling the font fragments.

sumotoy, 2016-02-28:
These are just prepocessor directives and in the case of font are just placeholders.
The reason it's simple, if an AVR it's detected all these placeholder became PROGMEM, if an ARM it's detected they simply vanish.
But this happen because preprocessor directives have to be interpreted in 'order as files are includes', your compiler just trigger the error 'before' everithing happen, this happen actually only with your compiler and should happen for every attemp to use preprocessor placeholders that are legal and used in the industry!
Why this happen? Maybe the author of your program didn't want to deal with file garbage since when compiler starts and there's an error it have (in the real word) erase all the files generated, but OSX and Apple uses SSD and they have to avoid file generations (an Apple developer directive of 4 years ago if I remember well), checking preprocessor syntax before start compiling it's just a clever idea? To me NO! But not just because I'm using placeholders but you will loose an important feature of preprocessor directives just becose the programmer 'it's lazy'?
But I can be wrong about that, maybe the author did just a mistake or worst decided to avoid completely preprocessor directive placeholders, who knows...
You was the only one to report an error in preprocessor, tested personally on win (4 different compilers), linux (same), OSX (paul stoffregen have OSX and compiled succesfully on arduino so it's included in next teensyduino), even online by using Spark toolchain, I've tried also the ST toolchan on a future beta that will support ST and... nothing, everithing compile.
So what to do? Actually I'm not gonna change 30 files just for this I'm sorry, if you use an ARM, just replace any __PRGMTAG_ with nothing, if you are AVR replace any __PRGMTAG_ with PROGMEM.

sbfreddie, 2016-02-28:
Max:
I just checked the "fonts" folder in Paul's (Paul Stoffregen) version of RA8875 and they are not the same as your version, at least on Mac OS X.  He has changed many of the files names to xxxFixed, they are now .h files, and looking inside they are completely different.  Attached is two files, one from Paul's version of RA8875 and the other from yours, as you can see they are different.  (Please remove the .txt from these files as github does not accept .h or .c files)

Thanks,
Freddie
[dubstreptix_14.c.txt](https://github.com/sumotoy/RA8875/files/150055/dubstreptix_14.c.txt)
[DubsteptrixFixed__8.h.txt](https://github.com/sumotoy/RA8875/files/150057/DubsteptrixFixed__8.h.txt)

sumotoy, 2016-02-28:
The RA8875 will be released with the final version of teensyduino 1.27 I think, actually there's beta 3
The 2 fonts use completely different approach and my library cannot deal with this font.
Can you give me the link of the RA8875 library you find? The only one I can find it's an old Adafruit version that doesn't use external fonts...

sbfreddie, 2016-02-28:
Max:
This font folder is included in the .dmg download from Paul's website.  When you installed it into Arduino on OS X it puts it in /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/RA8875/fonts

I attached the fonts folder which has been zipped.

Thanks,
Freddie
[fonts.zip](https://github.com/sumotoy/RA8875/files/150121/fonts.zip)

sumotoy, 2016-02-28:
Looks like he used an old version with the old font rendering engine
0.69b64, now I'm stuck with 0.70 and prerelease 0.79b11, almost 30 version ahead!
Current version it's much faster with font rendering