sumotoy/RA8875 #86: Resistive touch screen issue.

chuckmynatt, 2016-01-13:
Hi Max. I love your library but I'm having one issue. I'm using an Adafruit RA8875 board with the Adafruit 7" touch screen with resistive touch. My issue is that tft.touched() almost always returns 0. If I am using code similar to this:
while (tft.touched()) 
 { 
  do something
}
If I touch and hold , in other words touch and not remove my finger, the program will enter the while loop but exit after the first pass through the loop. If I place a Serial.print(tft.touched() inside the while loop, it always returns zero even though I'm still pressing on the screen. Strange that it will enter the loop as if tft.touched() is true but further testing returns false.

sumotoy, 2016-01-14:
I'm currently outside for work and no access to my computer but did you try the example provided?
It works or not? 
Also may help know CPU, if you are using interrupt (or better the display init code).

Sent by iPhone

chuckmynatt, 2016-01-15:
Sorry. Noticed I didn't specify a board until after I posted. I'm using a Due. The example works, somewhat. When I touch the screen, it paints the pixels under my finger but also random pixels extending horizontally from my touch point. I've successfully completed the calibration with no issues.
  In case it's helpful, my calibration values are:
#define TOUCSRCAL_XLOW  	975
#define TOUCSRCAL_YLOW 	884
#define TOUCSRCAL_XHIGH	85
#define TOUCSRCAL_YHIGH	151

 Here are the initialization parameters I'm using:
#include <RA8875.h>
#define RA8875_INT 3 
#define RA8875_CS 10 
#define RA8875_RESET 9

And in setup()
  tft.begin(Adafruit_800x480);
  tft.useINT(RA8875_INT);
  tft.touchBegin();
  tft.touchEnable(true);
  tft.enableISR(true);

All wiring confirmed to be on the correct respective pins.

Thanks Max

sumotoy, 2016-01-16:
I will back at home in a week or so. I cannot access any of my stuff here where I am, I just have my phone. But I can tell that resistive touch have worked for many user on different CPU's (included DUE), so maybe you need a pullup resistor of 10K on the INT pin, I raccomand to use INT (even if I've tested the soft based succesfully).

morpheous87, 2016-02-05:
Hi Max,

I'm getting troubles with the resistive touch too. I've tried on ER-TFTM050-3 from buysdisplay.com. With their samples it's working, but using your library is still  not possible to get it running. I didn't try with 10K pull-up resistor, but could this be the reason, if it's working with the other library?
I have also troubles with converting 24bit images. I successfully printed you predefined image, but when I tried to convert with LCD converter, I was unable. The preset you mention in the sketch - "direct24bit_plain" is probably your own, can you provide some more information about convertion?

Cheers!

sumotoy, 2016-02-05:
Not possible to help with such as informations, as far I know the buydisplay examples are mine and it's an old beta of this library (maybe they changed recently).

CPU?
Initialization code?

About conversion, I will complete wiki this weekend...

sumotoy, 2016-02-05:
One common mistake it's the releationship between INT pin and INT number, most arm can attach any int to any pin but AVR and some other CPU has a precise relation, for example using pin 2 in uno will result INT 0 but Teensy 3 can use any number.
Changing pin can break this relation so if it's your case, check RA8875UserSettings file for:
#if !defined(_AVOID_TOUCHSCREEN)
	#define __RA8875ISRINT 0 //// [default 0, pin 2 on Arduino UNO]
#endif
Accordly your processor (and pin used) you should put the right value of __RA8875ISRINT
I'm not a CPU enciclopedy so I don't know exact this relation on every CPU, check your board documents.

morpheous87, 2016-02-05:
Hi,
I had to mention I'm using Arduino DUE, my mistake...
So, I'm trying to use your beta 0.70b11 where you added a restive touch example - easyPaint.ino.

My configuration is this one:
`#define RA8875_INT 23 
#define RA8875_CS 22 
#define RA8875_RESET 24`

And I'm receiving compilation error - 

> 
> easyPaint:64: error: 'class RA8875' has no member named 'touchBegin'
> 
>    tft.touchBegin(RA8875_INT);//enable Touch support!

With this configuration the resistive touch is working with BuyDisplay provided library named "ER-TFTM050-3_Arduino_Libraries_Examples"

sumotoy, 2016-02-05:
try new version, just updated some example.

sumotoy, 2016-02-05:
ER-TFTM050-3_Arduino_Libraries_Examples are exact this library version 0.55! May work but it's light years from current one.

morpheous87, 2016-02-07:
I've downloaded the latest version, where you added a new example - _resistiveTouch/touchTest

I'm still receiving compilation error -

> Arduino: 1.6.7 (Windows 7), Board: "Arduino Due (Programming Port)"

C:\Program Files (x86)\Arduino\libraries\RA8875-0.70b11\examples\_due\_resistiveTouchScreen\touchTest\touchTest.ino: In function 'void setup()':

touchTest:26: error: 'class RA8875' has no member named 'touchBegin'

   tft.touchBegin();//enable touch support for RA8875

       ^

touchTest:27: error: 'class RA8875' has no member named 'touchCalibrated'

   if (!tft.touchCalibrated()) {//already calibrated?

            ^

C:\Program Files (x86)\Arduino\libraries\RA8875-0.70b11\examples\_due\_resistiveTouchScreen\touchTest\touchTest.ino: In function 'void loop()':

touchTest:37: error: 'class RA8875' has no member named 'touched'

   if (tft.touched()) {

           ^

touchTest:39: error: 'class RA8875' has no member named 'touchReadPixel'

     tft.touchReadPixel(&tx,&ty);//read directly in pixel

         ^

exit status 1
'class RA8875' has no member named 'touchBegin'

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

sumotoy, 2016-02-07:
Sounds like you don't enabled the touch screen.
Did you go to RA8875UserSettings.h and enabled resistive touch screen?

#define USE_RA8875_TOUCH//resistive touch screen
//#define USE_FT5206_TOUCH//capacitive touch screen

T

morpheous87, 2016-02-07:
It's very strange problem, because I'm looking into the library and I see all those methods, that the compiler complains they are missing, what could be the reason?

morpheous87, 2016-02-07:
Omg, yes, the problem was the RA8875UserSettings.h !!! Because I've tried to change it in the libraries folder, somehow I hadn't rights to save the file with the modifications, and I din't notice that - so stupid ...

closed 2016-02-07 by sumotoy

dennischrist, 2016-10-23:
Hi Sumotoy, does this mean I have to set the INT nr in this part:

#if !defined(_AVOID_TOUCHSCREEN)
#define __RA8875ISRINT 0 //// [default 0, pin 2 on Arduino UNO]
#endif

in stead of in the sketch?

I'm a newbie so your help is much appreciated! ;)

D