sumotoy/RA8875 #87: Problem with arduino MEGA2560

cocodev, 2016-01-30:
Hi,
I've got an issue with my arduino MEGA2560 using your library: you see, when I drive my RA8875 with the Adafruit library, all works fine but when I try to drive my 800x480 TFT screen with your library without changing any wiring, nothing apparears on the screen: it remained black.
Thanks in advance for helping

sumotoy, 2016-01-30:
Just connected my 7" to a mega compatible CPU and works, many users has reported works as well so it should work for you.
You have to provide more informations if you need assistence!

cocodev, 2016-01-31:
Ok, everythings works fine now! I just forgotten to replace 'RA8875_800x480' by 'Adafruit_800x480' in the setup function. I'm so excited to use your library now! Sorry and thanks to reply!

closed 2016-01-31 by cocodev