sumotoy/RA8875 #88: RA8875 touch issue when using SDfat library

Qdeathstar, 2016-02-07:
Hello, 

When using both the SDfat library and the Sumotoy RA8875 library, the touchReadAdc function returns 1 no matter where i press on the screen.  If i edit out SD.begin the touch will work.

Also, i have tested by screen using Adafruit's library and the default SD library and touch works fine with those libraries.

Here is my code:

[ra8875_bitmapSumotoy 2.zip](https://github.com/sumotoy/RA8875/files/120615/ra8875_bitmapSumotoy.2.zip)

sumotoy, 2016-02-07:
1) You read wiki?
2) You isolated the RA8875 with the extra circuit described in wiki?

Qdeathstar, 2016-02-07:
yes.

I have pull up resistors on the CS lines for the SD/Wifi shield.

I also have an shift register installed to isolate the ra8875

I set it up like this:

http://www.dorkbotpdx.org/blog/paul/better_spi_bus_design_in_3_steps

I didnt have 10k resistors for the test of the miso line, so I used 8.5k resistors and I got 2.2. on the miso line

sumotoy, 2016-02-07:
No, wrong.
Read my wiki please....

Qdeathstar, 2016-02-07:
sorry I i wasn't, clear, I have the ra8875 like this:

https://github.com/sumotoy/RA8875/wiki/Fix-compatibility-with-other-SPI-devices

sumotoy, 2016-02-07:
You talking about a resistor on miso line.... That's strange to me, don't have any resistor on that line, only the CS has to be pullup.
If you are using an SD you need to pullup the CS for the SD as well.
So you got 2.2 of what? voltage?

Qdeathstar, 2016-02-07:
Sorry, I'm a newbie at electronics.

The resistor on the MISO line was only there to test if MISO was being tristate correctly. I get 2.2 volts when neither the SD card or the RA8875 is "working" so i'm assuming the miso tristate behavior is correct. I have removed the resistors after verifying that behavior.

I have pull up restores on the CS line for the SD. The display works (the image can be loaded on to the screen, but 

readTouchAdc(&tx, &ty);
Serial.println(tx);

prints 1.

I FIXED that problem, using your hints. I thought i read somewhere it said to use the SDfat library, but actually i am supposed to paul's SD library. I now am able to get the correct signals from readTouchAdc, but only if i slow the teensy down to 24mhz. 

Is that because i am using a crappy shield and breadboard? (Keys C3300 Wifishield w/SD card)

sumotoy, 2016-02-07:
Yes, SD cards have weak signals and it can be a nightmare work with them.
As I wroted in wiki, really short connections, very clean assembly and high quality SD holder (NOT the chinese crappy stuff or SD holders with resistors or level changer mounted).
Also use branded and high quality SD cards.
When it works by speeding down SPI would mean that you have missed something as described above...

closed 2016-02-07 by sumotoy