sumotoy/RA8875 #90: Issue with Pattern example?

Qdeathstar, 2016-02-09:
Hello,

I am trying to figure out what the pattern function does. According to the RA8875 data sheet, you can set a pattern, then display it repeatedly over a certain area that you set. However, i can't figure out how do it in your example.

[code]
/*
Just a basic example, an hello world example.
*/

#include <SPI.h>
#include <RA8875.h>

#include "pattern16x16.h"

#define RA8875_CS 10
#define RA8875_RESET 20//any pin or 255 to disable it!

RA8875 tft = RA8875(RA8875_CS, RA8875_RESET);

void setup()
{
  Serial.begin(38400);
  //long unsigned debug_start = millis ();
  //while (!Serial && ((millis () - debug_start) <= 5000)) ;
  Serial.println("RA8875 start");

  tft.begin(Adafruit_800x480);
  tft.fillWindow(RA8875_BLUE);
  tft.writePattern(100, 100, _pattern16x16, 16, false); //this write a pattern 16x16 at 100,100
  tft.BTE_layer(SOURCE, 1);
  tft.BTE_layer(DEST, 1);
  tft.BTE_moveFrom(100, 100);
  tft.BTE_moveTo(300, 300);
  tft.BTE_size(16, 16);
  tft.setTransparentColor(0xF800);
  tft.BTE_ropcode(0xC5);
  tft.BTE_enable(true);

  tft.BTE_layer(SOURCE, 1);
  tft.BTE_layer(DEST, 1);
  tft.BTE_moveFrom(100, 100);
  tft.BTE_moveTo(400, 800);
  tft.BTE_size(16, 16);
  tft.setTransparentColor(0xF800);
  tft.BTE_ropcode(0xC4);
  tft.BTE_enable(true);
}

void loop()
{

}

[/code]

and 

[code]
/*
Just a basic example, an hello world example.
*/

#include <SPI.h>
#include <RA8875.h>

#include "pattern16x16.h"

#define RA8875_CS 10
#define RA8875_RESET 20//any pin or 255 to disable it!

RA8875 tft = RA8875(RA8875_CS, RA8875_RESET);

void setup()
{
  Serial.begin(38400);
  //long unsigned debug_start = millis ();
  //while (!Serial && ((millis () - debug_start) <= 5000)) ;
  Serial.println("RA8875 start");

  tft.begin(Adafruit_800x480);
  tft.fillWindow(RA8875_BLUE);
  tft.writePattern(100, 100, _pattern16x16, 16, false); //this write a pattern 16x16 at 100,100
}

void loop()
{

}
[/code]

produce the same affect on my screen, like the BTE functions aren't doing anything..

sumotoy, 2016-02-09:
They both print the pattern 16x16 at 100,100 once, as suppose, correctly, both tested a minute ago.
How to fill the screen with the pattern? Well, you have to figure out, I just wroted the library and commands and provided some examples.
This is not an issue, sorry...

closed 2016-02-09 by sumotoy

Qdeathstar, 2016-02-09:
I figured that out, but why does the bte code have no affect on the sketch?  Why include it in the example if it doesn't do anything.

sumotoy, 2016-02-09:
try this...

```
/*
Just a basic example, an hello world example.
*/

#include <SPI.h>
#include <RA8875.h>

#include "pattern16x16.h"

#define RA8875_CS 10
#define RA8875_RESET 8//any pin or 255 to disable it!

RA8875 tft = RA8875(RA8875_CS, RA8875_RESET);

void setup()
{
  Serial.begin(38400);
  //long unsigned debug_start = millis ();
  //while (!Serial && ((millis () - debug_start) <= 5000)) ;
  Serial.println("RA8875 start");
  tft.begin(RA8875_800x480);

}

void loop()
{
  tft.clearScreen(RA8875_BLUE);
  tft.writePattern(100, 100, _pattern16x16, 16, false); //this write a pattern 16x16 at 100,100
  tft.BTE_move(100, 100, 16, 16, 200, 200, 1, 1, true, 0xC5);
  delay(1000);
  tft.clearScreen(RA8875_BLUE);
  tft.writePattern(200, 200, _pattern16x16, 16, false);
  tft.BTE_move(200, 200, 16, 16, 300, 300, 1, 1, true, 0xC4);
  delay(1000);
}
```

BTE has been updated an some example it's old

Qdeathstar, 2016-02-09:
Thank you, that works much better