sumotoy/RA8875 #91: LCD Image converter tutorial needed

morpheous87, 2016-02-09:
Hi, you mentioned somewhere that you are going to make a wiki for image convert. I've tried lots of things, but I ended up with images converted, but in wrong colors. For example blue image was displayed in red/orange. I hadn't problems with green but everything is strange, I'm using one settings and the different colors are converted different.

A short wiki will be very, very helpful.

Regrads

sumotoy, 2016-02-09:
It takes time since the template it's the same of the icon and has to be tuned but if you are in the hurry...
1) Copy the new sumotoy_image.tmpl from _utility folder inside lcd-image-converter folder.
2) launch lcd-image-converter
3) Menu File->Open
4) On bottom right switch from xlm to Images and select your image (24bit)
5) Menu Options->Conversion
Tag Prepare:set color, Top To Bottom, Forward
Tag Matrix: Select Color
Tag reordering: do not touch
Tag Image: Block Size 24Bit, Little Endian
Tag Font: do not change
Tag Templates: for image select the template you put in the folder before. You may need to restart the app if the whole thing do not work, not sure if template it's active immediately.
6) Now click ok, save changes.
7) Menu File->Convert

You have now a file yourImage.c open it
Inside there's a couple of arrays, first one it's the image converted, second one has parameters needed for display it.

static const uint32_t image_data_yourimage[32640]   = {
xxxxx.xxxx.x.x.x..x.x.x..x
};// 24

copy this one inside load24bitImages sketch

You need to modify drawArray accordly
drawArray(image_data_yourimage,yourimageArraySize,yourimageWidth,tft.width()/2-yourimageWidth/2,tft.height()/2-yourimageHeight/2);

Be in mind that this use tons of memory so do NOT convert large images, it will simply do not work.
I'm not reacheable this week, gonna go outside country for work.

morpheous87, 2016-02-10:
Great, that step-by-step worked! I successfully converted an image and it was properly displayed. One more question - because I still wait my separate SD card board (didn't try to get alive the on on the display board), I thought that those image that I'm testing  with the LCD converter will appear the same way if I read them from the SD?

Qdeathstar, 2016-02-14:
I'm pretty sure (from my attempts) it is very slow and not worth it. It takes about 14 seconds to load an image from the sd onto the screen, and each time you want to change the text on the screen you have to redraw the screen... Unless you use layers which means you'll have an image using only a 256 color rgb pallet, which still takes a long time to load.

You'd have to use 8bit images to do anything useful, eg, aside from show a picture... So I ended up ditching the idea.

sumotoy, 2016-02-16:
Old story...If you read wiki I already said this. The RA8875 it's not a framebuffer chip, it's more a controller and not suitable for loading images, even send at 8 bit will not change the speed singe it accept only 16 bit stream (and internally the RA8875 will convert in seamless 16 bit).
Storing images in a SPI Memory chip and let RA8875 read from his second SPI dedicated channel with DMA it's much faster but you have to program SPI chip in advance, bla, bla, bla....