sumotoy/RA8875 #92: help wanted:  fillRectangle() with gradient and strange pattern behavior?

Qdeathstar, 2016-02-12:
My first question is if there is a function to fill rectangles with gradiant?

The second question is that, after setting a pattern, in order to dragon the screen you have to re-set the active window to the whole size, otherwise it will only be the size of the last pattern... is that how it is supposed to work?

sumotoy, 2016-02-20:
fillRectangle it's an hardware accellerated function, it's possible to build another function that uses grandient but will be much slower, I will see what I can do in next version.
About pattern, this is a good question, you are right, the function need to use the active window so in theory I can reset the active window after the function, but if you are using the draw pattern repetively this will impact performances a lot, I will look into this since pattern stuff was developed from datasheet and never really tested deeply.

Qdeathstar, 2016-02-20:
Your function would be something like drawing a series of lines instead of a rectangle, correct?

As gas as the fill pattern, I thought the ra8875 data sheet said you could only use four patterns at a time?

morpheous87, 2016-03-08:
Is there a fast way to draw a button or icon with gradient? Something like RoundRectangle() with gradient? Something like this
 [image]

Or it will be better to use the new icon functionality, which will be introduced soon?

sumotoy, 2016-03-09:
As I said, rectangles are hardware based, this is why they are fast rendered. It's possible create another function that uses lines but it will be much (a lot!) slower, the hardware parameters doesn't allow grandients.

Jim-McDonald, 2016-03-10:
> RAiO will probably release a new chip in the near future that have same RA8875 functionalities but faster and rumors of grandients, more internal memory and antialiasing(!) seems confirmed..

Any link to this upcoming controller please? :)

sumotoy, 2016-03-11:
Just rumors, talking between technicians. Btw it's long time RAiO don't have new products so I suppose this will happen soon.

sumotoy, 2016-03-11:
Qdeathstar,
even pattern on RA8875 are in a way limited, first it's memory, they use part of the internal ram used for everithing, second problem is they use the part of the internal engine that deal with layers and layers don't work at 100% if your screen it's more than 272x480 because in the real world more ram it's needed. At RAiO they do miracles to fit everithing inside such a small ram but at the end shit happens so there's several problems when using large screen.
In the end, yes, pattern have limited memory slots but don't use together layers or shit happens!