sumotoy/RA8875 #96: Slight Rotation / Pixels Off

wzinc, 2016-02-29:
I'm having an issue with the RA8875 driver. The screen is slightly rotated and this makes the pixels appear to skew with strange colors. The pixels are in pieces, with their RGB components rendering in adjacent pixels. This issue is not present with the Adafruit driver - text and graphics render fine and there is no slight rotation. Even so, because of the suggestion on this page (https://github.com/sumotoy/RA8875/wiki/About-RA8875-SPI-Speed), I rewired the RA8875 straight to my Arduino UNO, bypassing my breadboard - no luck. I messed with pixel clock / SPI speeds to match the Adafruit driver with no success, but that was just a wild guess. I don't really know where to begin to fix this issue; I'd appreciate it if someone could point me in the right direction. Again, this is not present in the Adafruit driver; that one works for some reason...

Look at the edges of the screen; you can see a very small clockwise rotation:
[image]

The effect is much more evident with text on the screen:
[image]

Thanks!

sumotoy, 2016-02-29:
Looks like a faulty screen! Just to be sure, what happen if you use the Adafruit RA8875 library? The RA8875 create all the graphics inside and has it's own memory, it's impossible to me recreate a screen slight rotated, the only way it's write pixel per pixel inside it's ram and would take a lot of time, this mean that something defective inside or the flat cable that connect the screen it was soldered offsetted.

sumotoy, 2016-02-29:
Ok, this not happen with Adafruit, I see... Well this intrigue me.
What version of the library are you using? Let see if I can solve this or at list know what trigger the problem.
For sure there's a kind of hardware defect triggered by something I would like to discover.
Another question. how you power up the screen? Can you measure the power voltage during all startup?

wzinc, 2016-03-01:
Thank you for getting back to me. I'm using RA8875-0.70b11.

Here's a screenshot of the display using the Adafruit driver:
[image]
Everything seems to be working on with this one, so I don't think it's a hardware issue. All I could think of is SPI speed or type or pixel-clock divisions.

I am powering the screen with the Arduino UNO 5v and GND. Here's the power on startup:
[image]
I powered it on twice and off in this image.

sumotoy, 2016-03-01:
1) The screenshot for power it was taken using my library or adafruit?
2) I cannot recognize the display. Please give me all possible info about that.
3) If screenshot was taken with Adafruit, please use my library and measure till display shows graphic.

Acctually hundreds of people it's using the library in the current form and never report something like that, you are also using UNO, this mean SPI speed at 8Mhz, pretty slow for this chip, should be almost the same speed of Adafruit more or less, I've tested at list 10 different displays at 20Mhz and never a problem!
This is why I would like to investigate.

wzinc, 2016-03-02:
Here's the front and back of the screen:
[image]
[image]

I ordered them from Adafruit:
board: https://www.adafruit.com/products/1590
screen: https://www.adafruit.com/products/1591

Here's the power on / off with the Adafruit driver:
[image]

And on / off with your driver (I don't think there's much difference):
[image]

Thank you again for your time. Let me know if you want me to try anything else.

sumotoy, 2016-03-02:
Ok, maybe I have an idea, the RA8875 receive initialization parameters about how to deal with the screen during startup, in these parameters there's a lot of timing critical parameters since RA8875 receive SPI command macros but in short it deals directly with the screen.
I think that this screen needs slight different parameters in the initialization, this has nothing to do with SPI speed...

sumotoy, 2016-03-04:
Sorry for delay, I was quite busy. Later in the day I will try to prepare a variation for your display, keep in touch to see if this will fix the problem

wzinc, 2016-03-04:
Sure, thank you for your time. I am also thinking of getting a display from buy display.com based on your recommendation: https://github.com/sumotoy/RA8875/wiki/Various-RA8875-based-boards-tested-with-this-library

sumotoy, 2016-03-04:
Ok, this will work, btw I have also a couple of adafruit and both working but display it's a bit different, maybe they change model during the years?
One raccomandation, buydisplay recently changed display as well, several users report that display has the capacitive touch flat cable that is actually loose (before was soldered on the board and easy to access). This force to buy some kind of adaptor (see https://forum.pjrc.com/threads/30993-RA8875-library). I don't know if still selling this display variation so please check yourself!

wzinc, 2016-03-07:
Good news; I figured it out. It was the second phase register, RA8875_PLLC2, well RA8875_PLLC1+1 in your driver.

Docs (p. 39): https://www.adafruit.com/datasheets/RA8875_DS_V19_Eng.pdf

[RA8875.cpp:560](https://github.com/sumotoy/RA8875/blob/0.70/RA8875.cpp#L560):
`_setSysClock(sysClockPar[_initIndex][0],initStrings[_initIndex][1],initStrings[_initIndex][2]);`

The second value needs to be 0x02. When it gets down to RA8875::_setSysClock, that value is fed into RA8875_PLLC1+1. FYI, Adafruit sets their PLL2 to 0x02 (RA8875_PLLC2_DIV4 in [Adafruit_RA8875-master/Adafruit_RA8875.h](https://github.com/adafruit/Adafruit_RA8875/blob/master/Adafruit_RA8875.h#L193));

0x00 - blank screen
0x01 - strange rotation; original problem
0x02 - **IT WORKS!** (see screenshots below)
0x03 - also works
0x04 - blank screen
0x05 - blank screen
0x06 - blank screen
0x07 - blank screen

I changed [sysClockPar](https://github.com/sumotoy/RA8875/blob/0.70/_settings/RA8875Registers.h#L59) to:
```
const static uint8_t sysClockPar[3][2] = {
	//{0x0B,0x01},//0 -> 320x240		->
    //{0x0B,0x01},//1 -> 480x272		    -> 0
    {0x0B,0x02},//1 -> 480x272		    -> 0
    //       ^
	//{0x0B,0x01},//2 -> 640x480			
	{0x0B,0x01},//3 -> 800x480			-> 1
	{0x0B,0x01} //4 -> 800x480_ALT		-> 2
	};
```

[image]
[image]

I don't know what effect this may have on drawing, etc. Let me know if there's anything you'd like me to try. Thank you again for your time.

sumotoy, 2016-03-07:
Oh, good you get it! I was preparing a new beta with the same changes (and small changes as weel plus checking Mr TOM observations).
So at the end was just a parameter.
This will not impact anithing, it's just an instruction that tells RA8875 how to deal with the display, the parameter you change it's ok, for most display the original parameter it's ok but probably this is older and need the proposed change. I will include an option in the upcoming release for dealing with this.
What about the rest? Everithing shows as it should or you gest any other problems?

wzinc, 2016-03-07:
The 3D solid example for UNO just shows a color-changing dot in the upper left of the screen, but that may be meant for an 800x480 screen. I'm also having an issue with the internal touch screen driver. It seems to think the screen is rotated, and it puts the touches at a 90° angle to where you tap. Other than simple calibration, I haven't tried to solve it yet. I'll adjust some params and let you know if I find any other issues. It could just be my setup. I did order an 800x480 screen from BuyDisplay.

sumotoy, 2016-03-07:
The touch screen can be inverted in your screen, It happened before. I can easy add a user setting for that.
The 3D example normally don't work at all with UNO, I'm surprised you get compiled, this caused by an error in the math 2D->3D transformation. I have another example that should work.
Try the benchmark and see if everithng it's showed correctly, it uses all the primitives and if there's any error you should note it.

wzinc, 2016-03-08:
FYI, the [benchmark example](https://github.com/sumotoy/RA8875/blob/0.70/Examples/_UNO/benchmarks/benchmarks.ino) is missing a lot of code, but I was able to get it out of a [previous revision](https://github.com/sumotoy/RA8875/blob/013360bfae394ee08a33707c7a84f89a2447d7b8/Examples/_UNO/benchmarks/tests.ino). Everything worked ok.

I'm going to work on the touchscreen now.

benchmark.ino output:
```
screen:480x272 - rotation:0
Benchmark                Time (microseconds)
Screen fill              5276
Test Pixel               216
Test Pixels              81168
Text                     17584
Lines                    66116
Horiz/Vert Lines         61132
Rectangles (outline)     55164
Rectangles (filled)      55036
Circles (filled)         114820
Circles (outline)        113928
Triangles (outline)      17520
Triangles (filled)       26280
Rounded rects (outline)  191288
Rounded rects (filled)   275148
--------------------------------

screen:272x480 - rotation:1
Benchmark                Time (microseconds)
Screen fill              5364
Test Pixel               208
Test Pixels              45996
Text                     17572
Lines                    66032
Horiz/Vert Lines         61216
Rectangles (outline)     55152
Rectangles (filled)      55064
Circles (filled)         113264
Circles (outline)        113368
Triangles (outline)      16672
Triangles (filled)       25232
Rounded rects (outline)  191496
Rounded rects (filled)   275244
--------------------------------

screen:480x272 - rotation:2
Benchmark                Time (microseconds)
Screen fill              5360
Test Pixel               208
Test Pixels              81176
Text                     17588
Lines                    66120
Horiz/Vert Lines         61148
Rectangles (outline)     55048
Rectangles (filled)      55064
Circles (filled)         114780
Circles (outline)        113904
Triangles (outline)      17560
Triangles (filled)       26256
Rounded rects (outline)  191408
Rounded rects (filled)   275108
--------------------------------

screen:272x480 - rotation:3
Benchmark                Time (microseconds)
Screen fill              5356
Test Pixel               200
Test Pixels              46004
Text                     17568
Lines                    66304
Horiz/Vert Lines         61220
Rectangles (outline)     55180
Rectangles (filled)      55132
Circles (filled)         113348
Circles (outline)        113360
Triangles (outline)      16640
Triangles (filled)       25244
Rounded rects (outline)  191472
Rounded rects (filled)   275008
```

sumotoy, 2016-03-09:
Try last beta, I have added a new option in user settings for dealing with display like yours.

closed 2016-03-09 by sumotoy

wzinc, 2016-03-10:
Swap the include order in [RA8875.h:190](https://github.com/sumotoy/RA8875/blob/0.79b11p10/RA8875.h#L190):

```
/* ----------------------------DO NOT TOUCH ANITHING FROM HERE ------------------------*/

#include "_includes/font.h"
#include "_includes/RA8875ColorPresets.h"
#include "_settings/RA8875UserSettings.h"
#include "_includes/RA8875Registers.h"
```

The USE_ADAFRUIT_PHASE_FIXUP flag isn't being respected.

sumotoy, 2016-03-10:
Oops! My bad. Fixed