sumotoy/RA8875 #105: Please Document "roundGaugeTicker" Function

sbfreddie, 2016-05-14:
Max:
This Function has 7 variables passed into it.
Please document the meaning of the the variables passed into this function like you have documented the other functions.  Please not not scrimp on the words, the more information the better.

If you would like, I would like to help you with the documentation of the individual functions contained in this very valuable display driver.

Thanks,
Ed

sumotoy, 2016-05-15:
Sure!

sbfreddie, 2016-10-05:
Max:
Hello again, this is Ed.
Can you please explain to me what these 7 variables are for.
roundGaugeTicker(uint16_t x, uint16_t y, uint16_t r, int from, int to, float dev,uint16_t color)
x = ?
y = ?
r = ?
from = ?
to = ?
dev = ?
color = The color to use for the ticks.