sumotoy/RA8875 #109: ER-AS-RA8875 + UNO + 9" 800x480 ER-TFTM090-2 buy Display

JamiePoc, 2016-07-03:
Please help
I have the following
ER-AS-RA8875 Shield
[image]

UNO 

9" 800x480 ER-TFTM090-2 buy Display
[http://www.buydisplay.com/default/serial-spi-arduino-9-inch-tft-lcd-touch-shield-ra8875-for-mega-due-uno](url)
[image]

I am running RA8875-0.70b11
I have been working on this for a week now and still can not get anything on the display.

Any assistance / suggestions/ pointers would be greatly appreciated, I am now at the point where the whole lot is going to be in the bin.

sumotoy, 2016-07-04:
Gasp....
The display is a 800x480 and should be supported but the board they added is something totally new to me. 
RA8875 based display can be frustrating, if not inited they just remain black and no life sign since all is controlled by display driver, background included.
I have ordered one of these but I may take a look on adapter board jumpers (since it have a lot) in the next days, I'm actually outside home, the piggy board, at first look, it's just a level adapter, useful for UNO, useless for DUE, on the board I've see another regulator, It looks a 5->3v3 but I don't know if used for the logic board or even for display.

Remember that these display's consume A LOT of current and for sure UNO is not able to power this display by using USB, if you are using an external supply it's ok, be sure can supply enough current!

For start I would do this:
1) Ensure that display is in 4 SPI pin mode
2) Ensure display is set for 5V supply, the internal regulator sometimes drops too much when supply is set at 3V3
3) Check if backlight is controlled by RA8875 (it must be like this)
4) Check with tester all supplies, some users experienced an amazing voltage drop when backlight try to turn on and since the backlight is controlled by RA8875 the result is a black screen since chip won't start up.
Still not work?.....
I will use an external level converter (like CD4050) and connect UNO trough this converter directly to the display, be extra careful when you do this, display logic is limited to 3v3 signals, more than that will damage it.
As soon I have some spare time I can look to the piggy board circuit and the damn tons of jumpers.

JamiePoc, 2016-07-06:
Thank you for the reply, yes all checks out, the seperate PSU I am using is from a mobile phone running at 5v with a max current of 2.4A so should be fine.
I do think the problem is in the new shield but I can not work out what's wrong.
I would love to get this running and I am spending all my spare time trying, if I do get somthing I will post here, I will check back every couple of days to see how you have got on with yours.

May simple be a faulty board, so if you can get yours running and I do the same and mine still does not work then I will know this to be the case.

Thanks again.

sumotoy, 2016-07-06:
It takes time to arrive from china!
However, the best thing you can do is double check everithing, arm with patience and try without the piggy board.
RA8875 looks dead until it get the correct signals, I have at list 10 different display's here, all working, never seen one faulty but I always take care to avoid 5v signals into RA8875 logic.

242bpm, 2016-07-16:
I have the same shield and got my 7" eastrising/buydisplay TFT (except for the capacitive touch panel) running with Uno and Mega. I just edited Pins like this:

#define RA8875_CS 10 
#define RA8875_RESET 9

// capacitive touch panel ????
// tried so many things but didn't got it yet ... i keep trying ;-)
// also tried a spare copy of the TFT from different production run - no luck
// i assume it sould be:
#define RA8875_INT 7

StephaneRod, 2016-09-15:
I got the same problem as Jamiepoc
I have a 8" TFT + RA8875 
I tried it with 
1) an arduino uno from arduino.org and a usb cable : works fine
2) an arduino uno from arduino.org and a 5V 3A power supply connected to the UNO : black screen
I tried it with an arduino uno WIFI from arduino.org and usb for uno : works 3s and stop. screen is black
I tried it with a uno copy and USB  : works 3s and Stop. Screen becomes black

Sumotoy, Did you had a chance to get your board and lake it run ?
 JamiePoc, did you make your board running ?
Thanks

dennischrist, 2016-10-23:
Hi everyone,

I bought the 8inch resistive touchscreen from buydisplay with the shield. For me it works with these settings and their own ER-AS-RA8875 library :

//Arduino DUE,Arduino mega2560
#define RA8875_INT 4 
#define RA8875_CS 10 
#define RA8875_RESET 9

When I throw out the buydisplay library and use your RA8875 library, the screen works, but touch doesn't. No matter what pin I try for INT, touch will not work.

Any ideas?

arunnad007, 2016-11-01:
@dennischrist ,facing  same problem
Did you find the solution???
Error:  tft not declared in this scope
          'tempData' was not declared in this scope

Are you getting the same errors??

arunnad007, 2016-11-01:
@everyone

Any help will be appreciated guys, tq

JamiePoc, 2016-11-01:
Hi
The solution was to use two PSU's one powering the Ardunio, the second the monitor via the power connector on the side of the TFT. Now works 100% of the time.

FranzEckert, 2017-10-24:
I just got this shield and tft screen too, i will hook up a much more powerfull powersuply tomorrow. 
But, can somebody guide me to the libraries needed, i have the Arduino Uno and a Mega. It seems the UNO says the libraries from the seller on Ebay are not working.
To sumotoy, how do i do step 1 to 3?? Do i just connect the shield to the Uno and run some code?
To Jamiepoc, where did you connect the power to the tft?

DereckWildstar, 2018-03-16:
@dennischrist , 
Hy dennischrist, did you find solution about touch ? i have same problem!!
thanks for your help
Dereck

GayanDissanayaka, 2019-06-05:
bought the 7inch RoHS ER-TFTM070-5 ver2.0  touchscreen with the shield. 

the screen works, but touch doesn't. 
pleas help

wb0wuq, 2019-06-05:
Is it the resistive or capacitive touch screen?

Tim

GayanDissanayaka, 2019-06-06:
thank for your consideration.
It is  resistivite touch screen

gcharles81, 2019-06-06:
You can use any 3rd party library for your touchscreen

Regards
Charles

filou331ci, 2023-08-15:
Hi
I am sorry , very old post , I have the same shield ES-RA-8875 from buydisplay with a Mega 2560 rev 3.
[https://www.buydisplay.com/7-inch-1024x600-tft-touch-shield-for-arduino-capacitive-touch-screen](url) 1024x600 LCD capacitive touch
The shield and the Library are for Due ... 
Trying to make screen and shield working on Mega , their Ra8876 Library is out of use ... 
The Mega even does'nt boot.
Thank you for all advices about Libraries to try for Uno/Mega
I have had a mail discussion with support , they answer rapidly and help to solve the issue , although their website products description is not correct ...
They send me free an ER-AS-5517 shield for Uno/Mega with Library.
You will not find it , it is only sale on bundles, but the doc is here : [https://www.buydisplay.com/download/manual/ER-AS-5517_Datasheet.pdf](url)
But my first goal is to use RA8875 shield ... 
Thank you for your help , I am not abble to write Libraries , just sometimes to edit them successfully .
Regards
Philippe from France
Apologize for my english ;)

trevorbattle, 2023-08-15:
Hello / Bonjour Philippe!

It has been forever since I've worked with my boards, but I know for sure that I used an RA8875 from buydisplay with an Arduino MEGA. 

My screen was much smaller and simpler though, but it worked. 

Does the MEGA boot on its own? If it does, but does not boot when the screen is connected, it must be the wiring is wrong.

The screen should not stop the MEGA from booting unless it is disturbing pins the MEGA needs. It should only be connected to SPI pins.

filou331ci, 2023-08-15:
Hi Trevor
Thank you very much to take care of my issue.
I have an Adafruit RA8875 driver for adafruit 7"tft 800x480 screen, no problem , works fine, nearly all my sketches are written with it.
For the East Rising 8875 shield , I check the pins between Mega and Due no more differences , excepting the default pins for SPI are not the same Mega/Due , the SCLK divider idem , the SPI code idem and so on ...
I am short in time now , but in september I will have time to work on it.
My project is to do an Engine controller for my Mustang 1965 Fastback. I can do it with 800x400 résistive touch screen Adafruit , but east rising screen is so nice ... 
[https://www.trechph.ovh/images/mustang/IMG_6080.jpeg](url)
[https://www.trechph.ovh/images/mustang/IMG_6080.jpeg](url)