sumotoy/RA8875 #110: How to use the ISR?

ShockDog, 2016-07-19:
Apologies if this is a ridiculous noob question... I am a bit lost regarding the ISR functionality. As well as including:

  tft.useINT(RA8875_INT);
  tft.touchBegin();
  tft.enableISR(true);

I also added the following:

  attachInterrupt(digitalPinToInterrupt(RA8875_INT), touchISR, CHANGE);

The action that I want to occur on a touch is put in touchISR(). Is this the best way to do this? Or is there some other way to specify the ISR? If not, then what is the purpose of the tft.touchBegin(); and tft.enableISR(true); lines?

sumotoy, 2016-07-20:
> I also added the following:

> attachInterrupt(digitalPinToInterrupt(RA8875_INT), touchISR, CHANGE);

This is the problem, I'm currently handle interrupts internally and you write your own interrupt routine externally?!
Why?!
I'm handle internally because, after int has been triggered, some code conditions have to be performed fast, like reading registers, clear flags, write results in variables and rearm registers and interrupt.
If you want to handle interrupt separately you have to do externally all this stuff using a right sequence or your RA8875 will be stuck.
There's some examples about this in library, maybe I can comment better....

ShockDog, 2016-08-09:
Thanks for the reply. Apologies, I completely missed the example utilising the internally handled interrupt! Now I have the basics down I will play around with it.