sumotoy/RA8875 #112: Touch being disabled after tft.print

Comiter, 2016-07-30:
Hi all,
I'm in the process to write some code that uses one of the BuyDisplays with RA8875 controller, running on a Arduino Pro Mini with Atmega 328 running on 5v, 16MHz.
The program I am trying to write is basically a calibration program, where the user are supposed to push on red dots on the screen, while the program takes average readings of the ADC value of the RA8875. A total of 9 dots are collected. The problem I have, is that for some reason the touch is disabled after I have written some text to the Display, and I'm not able to re-enable it by code (touchEnable(true) ). If I remove the lines writing instructions and data (text) to the user, the program works fine.

So, the question is: Is there any requirements for delay between commands that is not handled by the library?  I'm currently using v0.79b11p10. I have also connected a 10k pullup on the INT pin.

Best Regards
Vidar

Comiter, 2016-07-30:
Seems like I figured out the issue.
According to [this post](https://github.com/sumotoy/RA8875/issues/51), the RA8875 need to be in graphic mode to work?

Anyhow, adding the following code to the end of void RA8875::_textWrite(const char* buffer, uint16_t len) worked for me:

`#if defined(USE_RA8875_TOUCH)
	  if(!_textMode )
	  _setTextMode(false);//  go to graphic
	#endif`

sumotoy, 2016-08-04:
Mmm, good catch, I'm gonna check this week!

ukemina, 2016-09-27:
Hello 
I'm the same problem.
I try to write in RA8875.h file the #if defined...... but i have the next erro when compiled my scketch in arduino mega 2560 

Arduino:1.6.13 Hourly Build 2016/09/23 03:33 (Windows 10), Tarjeta:"Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from C:\UGE\ARDUINO\CONMUTADOR\CONMUTADOR.ino:243:0:

C:\Users\yo\Documents\Arduino\libraries\RA8875-0.79b11p10/RA8875.h:604:3: error: expected unqualified-id before 'if'

   if(_textMode ) _setTextMode(false);// go to graphic 

   ^

exit status 1
Error compilación en tarjeta Arduino/Genuino Mega or Mega 2560.

Can you help me?
Thank you

ukemina, 2016-09-27:
i found a solution for work ok.
enable the touch ,read pixel. disableb the touch, print tft and finally enable again the touch, and work fine.