sumotoy/RA8875 #113: Flicker when updating pointer

RuneKyndal, 2016-08-18:
Hi guys.    I'm using this library to make a screen with "analog looking" gauges

my pointer is made from a fillTriangle at the tip. and a fillQuad for the body
based on the Cos/sin helper functions to define the coordinates.
and its working fine. with a nice sweeping "fat" needle.

but I'm getting flicker when i redraw my needle.
I'm using Layer 2 as the gauge outlines, Ticks, writing  and am only updating
a few items on layer 1   
I draw the needle in white.    then  Draw it again in "black / transparant"
then update the needle angle.   Draw it again

i have tried different combinations of color / redraw transparency etc.  and I cant get rid of  the redraw flicker?

> tft.writeTo(L1);
>//ERASE pointer tip
> tft.fillTriangle(pointer_Ax,pointer_Ay,pointer_Bx,pointer_By,pointer_Cx,pointer_Cy,0xFFFFFE); 
>//ERASE pointer body
> tft.fillQuad(pointer_Bx, pointer_By, pointer_Cx, pointer_Cy, pointer_Dx, pointer_Dy, pointer_Ex, pointer_Ey, 0xFFFFFE, 1);  

> pointer_update();

>//DRAW pointer tiP
>tft.fillTriangle(pointer_Ax,pointer_Ay,pointer_Bx,pointer_By,pointer_Cx,pointer_Cy,RA8875_WHITE);
>//DRAW pointer body
> tft.fillQuad(pointer_Bx, pointer_By, pointer_Cx, pointer_Cy, pointer_Dx, pointer_Dy, pointer_Ex, pointer_Ey, RA8875_WHITE, 1); 

any suggestions ?  ideas would be greatly appreciated

/Rune Kyndal

darmfield, 2023-03-08:
Did you ever figure something out for this?

RuneKyndal, 2023-03-08:
unfortunately i did not.   
I have not worked with the LCD in several years now.