sumotoy/RA8875 #115: Teensy 3.5 & Teensy 3.6 do not work at clock speeds above 96MHz

sbfreddie, 2016-10-12:
Using Teensy 3.5 clocked above 96 MHz does not work, I suspected you will need to adjust the SPI clock for the higher clock speeds of Teensy 3.5 and Teensy 3.6.

Thanks,
Ed

bossredman, 2016-11-25:
Hi,

I'm currently trying to get my project to complie with a 3.6 -but it fails for compatiblilty issues with the `RA8875 tft = RA8875(RA8875_CS2, RA8875_RESET);` line.

I see the library only states support for TEENSY 3, TEENSY 3.1, TEENSY 3.2, TEENSY LC and so looks like the library detects it as a Generic Arm.

Did you have have to mod the library in any way to get yours to work?

sbfreddie, 2017-01-04:
I don't have any problems compiling RA8875 with the Teensy 3.5 or the 3.6's.  I use embedXcode with Xcode on my Mac's.  There is a problem trying to build the fonts as the Xcode compiler is different from the compiler used in Arduino.  I just remove the font folder and it builds with no errors.  Perhaps you are not using his latest branch: 

0.70b11p10 changes...

Added initial support for incoming Teensy's
Some changes from MrTom accepted, should fix an hardware bug in trianles and some other minor things
Created _includes folder and moved from settings some file that should never change, this will help users to save their settings between updates.

In June he added the changes for Teensy 3.5 and Teensy 3.6.

Thanks,
Ed