sumotoy/RA8875 #121: 25Q128FV

umutgemisi, 2017-03-12:
i try before show bmp file with arduino shield for er-tftm090 and micro sd reader on it but it was very slow.
25Q128FV  pre programmed with that
[image]
inside 24bit bmp bin file.i used to create bin file with BCB_RA8875_AP_Release from RAIO,  and soldered to ER-TFTM090 from buydisplay tft screen but im searching for one week to find a solution to show image on display 
are there any friend help me about that i need code for arduino uno for call image bin file to show on tft?
Thanks for helps :(

umutgemisi, 2017-03-13:
iste zart diye kalirsin boyle aga

riuson, 2017-03-14:
@umutgemisi, try read [using tft.pushColor to display PROGMEM rgb565 bitmaps](https://forum.arduino.cc/index.php?topic=424921.0)
If you can't find code, just write it.

umutgemisi, 2017-03-14:
thank you for the link but its about show images from sd. i have a 25q128fv chip and a bin file inside it. bin file created with [BCB_RA8875_AP_Release](http://www.raio.com.tw/Data/Datasheet/RA88%20Series/Demo_Program/RA8875_DemoProgram/BCB_RA8875_AP_Release.rar). its bmp to bin converter tool. and is there any way bin image file to show on tft. 25q128fv soldered on [ER-TFTM090](http://www.buydisplay.com/default/serial-spi-arduino-9-inch-tft-lcd-touch-shield-ra8875-for-mega-due-uno) thats more fast way than the sd card

MikeBorman, 2017-03-23:
It is also interesting how to display an image from an external SPI flash memory.
In the datasheet on RA8875 has the function of demand DMA, but how to use it in the library examples, there :-(

umutgemisi, 2017-03-27:
Are there any friend help me about to use flash :)

schuppeste, 2017-03-30:
RA8875::drawFlashImage(int16_t x,int16_t y,int16_t w,int16_t h,uint8_t picnum)

You can write your own Function with "Startadress" instead "pcnum".

umutgemisi, 2017-03-30:
im starting to try :) @MikeBorman  look at here ---> https://github.com/schuppeste/RA8875Ada-BTE

umutgemisi, 2017-03-30:
its work!!! Thank you so much friend you are really helped me :)

schuppeste, 2017-03-31:
Good, I add the hidden Function BTE_MOVE to Copy Data between Layers or  Clippings to Clippings on Screen, without flash.

umutgemisi, 2017-04-01:
it doesnt works on sumotoy's library but adafruit works

schuppeste, 2017-04-03:
I used Sumotoy first, its better and contains more functions, but I did not make it to combined missed functions from Ralinks Example Code.. i changed to Adafruit Library and Rewrite (Fork) the Code.. its only based on Adafruit Library, not the same.

umutgemisi, 2017-04-03:
Is that simple to use with sumotoy yours is very easy drawpicown. Im trying sumotoys drawflashimage but its giving terrible damaged bitmap to tft.

MikeBorman, 2017-05-06:
Thank you, Schuppest!
I did it all :-)
Tell me, how can I increase the data transfer rate from the SPI flash?
Or does the BTE_MOVE function use the full speed of the RA8875 chip?

anujgargiitk, 2018-10-10:
Hello, 
I am doing exactly what umutgemisi did but not getting any image. In fact I burned the winbond w25q128jv chip with all 0s so as to test if I am able to read data using DMA (should display as complete black screen) and the above library by Schuppest. However the data is not getting transferred and I may know the problem but not the solution. The winbond IC require read instruction either 03h or 0Bh or 3Bh depending upon standard SPI, fast read mode and dual SPI before the start address to get in read mode. However I could not find a way to write this instruction from RA8875 DMA to the winbond IC. Any advise would be greatly appreciated.
Best regards
Anuj Garg

anujgargiitk, 2018-10-10:
I have gone through 10-12 datasheets of Flash manufacturers and all of them seem to have same command for read instruction. It is certainly possible that RA8875 DMA mode already have these commands embedded in it and therefore it worked for others. But why I am getting pink screen when I have written the winbond chip with all 0s. In fact I have 2 screens. So the one without any flash IC soldered is also giving the same display indicating that pink color is some default color and flash IC is not sending any data at all.

gcharles81, 2018-10-11:
Let me check my code I make sever changes than I will fork out this branch and check in my changes - again mine works but changes might not be fully compatible btw my code tested on teensy 3

anujgargiitk, 2018-10-11:
Hello Charles, Thanks, but it worked couple of hours back after full night of trials and errors. Basically the problem was the hardware. In BuyDisplay 5inch screen there are 2 set of pads to solder the Font and Flash IC respectively. Previously I was soldering the Flash chip on pads meant for flash  (Chip 1 for RA8875 DMA) and the corresponding 05H port was set to values from 87h to 8Fh (for trial and error but it didnt worked). Now I have soldered the IC to pads meant for Font IC and have changed the corresponding 05h register in the Schuppest library DMA_Block_Move to Fh (to change chip select to chip 0 and Dual Mode 1) and **it worked for small images** but when I tried to read multiple images it started giving very bad noise on the screen (basically while horizontal lines in the image) then I changed the 05h port value to Eh to change to slower speed mode i.e. Dual Mode 0 and it now working like a charm.

So overall my conclusion is that SPI traces on the BuyDisplay screen is not designed properly and have a lot of inductance. Since the traces for Font IC are shorter in length as compared to Flash IC the inductance must be within acceptable tolerance for the Flash/RA8875 IC. Another reason I believe in this theory is that slower speed with Dual Mode 0 eliminated the issue with large image data being read at a time even with font IC pads.

What do you think? Is my conclusion correct or just shear luck?

gcharles81, 2018-10-11:
Hello anuj

I experienced same issue when loading multiple images and i found out that
time that the cursor was not being set correct , I found out I had modified
the dispicown & BTE move function a bit to overcome that now I can load
very fast consecutive images below is the actual code I am using for these
2 Functions after I ported to Sumotoy library , i had to add a GET and SET
X&Y cursor function.

// in file .cpp
void RA8875::dispicown(uint16_t x,uint16_t y, uint16_t w,uint16_t
h,uint64_t start)
{  if (_textMode) _setTextMode(false);//we are in text mode?
 _waitPoll(0xBF,0x01);

 DMA_block_move(x,y,w,h,w,start);

}
///////////////The FLASH reading area   setting
void RA8875::DMA_block_move(uint16_t X,uint16_t Y,uint16_t BWR,uint16_t
BHR,uint16_t SPWR,uint64_t start_address)
{

//Get cursor position
int16_t x7, y7;
getCursor0(x7, y7);
//Get window position
int16_t x1, y1, x2, y2;
getActiveWindow(x1, x2,y1,y2);
  ///Write_Dir(0X06,0X11);//FLASH frequency setting
  // Write_Dir(0X05,0X87);//FLASH setting

    Write_Dir(0X06,0X00);//FLASH frequency setting
    Write_Dir(0X05,0Xac);//FLASH setting  this solved my issues babe
wohhhhho
    writeCommand(0xBF);
   _writeData(0x02);

    XY_Coordinate(X,Y);
    Active_Window(X,X+BWR-1,Y,Y+BHR);
   DMA_block_mode_size_setting(BWR,BHR,SPWR);
   DMA_Start_address_setting(start_address);

    writeCommand(0xBF);
   _writeData(0x03);

   Chk_DMA_Busy();

//Set cursor position
delay(10);
setCursor(x7, y7);
//Set window position
delay(10);
setActiveWindow(x1, x2, y1, y2);

 }

I have to find some time to upload this to my git

Regards

gcharles81, 2018-10-11:
Hello Check the below link I uploaded my changes , note this has to be
cleaned properly later

https://github.com/gcharles81/RA8875/tree/gcharles81-patch-1

Regards

anujgargiitk, 2018-10-11:
Thanks Charles :)