sumotoy/RA8875 #127: RA8875::_initializeFT5206(void)   Loop problems?

GremlinsInGame, 2017-10-05:
```
void RA8875::_initializeFT5206(void)
{
	uint8_t i;
	for (i=0x80;i<=0x89;i++){
		_sendRegFT5206(i,_FT5206REgisters[i-0x80]);
	}
	_sendRegFT5206(0x00,0x00);//Device Mode
}
```

**Gives Warnings:**
RA8875.cpp: In member function void RA8875::_initializeFT5206()
RA8875.cpp: 5265:43: warning: iteration 9u invokes undefined behavior [-Waggressive-loop-optimizations]
   _sendRegFT5206(i,_FT5206REgisters[i-0x80])
RA8875.cpp:5264: note  containing loop
   for (i=0x80;i<=0x89;i++){

Could this be a problem? And how to solve?

M-rTom, 2017-10-05:
What happens if you try something like this?

 

for (i=0;i<=9;i++)
  _sendRegFT5206(0x80+i,_FT5206REgisters[i]);

 

 

Tom

GremlinsInGame, 2017-10-07:
Thank you for the input. I adjusted my code based on your suggestion and then
I just realized that in RA8875UserSettings.h that the _FT5206REgisters[] Array is getting filled with 9 Values but the For Loop is being iterated 10 times..  i<=9
This is/was stuffing a random memory value into the FT5205 0x89 register.
I imagine this causes problems for people.

Also,
Some of the the values that are in _FT5206REgisters[] don't seem to be correct. 
Default Values that are listed in the FT5206 Datasheet seem to work better for me.

seamaster101, 2017-11-01:
Hi GremlinsInGame,
I read your post for soem of the values for _FT5206REgisters[]. Below is what i found in RA8875UserSettings.h. Is that what you are referring too? 
`#include "Wire.h"//include the support for FT5206
		static const uint8_t _FT5206REgisters[9] = {
			0x16,0x3C,0xE9,0x01,0x01,0xA0,0x0A,0x06,0x28
		};`

what are the new values that you are using? I did look in the datasheet but wasn't sure what to replace these values with.
can you post the line for the values that you are using?
Thanks!

GremlinsInGame, 2017-11-02:
The Values I am using now are:
```
static const uint8_t _FT5206REgisters[9] = {
0x3C,0x30,0x02,0x5C,0x0A, 0x02,0x00,0x02,0x0C
};
```
You can find the Default Values and descriptions of the Registers starting on Page 34 of this datasheet.
http://www.buydisplay.com/download/ic/FT5206.pdf

I think the Defaults for the FT5206 are (In Decimal)
//70,  60,  16,  60,  10,  20,  0,   2,   12