sumotoy/RA8875 #129: EastRising 5" 480x272 E325364 not initializing?

MrBambix, 2018-03-02:
Hi Sumotoy,

Not sure if this is the right place to ask nor do i want to be bothering you... Just mention it and I will take the post away.

Anyway I am trying to use an EastRising 5" display with your library with product number 325364, ebay URL:

https://www.ebay.com/itm/Serial-SPI-5-5-0-inch-TFT-LCD-Module-Display-w-Capacitive-Touch-Panel-Tutorial-/291861890941?hash=item43f453077d&tfrom=302058660713&tpos=unknow&ttype=price&talgo=origal

Anyway, it seems it just does not switch on or the backlight stays off for some reason. This is configured for 4 wire spi allthough I tried with additional RST and INT on the JP3 header.

Allmost feel like giving up and maybe my display is faulty but if you have any ideas to try that would be great.

btw it is running on external 3.3V power.

Thanks for looking and regards 

Leo

riuson, 2018-03-02:
@MrBambix, when you buy this display, what power suply you have specified? 5V or 3.3V?
Because their PCB can have possibility use one or another, by selection right jumpers.

MrBambix, 2018-03-02:
Hi Vladimir,

 

Thanks for your reply!

 

Specified to 3.3V looks right according to the documentation that they indeed set it to 3.3V U2 regulator and C’s present on this board though.

 

Greetings,

Leo

riuson, 2018-03-02:
Please give a link to exactly your display on their home site: https://www.buydisplay.com/default/tft-display/5-inch
And attach photo of PCB from component's side.

MrBambix, 2018-03-02:
Hi Vladimir,

 

Bought if of ebay through the link below:

 

https://www.ebay.com/itm/5-5-0-inch-TFT-LCD-Module-Display-w-RA8875-Touch-Panel-I2C-Serial-SPI-Tutorial/291812803605?ssPageName=STRK%3AMEBIDX%3AIT <https://www.ebay.com/itm/5-5-0-inch-TFT-LCD-Module-Display-w-RA8875-Touch-Panel-I2C-Serial-SPI-Tutorial/291812803605?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2060353.m2749.l2649> &_trksid=p2060353.m2749.l2649

 

Please find below the picture of the display.

 

Thanks for your help !

 

Greetz

Leo

riuson, 2018-03-02:
@MrBambix, images from email's are not supported on GitHub. You need to login here.
Your link to ebay just shows common info about display. As I understand, 3.3V power supply option was specified.

MrBambix, 2018-03-02:
Vladimir,

https://www.buydisplay.com/default/tft-5-inch-lcd-display-module-controller-board-serial-i2c-ra8875

Is this link on the buydisplay website. On my request they have made it ready for 4 wire-SPI and 3.3V for which the jumpers seem to be set correctly as far as i can see from the documentation.

regards,
Leo

MrBambix, 2018-03-02:
[image]

riuson, 2018-03-02:
Ok.

> J8 **Short**: Vdd=3.3V Power Supply
> J8 **Open**: Vdd=5V Power Supply.

As I can see, it is short, 3.3V.
Also you can check, what your power supply delivering 3.3 V with 450 mA successfully and voltage is stable.
And... Check this:

- Pin 14 BL_Control on FFC connector CON2, 
- Backlight Conrol:
  - J15 Short, J16 Open: Select Backlight Control Signal with External Input
  - J15 Open, J16 Short: Select Backlight Control Signal with RA8875 'PWM'

MrBambix, 2018-03-02:
Hi Vladimir,

 

J8 is closed so indeed it is at 3,3V.

For the backlight control jumpers, J16 closed and j15 open. Am I correct in that that is the way it should be for the library to control the backlight ?

 

My lab powersupply seems to be stable at 3,3V but only 0,02Amps drawn with Arduino mega on via usb and the lab supply powering the display directly. Doesn’t change a bit it seems…

 

Not sure what you mean by: Pin 14 BL_Control on FFC connector CON2 ?

I currently have it setup with : INT on pin 11 and RESET on pin 12 of CON2 btw as per description on the wiki tried utilizing those extra in the Arduino sketch no difference in behavior though.. 

 

Thanks and greetings,

Leo

riuson, 2018-03-02:
> For the backlight control jumpers, J16 closed and j15 open. Am I correct in that that is the way it should be for the library to control the backlight ?

It should be controlled via RA8875. Perhaps this is the responsibility of the library. I do not know, because I'm not familiar with this library.
You can check average voltage or signal (with oscilloscope) on RA8875 PWM output.

> Not sure what you mean by: Pin 14 BL_Control on FFC connector CON2 ?

Taking into account the state of the jumpers J15 & J16, this pin no longer affects.

MrBambix, 2018-03-03:
Well,

Did a complete rewire and now I am a bit further. The backlight comes on but so far only a white screen nothing drawn upon it..

regards,
Leo

MrBambix, 2018-03-03:
Wow accidentally i put on a bit more voltage then 3.3V, where this board should be configured for and voila it started working at 3.8V ! weird but ow well. Quite sure the jumpers on the board are set for 3.3V and the voltage regulator also puts out 3.3V ow well.. .Thanks for thinking along with me here and maybe this information will help others. Regards, Leo

riuson, 2018-03-03:
> the voltage regulator also puts out 3.3V ow well..

No...
When jumpers configured to 5.0V, there is U2 regulator (5.0 -> 3.3) is used.
When jumpers configured to 3.3V, then U2 regulator is just shortened (by J8) and not used.

MrBambix, 2018-03-03:
Hi

Wel that is somewhat strange then as there is definitely coming 3.3v into u2 and J8 is shortened... Will investigate Some more tomorrow 😁

Regards

Leo

Verstuurd vanaf mijn iPhone