sumotoy/RA8875 #134: RA8875 TFT Capacitive touch screen not responding

Amomah, 2018-08-28:
Hi There,

Can anyone help please me understand why only one of the two Codes work with my 7" TFT RA8875 Capactive TFT touch Screen and already uncommented RA8875UserSettings.h file and uncomment USE_FT5206_TOUCH. I am using Mega 2560 Arduino.

There are 2 points I seek understanding (Code1 = works) and (Code2 = doesn't work):

**1- The following Code work downloaded from BuyDisplayOnline website;**

/***************************************************
//Web: http://www.buydisplay.com
EastRising Technology Co.,LTD
Examples for ER-TFTM070-5 with Capacitive Touch Panel 
Display is Hardward SPI 4-Wire SPI Interface and 5V Power Supply
Capacitive Touch Panel is I2C Interface
Tested and worked with:
Works with Arduino 1.6.0 IDE 
****************************************************/
#include <stdint.h>
#include <SPI.h>
#include <Wire.h>

uint8_t addr  = 0x38;

#include "Adafruit_GFX.h"
#include "Adafruit_RA8875.h"

//LCD:hardware SPI  CTP:hardware IIC
//Arduino DUE,Arduino mega2560,Arduino UNO
#define RA8875_INT 4
#define RA8875_CS 10 
#define RA8875_RESET 9
#define FT5206_WAKE 6
#define FT5206_INT   7    
Adafruit_RA8875 tft = Adafruit_RA8875(RA8875_CS, RA8875_RESET);
uint16_t tx, ty;

enum {
  eNORMAL = 0,
  eTEST   = 0x04,
  eSYSTEM = 0x01
};

struct TouchLocation
{
  uint16_t x;
  uint16_t y;
};

TouchLocation touchLocations[5];

uint8_t readFT5206TouchRegister( uint8_t reg );
uint8_t readFT5206TouchLocation( TouchLocation * pLoc, uint8_t num );
uint8_t readFT5206TouchAddr( uint8_t regAddr, uint8_t * pBuf, uint8_t len );

uint32_t dist(const TouchLocation & loc);
uint32_t dist(const TouchLocation & loc1, const TouchLocation & loc2);

bool sameLoc( const TouchLocation & loc, const TouchLocation & loc2 );

uint8_t buf[30];

uint8_t readFT5206TouchRegister( uint8_t reg )
{
  Wire.beginTransmission(addr);
  Wire.write( reg );  // register 0
  uint8_t retVal = Wire.endTransmission();
  
  uint8_t returned = Wire.requestFrom(addr, uint8_t(1) );    // request 6 uint8_ts from slave device #2
  
  if (Wire.available())
[... 458 lines cut ...]
    tft.print(tft.getTouchState(),HEX);
    //you need to get the coordinates? We need a bidimensional array
    uint16_t coordinates[MAXTOUCHLIMIT][2];//to hold coordinates
    tft.getTScoordinates(coordinates);//done
    //now coordinates has the x,y of all touches
    //now draw something....
    uint16_t tempCol;
    for (uint8_t i=1;i<=tft.getTouches();i++){
      if (i == 1)tempCol = RA8875_RED;
      if (i == 2)tempCol = RA8875_GREEN;
      if (i == 3)tempCol = RA8875_MAGENTA;
      if (i == 4)tempCol = RA8875_CYAN;
      if (i == 5)tempCol = RA8875_YELLOW;
      tft.fillCircle(coordinates[i-1][0],coordinates[i-1][1],10,tempCol);
    }
    tft.enableCapISR();//rearm ISR if needed (touched(true))
    //otherwise it doesn't do nothing...
  }
  #endif
}

  ****************************************************/

Appreciate your time and help.
AMOMAH

atilla44, 2020-03-06:
Hello i have a 9inch ra8875 display with capacitive touch 4 wire i tried everything but the touch has no reaction i would be grateful for any help thanks

gilphilbert, 2020-10-31:
The buydisplay RA8875 capacitive touchscreen models don't use the same touchscreen as most other RA8875 boards; they use GLS1680 displays. They're not easy to get working, so you may need to hunt around for the right library