sumotoy/RA8875 #135: Several errors using custom font

Spomky, 2018-09-09:
Hi,

I created a custom font using the wiki page step by step tutorial.
I have a .c file that contains my font, but when I try to use it, I have the following compilation errors:

* error: unknown type name 'tFont'
* error: unknown type name 'tChar'
* error: unknown type name 'tImage'

Any idea?

riuson, 2018-09-09:
You have to include header file with structure's definition.
There is a strange way of programming in Arduion's area...

Examples/_Energia_IDE/renderTextExample3/renderTextExample3.ino :
```
#include <SPI.h>
#include <RA8875.h>
#include "fonts/akashi_36.c"
```
RA8875.h:
`#include "_settings/font.h"`

_settings/font.h:
`typedef struct  ... tImage;`

Spomky, 2018-09-09:
Thank you for the quick reply.
I’ve already included these header files

```
#include <SPI.h>
#include <RA8875.h>
#include "gilles.c"
```

It works fine using the fonts in the `fonts` folder, but not with that one I created.
`gilles.c` code is as follow:

<details>
```

/*******************************************************************************
* generated by lcd-image-converter rev.c6da9ce from 2018-02-11 00:03:45 +0500
* font
* filename: unsaved
* name: gilles
* family: Gill Sans MT
* size: 32
* style: Normal
* included characters:  !"#$%&'()*+,-./0123456789:;<=>?\x0040ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~\x007fÀÁÂÃÄÅÆÇÈÉÊËÌÍÎÏÐÑÒÓÔÕÖ×ØÙÚÛÜÝÞßàáâãäåæçèéêëìíîïðñòóôõö÷øùúûüýþÿ°
* antialiasing: no
* type: proportional
* encoding: ISO-8859-1
* unicode bom: no
*
* preset name: Monochrome
* data block size: 8 bit(s), uint8_t
* RLE compression enabled: no
* conversion type: Monochrome, Edge 128
* split to rows: yes
* bits per pixel: 1
*
* preprocess:
*  main scan direction: top_to_bottom
*  line scan direction: forward
*  inverse: no
*******************************************************************************/

 typedef struct {
     long int code;
     const tImage *image;
     } tChar;
 typedef struct {
     int length;
     const tChar *chars;
     } tFont;

#include <stdint.h>

#if (0x0 == 0x0)
static const uint8_t image_data_gilles_0x20[80] = {
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    // █████████
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80, 
    0xff, 0x80
};
static const tImage gilles_0x20 = { image_data_gilles_0x20,
    9, 40, 8};
#endif

…

const tFont gilles = { 161, gilles_array };
</details>

riuson, 2018-09-09:
I think this because you are trying compile font's source file.
But content of this file compiled inside other file:
```
#include <SPI.h>
#include <RA8875.h>
#include "gilles.c"
```
This is a very common mistake in the world of Arduino: [Problems with Bitmaps on TFT Display using GFX Library](https://forum.arduino.cc/index.php?topic=546304.0)

Spomky, 2018-09-09:
OK, I will read this thread.
I noticed that my file does not have any `__PRGMTAG_` definition.
I am wondering if I missed an option during the conversion of the font…

riuson, 2018-09-09:
All options are applied when font converted with right template file.

Spomky, 2018-09-09:
OK looks like the template was not correctly load (even if I quit and run the software again).
Sorry for the noise. Compilation without any error now.
Regards.

closed 2018-09-09 by Spomky