sumotoy/RA8875 #136: ESP32 and ESP8266

JaspaJami, 2018-09-21:
Any plans to continue with this project and to get it working in ESP32 and ESP8266?

JanKozub, 2018-12-30:
Anyone tested it with ESP32? I really need to know is it working.

knopserl, 2019-05-05:
I would also be interested in  fast SPI supporting library for the ESP32. But it seem that the further development has either stopped or is getting very low focus.

JanKozub, 2019-05-05:
I tested esp32 with RA8875 and collected a lot information about development of this setup. If you have and questions about it e-mail me here: j.kozub136@gmail.com. If you haven't bought parts yet i want to infom you that it's not that easy to use it with esp32 and you may want to change your setup.

knopserl, 2019-05-19:
Hi,
I'm looking for a good graphics library for a 5 or 7" display with 800x480 (=minimum or even larger) resolution which is fast, support SPI and touch which can run on an ESP32 MCU. Everything else is not an option. I neither want to use an Arduino or a Raspberry. I really want to use the ESP32 (I need the ESP-now protocol) and Bluetooth and other features of the ESP32. It has enough RAM and Flash thus it has no good alternative. And I also don't want to use an extra OS (like Debian etc).
I was using a 5" Nextion display with UART, but it also has a couple of drawbacks (bad font support if you don't pay extra per each font).
So I was researching on IoT MCU displays for 800x480 and only found the SD1963 and RA8875 types. Looking on the support, speed, age of the chips, I found the RA8875 much more  attractive as the old SD1963 (which is also slow, the RA8875 is much faster and more capeable). Thus I found this/your library as it support RA8875 SPI connected displays.However, I have found your message, that it is difficult to use it with ESP32. So before I order such as display (like below), I wanted to get your opinion, if I get it to run reasonable.
 https://www.buydisplay.com/default/5-inch-tft-lcd-module-800x480-display-controller-i2c-serial-spi
Thanks a lot for your hints/support.
Heinz

JanKozub, 2019-05-20:
Hey,
I tested my setup with 10 inch display with RA8875(i've got one for sale if you are interested and i can send some demos of it working) and at the beginning i want to say that you can't use this library. For what i know only working library for esp is one created by adafruit https://github.com/adafruit/Adafruit_RA8875. That's because esp doesn't support type of spi which is used in sumotoy's project. Unfortunately both libraries are not the best for complex projects what forced me to use raspberry. SPI is pretty fast and it's enaught for simple usages. This is basic information which should help you whether to go for it or not. In case that you are encouraged to buy setup similar to mine just message me and i can send you some samples.
Jan

knopserl, 2019-05-20:
@JanKozub Hi Jan, thanks for your quick reply and your hints. I have only found this (sumotoy) and the one from adafruit (you have mentioned) which seem to be options for the RA8875. I have also seached for any reasonable supported 5" dispaly with 800x480 resolution for IoT devices (not Raspberry). I have a couple of RPi 3+ but they use HDMI and I need the ESP32 and I have only found SD1963 and RA8875 based SPI displays. So I don't have much choice. The SD1963 looks even worse and so I was looking further at the RA8875 SPI display.
What are the problems with the SPI interface on the ESP32? I have not found incompatibilities with SPI on ESP32 to any other libraries? But if this library does not work, I certainly will switch to the Adafruit_RA8875.
What do you mean with "not the best for complex projects"? What are the complex things which don't work and what are the things which work well?
Unfortunatly 10" is to large  for me, I rather was looking for 5" or max. 7".
I would be interested in your samples!

regards Heinz

JanKozub, 2019-05-21:
Hi Heinz, by complex I meant that this library is very poor in functions and drawing many for example squares on screen might be visible for eyes. Also I want to mansion that creating a nice looking menu might be time consuming and hard when trying to make it perfect. Even my menu after many hours of work looked like one from 2005 and it wasn’t even close to these days standards. Basically that’s why I moved to raspberry, I wanted to make my menu modern and it wasn’t posing or with esp. For example now I’m programming my project in java and it’s 10 times faster and easier than programming esp. In your reply you wrote that you need some functions of esp like Bluetooth. If I were you I would just skip esp and start to think how to do it on raspberry because it possible for sure. For now I don’t have time to connect the display and send you samples of it working(if you really want these you will have to wait like 2 maybe 3 weeks) . If there is any other question you have, just reply under this topic and be sure that I’m gonna answer it.
Jan

gcharles81, 2019-05-21:
@knopserl This library is powerful but unfortunately it was never fixed to work on ESP and also creater stopped supporting it 2 years back  , 

I do use it and in my case the setup is I use teensy 3.6  to handle all the application and even load GUI icons from SPI flash its really fast and you can create nice applications 

In my case  I simply use an ESP8266 to provide Wifi connection to the Teeny , might be this setup is enough for you

knopserl, 2019-05-24:
@JanKozub I really want to use it (not on a RPi) and I want to use it on ESP32. So if you can send me your examples, that would be nice.
There is also a library from Adafruit (Adafruit_RA8875) I might also test this one.

knopserl, 2019-06-22:
@sumotoy I think you have created an incredible library which is feature rich and nothing come close to it. There is only one from Adafruit, but it has a fraction of this library.
As you know, the ESP32 is getting more and more popular and is the most powerfull SoC, you would do me and a lot of people a big favor in addin support also for ESP32.
THere is no other good chip available for 800x400 and MCU than the RA8875. The only altrenative is the much sower SD1963 but this taked 16 + many other pins and this is unusable.
Is there any chance to get EP32 support?

knopserl, 2019-07-08:
@JanKozub and @JaspaJami I got my RA8875 from buydisplay.com. a very nice 5 inch 800x480 touch display with external font rom and SPI interface.
I have tested several libraries with an ESP32 and this ony does not even compile. I have tested several versions. I found an old version from Max Costa (the author) which works fin with an ESP32, but a couple of features are missing in comparison to the current versions 0.70 to 0.79.
So Max seem to introduce a kind of FASTSPPORT features and added optimizations for several boards up to an ESP8266 but not ESP32. I have spent some hours but it's a LOT of work and not sure if I get it to run and if I will spent days of effort to compile and not even to run.
The Aadafruit RA8875 is not really a good alterantive, as a lot of features are missing and it has some bugs etc.
 

/*  www.buydisplay.com
	--------------------------------------------------
	RA8875 LCD/TFT Graphic Controller Driver Library
	--------------------------------------------------
	Version:0.55(early beta) introduces compatibility 
	with Energia IDE and his MCU's (tested!)
	++++++++++++++++++++++++++++++++++++++++++++++++++
	Written by: Max MC Costa for s.u.m.o.t.o.y

So either Max MC Cost could give us some hints or anybody else could also work on it?
I think the RA8875 and ESP32 are worth to be supported.

JanKozub, 2019-07-09:
@knopserl I disagree with your opinion about supporting ra8875 library for esp32 or any microprocessor this type. In my opiniom is too late for that, it’s 2019 and we have better alternatives for making working gui menus. If you want to make pretty GUI interface RA8875 is not enough anyway and I’m this case I would rather use raspberry pi especially when new version  came out and older ones gonna be cheeper. Maybe I would agree with you when we would be talking in area of displaying simple data but then you can download Tara and transfer it to arduino if you really want to use this library. That’s my thoughts on this topic and I would really like to know yours.

knopserl, 2019-07-12:
@JanKozub I disagree with your disagreement ;-) I ahve done acouple of projects with RPi and even better RPi clones. For a couple of projects, I don't even want to maintain a complete OS (like Raspian). It takes quite long to start, effort to maintian and keep it up to date with all updates/security fices, consume much more engery etc. etc. I have convinced even peipel how thought the RPi is the best option.
With the ESP32 I have WiFI, BLE 4.2, CAN Bus, HW Encryption, multitasking, I2C, I2S, SPI, SD Card etc. ect. etc. and it starts my firmware in 1sek.
So small Linux boards have it's use cases as well as SoC like the ESP32 (nd of course it cost around 3-4€).
I've got the RA8875 to run with some reasonable featureset, but I would like to get the latest version 0,79to run. But as the author of this best availble working library Max MC Costa sai, the new introduced accellerated SPI handling is very difficult to maintain and takes long tome to understand, needs rework and does currently not work with ESP8266 or ESP32. I'm not sure if I can do all he required changes. I would not need the support for all the other older SoC's.

I was also looking at uGFX as it claims to support also SoC like the ESP32 and also the RA8875 chip/displays. https://ugfx.io/  see the display controller under https://ugfx.io/platforms
My first attempts were not successful, but I was trying it with Arduino IDE, maybe I have more success with VS COde + Platform.io and I might not have the required knowledge yet.

knopserl, 2019-07-13:
@all OK, I got this lastest version to work with ESP32. It took me an hour and it was finally no so difficult.
My changes should not impact the other code and I have not tested everything such as the calibration with permanent storage, but I will do that later this night.
I've done tests with most of the drawing features and some text, but will do further testing with custom fonts, external ROM fonts etc. an let you know.
I don't know how to do a pull request and guess nobody will do it as the author seem to have left the support of this great lib.
But if anybody is interested, I can send the library or I just upload the archive in this post:

[RA8875-ESP32.zip](https://github.com/sumotoy/RA8875/files/3388826/RA8875-ESP32.zip)

knopserl, 2019-07-13:
> 
> 
> @knopserl This library is powerful but unfortunately it was never fixed to work on ESP and also creater stopped supporting it 2 years back ,
> 
> I do use it and in my case the setup is I use teensy 3.6 to handle all the application and even load GUI icons from SPI flash its really fast and you can create nice applications
> 
> In my case I simply use an ESP8266 to provide Wifi connection to the Teensy , might be this setup is enough for you
> 
> I also tested and confirm working on Arduino Due which can be also an good alternative to the Teensy , If you are interested I had to make some changes to make SPI flash work and below is the like top the updated library
> https://github.com/gcharles81/RA8875/tree/gcharles81-patch-1

@gcharles81 you can use now give it a try with my uploaded lib in the thread with a single ESP32 without an expensive Teensy and a second ESP8266 just for WiFi.

knopserl, 2019-07-14:
@all I have now tested most of the examples on my ESP32 (LOLIN32) and did a lot of changes and enehancements. E.g. the resistive touch did not work etc. I have now created a new version which I call 0.80b1. Here is the archive, if aybody wants to use it with the ESP32. I have also attached a picture which "proves" that it works (the easyPaint example) attached you see my ESP32.
I'm using a 5" 800x480 display from buydisplay.com.
[RA8875_v080b1.zip](https://github.com/sumotoy/RA8875/files/3389945/RA8875_v080b1.zip)
[image]

JanKozub, 2019-07-14:
@knopserl Great job!

gcharles81, 2019-07-14:
> @gcharles81 you can use now give it a try with my uploaded lib in the thread with a single ESP32 without an expensive Teensy and a second ESP8266 just for WiFi.

@knopserl  Great job I will for sure try it coming days , I have to solder few wires to my 7inch than I can test also with my LOLIN D32 pro , can you share the used pins ?

knopserl, 2019-07-15:
@gcharles81 my initial test was done with a Wemos LOLIN32 and I used the default SPI pins for a ESP32 board which are: MISO=19, MOSI=23, SCK=18 and for SS/CS I used 17 (default is 5) because the internal LED is also on 5.
BTW, I did not get the resistive touch with the sumotoy's version. I have tried several combinations without the IRQ pin as the buydisplay.com 5" displays use internal IRQ. So I took some code from the Adafruit library and adopted it and this works fine.
Meanwhile, I got a hint from the buydisplay support (which always reacts fast). There is a IRQ pin 10 on the JP3 parallel port (which is not used as I have ordered the SPI 4-wire interface), with that, the interrupt version can be used. I'll try that later.
Another hint: as JanKozub mentioned, that no complex interfaces can be built with the embedded systems and RA8875: I have started my first tests with **littlevgl** and a 3.5 inch display (ILI9488 controller) and this works fine! It should also support the RA8875. Wih that, complex UI's (even with a drag and drop editor) with tab's scollbars, can be build with little memory usage. Perfect for the ESP32 ;-)
Give it a try (supported also by Arduino IDE)
https://littlevgl.com/
https://github.com/littlevgl/lv_arduino
https://www.arduinolibraries.info/libraries/littlev-gl
https://github.com/littlevgl
https://platformio.org/lib/show/6228/littlevgl

maybe we can help each other, as I don't see any other good option for 5"+ displays on ESP32.
For smaller displays, I use 3.5 ILI9488 (480x32) or 2.8" ILI9341 (320x240) which is cheap and easy using the TFT_eSPI library (which works with the littlevgl).

gcharles81, 2019-07-15:
@knopserl 

Thanks for the feedback give me few days and I will test both 5" and 7" with your version , 
I can than  show you ,  cool gui with icons on Ra8875 using SPI Flash , I have the setup done

gcharles81, 2019-07-16:
@knopserl  works great tested only 5 inch so far using graphics test and set max SPI speed to #define _FASTSSPORT in user settings below are the benchmark results

19:00:09.696 -> screen:800x480 - rotation:0
19:00:09.696 -> Benchmark                Time (microseconds)
19:00:09.696 -> Screen fill              16737
19:00:10.214 -> Test Pixel               174
19:00:10.726 -> Test Pixels              73463
19:00:11.318 -> Rectangles (outline)     286565
19:00:12.143 -> Triangles (outline)      36050
19:00:12.693 -> Triangles (filled)       77119

Soon I will try the 7 inch capacitive version and after that I show you how cool gui we can make , I modified a bit to make SPI flash working and loading fast Icons than you can really do cool Gui , we can collaborate if you want to improve this library further - I have to build my test setup  than for ESP32

knopserl, 2019-07-16:
@gcharles81 perfect, thanks for the testing and good feedbak. I will certainly use then your improved version with SPI flash etc.  and I'm also interested in your cool GUI's.

I encourage you to use the ESP32, as it has enough flash, (min 4, I have ordered 16MB modules and goes up to 64MB) has 430k RAM (with 8MB PSRAM option, I have such ESP32) and it has dual core 240MHZ. So quite a good HW base.

Meanwhile I'm also looking at littlevgl and uGFX. It works with ILI9341 and ILI9488 (320x240 and 480x320 TFT's) and I'm also trying to get it work with the 5" RA8875.

gcharles81, 2019-07-16:
@knopserl  any reason why you added this my 7" has capacitive touch and I cannot disable Restive touch

knopserl, 2019-07-16:
@gcharles81 Have you set/configured the capacitive touch usage in the RA8875\_settings\RA8875UserSettings.h?
see below?
And what compile error do you get?
The eeprom is only used for the resistive touch and for calibration and only if your MCU supports eeprom. I'm wonderuing, that you get a compile error and where do you get it?

/* [CHOOSE YOUR TOUCH SCREEN TYPE] +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
This library supports the RA8875 internal resistive touch screen driver or a FT5206
based capacitive touch screen driver. Below, you have to choose one of them.
The Resistive Touch screen it's drived directly by RA8875.
The Capacitive Touch Screen use a I2C chip called FT5206 (hardwired to address 0x38).
Please choose at list one (NOT both), if you comment both it's the same as _AVOID_TOUCHSCREEN*/

#define USE_RA8875_TOUCH	//resistive touch screen
//#define USE_FT5206_TOUCH	//capacitive touch screen

knopserl, 2019-07-18:
@gcharles81 Maybe you can also have a look at the issue with the mirrored font/text if rotation is used: https://github.com/sumotoy/RA8875/issues/140 (open this year).
I took a quick look and have not found where it fails, bt I confirm the error, if rotation 1 (I guess also 3 = vertical) is used, the text is mirrored an not usable.

leic3d, 2019-10-20:
@knopserl 
Hi, i am sorry to dig this out after some months. To the time of your conversation here, i started my first project with an ESP32 and to that time found an ILI9341 320x240 display from china and used littlevgl to make a full menu for a controller of some power supplies with RTC and so on.
After 2 months and many experiences the project is 95% finished and i seem to know what to make different next time. The biggest issue i find is the small display and the resistive touch, which is so much not 2019, so i decided to invest first of all in a better display. After searching the web, i ended up in the same solution like you: 5-7 inch screen with SPI, as all alternatives consume lots of pins and seem to be slower anyway. ESP32 is the way to go for easy controlling of some relais or optos, no need to have an OS. Also it seems the RA8875 is the only solution there. 
As i developed a code based on littlevgl already, i am very interested in getting such display to work on the ESP32 and have the porting for a working driver done. I found your posts on the lvgl forums (and github pages :) ) as well and am additionally surprised, that the amout of ppl, that want this combination to work is as small as 2-3. 
Would you be willing to share your recent status on this? Did you manage to port that into lvgl? Is the update time of such big screen useful at all? If it is useful, i can contribute to sth.
Thank you very much,
Michael

knopserl, 2019-11-15:
@leic3d I spent effort on the RA8875 code to get it run on ESP32 and it works so far (at least with the resistive touch). I have started to use the RA8875 with ESP32 and littlevgl, but had not yet enouh time and currently no urgency but I would like to have this combination running to support 5" displays with advanced UI libs (like littlevgl or ugfx).
There was a lady (Maker Melissa,  https://twitter.com/makermelissa) who got it to run (not using this library, but I guess the adafruit one). So far I have heard that it's quite slow and need to use the speed-up from this lib. So it needs effort to use this lib and use it with littlevgl. So if you give it a try, I might join the effort.

regards 
Heinz

erikhjertholm, 2019-11-20:
Hey guys! You seem to be the only ones on the interwebs that might be able to help me solve my problem. 

I am using the RA8875 together with a ESP32-wrover (for the extra spram) running a 7" 800x480 display with resistive touch from buydisplay. At the moment I am using the library from Adafruit, but I reeeeeeally need the **dual layer functionality** that is seems like the Sumotoy library supports. Do you think this is possible with my setup?

I am working on a graphical editor where you can use the standard shapes in excel to create a design, and then have the shapes translated into c++, even with support of grouping. If you can help me, I will of course share this when it is done. 

Also, are there any other advantages using the Sumotoy version instead of Adafruit? @knopserl @gcharles81

knopserl, 2019-11-25:
@erikhjertholm This sumotoy lib is a much enhanced lib compared to the Adafruit lib. But it did not work with ESP32, that was the reason why I spent some effort to modify it for ESP32. 
Here is my fork https://github.com/knopserl/RA8875/tree/0.80b1 give a try. It has a two layer functionality but I did not test it. I have by far not the deep inside as the original author but I think it was also not just improved by functionality but also by speed. There are some bugs like mirrored fonts when using rotation etc. But at least it's usable and I did some examples/demos which worked well.

Tszul, 2020-03-21:
Hello friends. I did it, thanks for your help. These are the first steps for my thesis to obtain the title of mechanical electrical engineer. I will continue working to obtain a professional interface.
[image]

knopserl, 2020-03-22:
@Tszul You can also contribute and try to fix the mirrored fonts if they are rotated.
I think it's a good library, as there are not many available for 5" displays.
THere is another 5" display available now with LT7683 controller which is also very fast. I have also sucessfully ported it to ESP32.
Here is the link to the controller company/download
http://www.levetop.tw/en/download1-268.html
and here the displays I have ordered and are working
https://de.aliexpress.com/item/4000282154239.html?spm=a2g0s.9042311.0.0.27424c4d76KZoQ

Tszul, 2020-03-24:
> 27424c4d76KZoQ

Thank you. I will try to contribute to the best of my ability and
upload contributions. 😄😎

pacmenDE, 2020-07-09:
Hy all,

did you manage to run the esp32 with ra8857 and lvgl?
I ve bought 7 inch display from buydisplay.com  lan can't get the device to run.
do you have any information for me.
Thanks and a nice day, patrik

Tszul, 2020-07-09:
Hi Patrick.

At the moment I have not continued working with this project. Not
ostentatiously, I did not use the LVGL library, only the library's own
functions, well, the truth is that I did not know how to link the
Arduino IDE with the LVGL libraries.

gilphilbert, 2020-10-23:
@knopserl, do you still have the code for the LT7683 controller? Some of the displays are capable of true color, which is pretty nice for an MCU-attached display.

I was thinking about getting one - sounds like you've had success with this display already. I'd have to write an LVGL driver for it though.

knopserl, 2020-11-15:
I have code for the LT7683 controller and I have adapted and tested it for ESP32. No I have no LVGL driver. Not enough time and not yet an urgent use case for it. But if you like to do it, I can send you the source.

petermccracken, 2021-10-29:
Hey @knopserl Can you share the lib you are using for the ESP32 and LT7683 displays ?
I've been playing around with the RA8875 and wouldn't mind having a go with one of these displays off Ali.
Thanks!

knopserl, 2021-10-31:
@petermccracken here are the designed libs
Link to my adapted RA8875 repository: https://github.com/knopserl/RA8875
and the ZIP archive with the ESP32 version of the LT768. Both work for me.
[ESP32_LT768.zip](https://github.com/sumotoy/RA8875/files/7448353/ESP32_LT768.zip)

The forked RA8875 lib has two limitations:
The rotation of text does not work propoerl and only a resistive touch is currently supported.
If you improve both, you can share the code with me.

knopserl, 2021-10-31:
@gilphilbert sorry, I did not read this thread for a while. See the ZIP-File in my above post for the LT768. If you have a LVGL driver, you might share the solution with me

TheConceptBoy, 2023-01-06:
Ok if anyone, like me, was googling in hopes of getting this to run as an alternative to the half assed Adafruit library, I just got it to work in Arduino IDE in Jan 6 2023.

I downloaded this version from the above provided link:
https://github.com/knopserl/RA8875/tree/0.80b1

First thing is that you'll need to modify the library folder
it has to have both _includes and _settings folders.
Then you need to move some files from one folder into the other like so:

[image]
[image]

Finally you need to go to RA8875Registers.h and uncomment the two lines that the compiler would complain about otherwise.

	#define RA8875_TPCR0_ENABLE           0x80
	#define RA8875_TPCR0_DISABLE          0x00