sumotoy/RA8875 #139: 5"TFT LCD Display Capacitive Touchscreen 800x480 with RA8875 Compatible?

Dancopy, 2019-03-19:
Is this library compatible with this card?
https://www.ebay.com/itm/Serial-5-TFT-LCD-Display-Capacitive-Touchscreen-800x480-with-RA8875-Pin-Header/302748193881

Thanks

GremlinsInGame, 2019-03-23:
Yes. I use the 4.3" and 7" Versions of that display all the time from buydisplay.com

Dancopy, 2019-03-23:
Thanks GremlinsInGame
Does it work with Nodemcu esp8266?

GremlinsInGame, 2019-03-23:
I haven't used those so can't be sure. I use the Teensy boards.

ik3umt, 2019-11-15:
I'm going to use Teensy3.2
Does this library work ?
Can this display be used ? https://www.buydisplay.com/default/tft-5-inch-lcd-display-module-controller-board-serial-i2c-ra8875

Thank you !

GremlinsInGame, 2019-11-16:
Yes. Works great with the Teensy 3.2. I just use the 4.3" and 7" Versions of that display but it should work the same.